Koenkk/zigbee2mqtt #449: WXKG12LM is "not supported" / has a zigbee model that _almost_ matches ones in shepard devices.js

kquinsland, 2018-10-05:
Hi there. I want to say thanks for putting this together.  It's been very useful / helpful for getting my HA setup running and free of hubs.  Appreciated!

I am currently running into an issue attempting to get a few of the *new* Aquara switches working (model: `WXKG12LM` - it has vibration sensing as well).

Here's the output of the z2m log.  I have debugging on:

```
cat configuration.yaml | grep level
  log_level: debug
```

I flashed the firmware on the radio about 4 hours before capturing this log output. I have been able to successfully pair several WXKG11LM (older gen aqara btns; only support 1,2,3,4x click).

```
  zigbee2mqtt:error 2018-10-5 02:48:19 Cannot get the Node Descriptor of the Device: 0x00158d0002012850 (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2018-10-5 02:48:44 Cannot get the Node Descriptor of the Device: 0x00158d0002012850 (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2018-10-5 02:49:12 Cannot get the Node Descriptor of the Device: 0x00158d0002012850 (Error: Timed out after 10000 ms)
  zigbee2mqtt:debug 2018-10-5 02:49:43 Recieved zigbee message of type 'devInterview' with data '"0x00158d0002012850"'
  zigbee2mqtt:info 2018-10-5 02:49:43 Connecting with device...
  zigbee2mqtt:info 2018-10-5 02:49:43 MQTT publish, topic: 'z2m/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-10-5 02:49:46 Recieved zigbee message of type 'devInterview' with data '"0x00158d0002012850"'
  zigbee2mqtt:info 2018-10-5 02:49:46 Connecting with device...
  zigbee2mqtt:info 2018-10-5 02:49:46 MQTT publish, topic: 'z2m/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-10-5 02:49:50 Recieved zigbee message of type 'devInterview' with data '"0x00158d0002012850"'
  zigbee2mqtt:info 2018-10-5 02:49:50 Connecting with device...
  zigbee2mqtt:info 2018-10-5 02:49:50 MQTT publish, topic: 'z2m/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-10-5 02:49:53 Recieved zigbee message of type 'devInterview' with data '"0x00158d0002012850"'
  zigbee2mqtt:info 2018-10-5 02:49:53 Connecting with device...
  zigbee2mqtt:info 2018-10-5 02:49:53 MQTT publish, topic: 'z2m/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-10-5 02:49:53 Recieved zigbee message of type 'devIncoming' with data '"0x00158d0002012850"' of device '0x746977735f726f736e65732e696d756c' (0x00158d0002012850)
  zigbee2mqtt:info 2018-10-5 02:49:53 Device incoming...
  zigbee2mqtt:info 2018-10-5 02:49:53 MQTT publish, topic: 'z2m/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2018-10-5 02:49:53 New device with address 0x00158d0002012850 connected!
  zigbee2mqtt:info 2018-10-5 02:49:53 MQTT publish, topic: 'z2m/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0002012850"}'
  zigbee2mqtt:warn 2018-10-5 02:49:53 Device with modelID '0x746977735f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-10-5 02:49:53 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-10-5 02:49:53 Recieved zigbee message of type 'devStatus' with data '"online"' of device '0x746977735f726f736e65732e696d756c' (0x00158d0002012850)
  zigbee2mqtt:warn 2018-10-5 02:49:53 Device with modelID '0x746977735f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-10-5 02:49:53 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-10-5 02:50:03 Recieved zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device '0x746977735f726f736e65732e696d756c' (0x00158d0002012848)
  zigbee2mqtt:warn 2018-10-5 02:50:03 Device with modelID '0x746977735f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-10-5 02:50:03 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-10-5 02:50:06 Recieved zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device '0x746977735f726f736e65732e696d756c' (0x00158d0002012850)
  zigbee2mqtt:warn 2018-10-5 02:50:06 Device with modelID '0x746977735f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-10-5 02:50:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-10-5 02:50:06 Recieved zigbee message of type 'devChange' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device '0x746977735f726f736e65732e696d756c' (0x00158d0002012850)
  zigbee2mqtt:warn 2018-10-5 02:50:06 Device with modelID '0x746977735f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-10-5 02:50:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

The first three logs are from when i started the pairing procedure (press the little button until blue-light flashes)

It looks like there's an issue with the line `Device with modelID '0x746977735f726f736e65732e696d756c' is not supported.`. I get this feeling that `0x746977...756c` is a bug of some sort?  This seems like it's very *close* to the other device names listed [here](https://github.com/Koenkk/zigbee-shepherd-converters/blob/9d33e18c7ce319080917626ff36fdff1d968a107/devices.js#L86):

```
echo "0x746977735f726f736e65732e696d756c" | xxd -r -p
tiws_rosnes.imul⏎
```

Is this a bug or possibly something in the firmware of the stick?  or is this just a bizarre thing from _my_ buttons?

Thoughts?

ryanbeaton, 2018-10-05:
It’s like the issue AnhDuc85 was having with the cube.
https://github.com/Koenkk/zigbee2mqtt/issues/388

“Fixed in the dev branch”

kquinsland, 2018-10-05:
Yep, i *just* found that. @ryanbeaton.

I guess this can be closed / more info here: https://github.com/Koenkk/zigbee2mqtt/issues/388#issuecomment-424858076

kquinsland, 2018-10-20:
after updating to release 0.1.8 i am now able to pair `WXKG12LM` sensors.  Closing this issue.

closed 2018-10-20 by kquinsland

PlayedIn, 2018-10-29:
@kquinsland  Have you been able to use the vibration state of the WXKG12LM?  I'd really like to find a way to trigger an automation with vibration. I didn't want to order it unless its a possibility.

kquinsland, 2018-10-29:
@PlayedIn i have not tried. I can see that the state does change on my HA instance though.  I can't imagine that it'll be that difficult, though; it should work exactly the same as detecting a single or double click.  Please open a support ticket on the HA forums to ask about how to do so.