Koenkk/zigbee2mqtt #453: Double response messages when publishing to zigbee2mqtt/<deviceid>/set

simonvanderveldt, 2018-10-06:
When I publish a message to `zigbee2mqtt/<deviceid>/set` I get two messages in return to `zigbee2mqtt/<deviceid>`, the first one seems to echo the command sent to `/set`, the second one seems to be the actual response from the device, for example when I publish `{"state": "off"}` to `zigbee2mqtt/<deviceid>/set` I get the following two responses:
```
zigbee2mqtt/<deviceid> {"state":"off"}
zigbee2mqtt/<deviceid> {"state":"OFF"}
```
These logs only show up if the payload is recognized, i.e. if I send `{"state":"foo"}` nothing happens.
The second message is always in full caps, the first one is the same as the message I sent to `/set`.

Is this expected behavior?

JLeandroFreitas, 2018-10-06:
I can confirm i got the same issue after i uppgraded to 0.1.6

simonvanderveldt, 2018-10-07:
Is this a duplicate of #360 ?

Koenkk, 2018-10-07:
This is expected behavior, zigbee2mqtt mocks a state message when the bulb is turned on/off, for two reasons:

- Some bulbs don't send their OFF state when they are turned off (in your case it did), this would mean that you won't get an acknowledge.
- In a situation where home assistant (or any other home automation software) bulb state is not in sync; e.g. actual state is ON, home assistant state is OFF. Home assistant would send an ON command but because the bulb is already ON, it would never acknowledge the ON command, in that case home assistant is stuck in the OFF state.

simonvanderveldt, 2018-10-07:
@Koenkk Ok, thanks for the information/context.
Would it be possible to add an option to disable this behavior? It makes it impossible to be sure you're getting the actual state from the device, pretty much negating the use of a command and a state topic to ensure separation between requesting changes and the confirmation that a state change has been applied.

> * In a situation where home assistant (or any other home automation software) bulb state is not in sync; e.g. actual state is ON, home assistant state is OFF. Home assistant would send an ON command [...]

How would one get into this state? I just tried a bunch of things, all of them without persistence in MQTT, which I would nornally never use for device state and Home Assistant picked it up just fine.
I have no clue how zigbee works, but wouldn't it be possible to always respond with the actual device's state, for example by getting the state from the bulb and publishing that? Or is that not possible?

Koenkk, 2018-10-08:
Reading is not an option as it would increase load on the network.

Note that the mocked state message is **only** send when the zigbee command has been executed successfully, I've never seen that after a successful command the bulb state was different than the command specified, which means in practice you can always trust this mocked state message.

closed 2018-12-14 by stale[bot]