Koenkk/zigbee2mqtt #454: How to reset Philips Hue without a Dimmer Switch?

orientalsniper, 2018-10-07:
I have it paired with the Philips Bridge, but I don't want to spend more money to buy the Dimmer Switch.

WinSCaP, 2018-10-07:
Well, in that case I have bad news for you, there is no real other solution. My advise would be to invest in a dimmerswitch because it's really handy with other ZigBee lights as well.

orientalsniper, 2018-10-07:
Just found this: http://thingsthataresmart.wiki/index.php?title=How_to_Reset_Philips_Hue_Bulbs

And it worked! But my zigbee2mqtt docker container is still not pairing the bulb, I'm going to have to open a new thread.

closed 2018-10-07 by orientalsniper

exetico, 2018-12-11:
@orientalsniper It's working here too, but i can't get it to connect with zigbee2mqtt. How did you manage to fix that part?

frankiej911, 2018-12-11:
I've also issues with the Philips Bulbs. No remote control, so i've tried it with the API debugging tool. Bulb is blinking 3 times and ending at full brightness. Turn off and on the bulb and it was not found by the Zigbee2mqtt and it's still posible to control the bulb with the Hue. Another touchlink true command was also working, same thing happened... But undiscoverable en still paired with the Hue.

exetico, 2018-12-11:
@frankiej911 - Same issue here. I've also tried to take the power in the middle of the blinks on different times, but it doesn't help.

My four spots is LTW013 according to Hue Essentials.

Do you really ting that we need to pick up a Philips Hue Switch to reset the buld? :-) Let me know if you make any progress!

orientalsniper, 2018-12-11:
It worked, but what I did was to eliminate the bulbs in the Hue App and zigbee2mqtt automatically pair with the bulbs, had to remove some bulbs multiple times.

orientalsniper, 2018-12-11:
> 
> 
> @orientalsniper It's working here too, but i can't get it to connect with zigbee2mqtt. How did you manage to fix that part? (My bulbs is model 9290012576 - i can't find them on the list)

If device is unsupported you have to manually add it to a file (forgot name) in the wiki.

orientalsniper, 2018-12-11:
Yeah, it was a combination of using the Philips API and removing the bulbs in the Hue app in my phone.

ygreq, 2021-06-30:
I have the same problem. I don't have a Hue Hub. I managed to reset it once and paired it with ZHA. Now I switched to Z2M and try to reset once more, but to no avail. 

Any suggestions on how to reset it with the on and off combination? Do you know how many times and how fast?
Thank you!

exetico, 2021-06-30:
Use touchlink? :-) 
https://www.zigbee2mqtt.io/information/touchlink.html