Koenkk/zigbee2mqtt #455: New version of WXKG03LM

HKC001, 2018-10-08:
Reporting a new version of WXKG03LM. Connected on both 0.1.6 and edge, but:

```
Device with modelID 'lumi.remote.b186acn01
```

The box it came in appears new as well. Previous switches came in a box with a white face and only the words 'Aqara' in black. This new one has an the word 'Aqara' in blue, an image of the switch, and Chinese + Korean text. Manufactured June 2018.

Koenkk, 2018-10-08:
Could you change https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L106 to 
```js
        zigbeeModel: ['lumi.sensor_86sw1\u0000lu', 'lumi.remote.b186acn01'],
```

and report if it works?

HKC001, 2018-10-08:
Sorry but I'm really new at this. Running Hass.IO and I couldn't find the devices.js file in any of the folders (using Samba share). Could you tell me how to access it?

Koenkk, 2018-10-08:
This should help: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices-on-Hass.io

HKC001, 2018-10-09:
Still not working unfortunately. Added to device.js and restarted container:

```
{
        zigbeeModel: ['lumi.sensor_86sw1\u0000lu', 'lumi.remote.b186acn01'],
        model: 'WXKG03LM',
        vendor: 'Xiaomi',
        description: 'Aqara single key wireless wall switch',
        supports: 'single click',
        fromZigbee: [fz.xiaomi_battery_3v, fz.WXKG03LM_click, fz.ignore_basic_change],
        toZigbee: [],
    },
```

Same error:
```
Device with modelID 'lumi.remote.b186acn01' is not supported.
```

Looking through the support new devices page now to see what else I can do, but any advice would be appreciated.

Koenkk, 2018-10-09:
Could you post your `data/database.db` file?

HKC001, 2018-10-09:
The database file in share>zigbee2mqtt, right?

```
{"id":13,"type":"EndDevice","ieeeAddr":"0x00158d000282a0f6","nwkAddr":17039,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b286acn01\u0000\u0000\u0000","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"66ev7PjOcknFMEUF"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018ed39af","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1539072762,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"HSQ4IYr2Wx0CtnEL"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d000237c63d","nwkAddr":27063,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,6,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"IEL1JBOMHcr45yRX"}
{"id":8,"type":"EndDevice","ieeeAddr":"0x00158d000252d579","nwkAddr":45413,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"JdWoH4yhoVwT8wul"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d00020f542e","nwkAddr":21575,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"Q3dBaxyn5psDpvfm"}
{"id":14,"type":"Router","ieeeAddr":"0x00158d00026a10a4","nwkAddr":56412,"manufId":4151,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.ctrl_neutral1","epList":[1,2,3,4,5,6,8],"status":"online","joinTime":1539072972,"endpoints":{"1":{"profId":260,"epId":1,"devId":6,"inClusterList":[0,1,2,3,10,25],"outClusterList":[0,10,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{"zclVersion":0,"appVersion":18,"stackVersion":2,"hwVersion":38,"manufacturerName":"LUMI","modelId":"lumi.ctrl_neutral1","dateCode":"11-11-2016","powerSource":4,"locationDesc":""}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":0,"mainsAlarmMask":0,"batteryVoltage":0}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":33,"lowTempThres":55,"highTempThres":60}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genTime":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,16],"outClusterList":[],"clusters":{"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":128}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nam [... cut ...] "genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"yN1bBdZlnEksOdjj"}
```

Koenkk, 2018-10-09:
> Could you change https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L106 to
> 
> ```js
>         zigbeeModel: ['lumi.sensor_86sw1\u0000lu', 'lumi.remote.b186acn01'],
> ```
> 
> and report if it works?

Can you change it to

```js
         zigbeeModel: ['lumi.sensor_86sw1\u0000lu', 'lumi.remote.b186acn01\u0000\u0000\u0000'],
```

Koenkk, 2018-10-10:
Can you try with:

```js
    {
        zigbeeModel: ['lumi.sensor_86sw1\u0000lu', 'lumi.remote.b186acn01\u0000\u0000\u0000'],
        model: 'WXKG03LM',
        vendor: 'Xiaomi',
        description: 'Aqara single key wireless wall switch',
        supports: 'single click',
        fromZigbee: [fz.xiaomi_battery_3v, fz.WXKG03LM_click, fz.ignore_basic_change, fz.WXKG11LM_action_click_multistate, fz.ignore_multistate_change],
        toZigbee: [],
    },
```

And check which of the following events are supported: single, double, hold/release (after holding for more than 400ms)

HKC001, 2018-10-10:
Woah, one click and zigbee2mqtt reset itself.

```
  zigbee2mqtt:info 2018-10-10 23:34:10 MQTT publish, topic: 'zig/dwd_ls_1', payload: '{"state_right":"OFF","linkquality":39}'
/app/lib/controller.js:265
                return c.cid === cid && c.type === message.type;
                         ^
TypeError: Cannot read property 'cid' of undefined
    at mappedModel.fromZigbee.filter (/app/lib/controller.js:265:26)
    at Array.filter (<anonymous>)
    at Controller.handleZigbeeMessage (/app/lib/controller.js:263:51)
    at Zigbee.handleMessage (/app/lib/zigbee.js:172:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/app/node_modules/zigbee-shepherd/lib/shepherd.js:160:14)
    at emitMany (events.js:147:13)
    at ZShepherd.emit (events.js:224:7)
    at dispatchIncomingMsg (/app/node_modules/zigbee-shepherd/lib/components/af.js:669:45)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@0.1.6 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.6 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!
     /root/.npm/_logs/2018-10-10T15_34_18_057Z-debug.log
2018-10-10T23:34:19: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2018-10-10T23:34:19: PM2 log: App [npm:0] starting in -fork mode-
2018-10-10T23:34:19: PM2 log: App [npm:0] online
> zigbee2mqtt@0.1.6 start /app
> node index.js
```

I did 'docker exec -it xxxxx bash' but couldn't find '/root/.npm/_logs/2018-10-10T15_34_18_057Z-debug.log'

Koenkk, 2018-10-11:
In the docker container, could you execute:

```js
cd /app/node_modules
rm -rf zigbee-shepherd-converters
git clone https://github.com/Koenkk/zigbee-shepherd-converters.git -b lumi.remote.b186acn01 
cd zigbee-shepherd-converters
npm install
```

and try again?

HKC001, 2018-10-14:
Sorry for the late reply, had to go out of town for work. Good news, it works! It registers single and double clicks and hold.

davidsig, 2018-10-14:
I think I might be experiencing something similar with a new wxkg02lm:

zigbee2mqtt:warn 2018-10-14 11:53:07 Device with modelID 'lumi.remote.b286acn01' is not supported.

Koenkk, 2018-10-15:
@davidsig that one is already supported: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L115 (is supported in zigbee2mqtt 0.1.7)

Koenkk, 2018-10-15:
@HKC001  thanks, merged!

closed 2018-10-15 by Koenkk