Koenkk/zigbee2mqtt #459: Add Samsung SmartThings Arrival Sensor support

pixeye33, 2018-10-08:
Hello,

I just received a Samsung SmartThings Arrival Sensor and i've tried to add support for it without success...
I've followed the "How to support new devices" page and the issue is not related to me not being able to read :)

This device is kind of special : it doen't have any input/output i can interact with (except a reset button witch sends a "devLeaving"). And that's probably the reason why i can't follow the standard procedure.

As soon as the device is first paired i've got this :
`zigbee2mqtt:debug 2018-10-8 17:38:56 Recieved zigbee message of type 'devInterview' with data '"0x24xxxxxxxxxxxxxx"'
zigbee2mqtt:debug 2018-10-8 17:38:57 Recieved zigbee message of type 'devInterview' with data '"0x24xxxxxxxxxxxxxx"'
zigbee2mqtt:debug 2018-10-8 17:39:01 Recieved zigbee message of type 'devInterview' with data '"0x24xxxxxxxxxxxxxx"'
zigbee2mqtt:debug 2018-10-8 17:39:02 Recieved zigbee message of type 'devInterview' with data '"0x24xxxxxxxxxxxxxx"'
zigbee2mqtt:debug 2018-10-8 17:39:02 Recieved zigbee message of type 'devInterview' with data '"0x24xxxxxxxxxxxxxx"'
zigbee2mqtt:debug 2018-10-8 17:39:03 Recieved zigbee message of type 'devInterview' with data '"0x24xxxxxxxxxxxxxx"'
zigbee2mqtt:debug 2018-10-8 17:39:03 Recieved zigbee message of type 'devIncoming' with data '"0x24xxxxxxxxxxxxxx"' of device 'tagv4' (0x24xxxxxxxxxxxxxx)
zigbee2mqtt:debug 2018-10-8 17:39:03 Recieved zigbee message of type 'devStatus' with data '"online"' of device 'tagv4' (0x24xxxxxxxxxxxxxx)`

and the device is paired fine : 
`zigbee2mqtt:info 2018-10-8 16:54:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x24fd5b00010a05a2"}'`

I can also remove it via the reset button :
`zigbee2mqtt:debug 2018-10-8 17:38:14 Recieved zigbee message of type 'devLeaving' with data '"0x24fd5b00010a05a2"'`

following that and the modification of devices.js to add tagv4 i've got no new message (even in debug mode)

My guess is that the devices only sends message of type devStatus (or something else) witch zigbee2mqtt chooses to ignore since the device is already paired.

I'm available to help debugging this if anyone wants to give it a try :)

pixeye33, 2018-10-15:
I discovered the existence of the networkmap feature : it keeps saying that my device is online, even though i pulled it's battery out.

I guess i'm gonna have to return this sensor to Amazon soon unless i get some help figuring this out ? :)

@Koenkk can you have a look ?

Koenkk, 2018-10-16:
@pixeye33 the sensors is probably working fine, my first guess is that it requires extra configuration to work.

Could you share your `data/database.db` file?

pixeye33, 2018-10-16:
@Koenkk i've created a clean docker instance only adding my sensor to make it easier to debug.

here is the database.db file (with my unique ID edited)
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/2484779/database.db.txt)

Device was on and did not move during the whole time (at least 5 min). For info it's supposed to send a packet every 30s.

Koenkk, 2018-10-17:
Can you try with the following added to `devices.js`.

```js
    {
        zigbeeModel: ['tagv4],
        model: 'tagv4,
        vendor: 'Samsung',
        description: 'SmartThings arrival sensor',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('genBinaryInput', 'presentValue', 10, 30, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

pixeye33, 2018-10-17:
Looks like we're making progress 👍 

exactly every 30sec (as this device is supposed to do !) i got a log saying
`No converter available for 'tagv4' with cid 'genBinaryInput', type 'attReport' and data '{"cid":"genBinaryInput","data":{"presentValue":0}}'`

i've seen no new input in database.db

Koenkk, 2018-10-19:
Could you try with: 

```bash
cd node_modules
rm -rf zigbee-shepherd-converters
git clone https://github.com/Koenkk/zigbee-shepherd-converters.git -b  F-ARR-US-2
cd zigbee-shepherd-converters
npm install
```
-> Start zigbee2mqtt

pixeye33, 2018-10-20:
btw, looks like the device sends battery info + can beep when reeving something.
https://community.smartthings.com/t/presence-activate-decativate/113706

but i'm fine with partial support instead of none at all 👍

Koenkk, 2018-10-21:
Can you update and check if the battery is reported now (a warning no known converter should be displayed now); this can take up to one hour.

Koenkk, 2018-10-23:
Let's first focus on the battery, what kind of battery does this device contain?

tb-killa, 2018-10-23:
Its a CR2032 Coin with 3V.

I use the same reporting for 4x EU-Lightyfy with 
``
xx.report('genPowerCfg', 'batteryVoltage', 1800, 3600, 0, cb),
``
because 30min MIN and 1H MAX is battery Comfortable :)

My Voltage Converter:

```
generic_power_battvolt: {
        cid: 'genPowerCfg',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
                return {voltage: msg.data.data['batteryVoltage']/10};
		},
	},
```

Koenkk, 2018-10-25:
I've made some updates, can you verify this:
- Battery should be working now
- `presence_timeout` is configurable now (same like `occupancy_timeout` https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration)
- Beep should be working now: publish to `zigbee2mqtt/[FRIENDLY_NAME]/set` payload `{"beep": 5}`. Where 5 is the seconds is should beep.

pixeye33, 2018-10-25:
Well i'm impressed : it mostly works 👍 
there is some errors left though : 

- first start of zigbee2mqtt with battery out

```
zigbee2mqtt:info 2018-10-25 22:16:34 MQTT publish, topic: 'zigbee2mqtt/0x24xxxxxxxxxxxxxx', payload: '{"presence":false,"linkquality":63}'
```

battery inserted

```
zigbee2mqtt:info 2018-10-25 22:17:03 MQTT publish, topic: 'zigbee2mqtt/0x24xxxxxxxxxxxxxx', payload: '{"presence":false,"linkquality":63,"battery":"9.00","voltage":2800}'
zigbee2mqtt:warn 2018-10-25 22:17:03 No converter available for 'STS-PRS-251' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":28}}'
zigbee2mqtt:warn 2018-10-25 22:17:03 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

notice how battery voltage is provided, but presence is still false (it does that every time battery is inserted), maybe presence should be set as true not matter what message is received ? (but it's also fine as it is)

```
zigbee2mqtt:info 2018-10-25 22:17:33 MQTT publish, topic: 'zigbee2mqtt/0x24xxxxxxxxxxxxxx', payload: '{"presence":true,"linkquality":94,"battery":"9.00","voltage":2800}'
```

- about the beeping feature, it works well !
for info it makes a series of slow beeps, not a continuous one.
a duration of 1 second is not enough to have a beep emited.

when i pushed "{"beep": 5}"

```
zigbee2mqtt:info 2018-10-25 22:31:59 Zigbee publish to '0x24xxxxxxxxxxxxxx', genIdentify - identify - {"identifytime":5} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:warn 2018-10-25 22:32:05 No converter available for 'STS-PRS-251' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":5}}'
zigbee2mqtt:warn 2018-10-25 22:32:05 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
zigbee2mqtt:info 2018-10-25 22:32:06 MQTT publish, topic: 'zigbee2mqtt/0x24xxxxxxxxxxxxxx', payload: '{"presence":true,"linkquality":81,"battery":"9.00","voltage":2800}'
zigbee2mqtt:warn 2018-10-25 22:32:06 No converter available for 'STS-PRS-251' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":4}}'
zigbee2mqtt:warn 2018-10-25 22:32:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

when i pushed "{"beep": 20}"

```
zigbee2mqtt:info 2018-10-25 22:33:13 Zigbee publish to '0x24xxxxxxxxxxxxxx', genIdentify - identify - {"identifytime":20} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:warn 2018-10-25 22:33:14 No converter available for 'STS-PRS-251' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":20}}'
zigbee2mqtt:warn 2018-10-25 22:33:14 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
zigbee2mqtt:warn 2018-10-25 22:33:15 No converter available for 'STS-PRS-251' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":19}}'
zigbee2mqtt:warn 2018-10-25 22:33:15 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

funny how device seems to respond duration asked then duration-1 (it does not countdown more than this) : you could just respond with "beeping":true and ignore this second message ? 

- about the presence_timeout (i did not set this value, as default is fine with me)

```
zigbee2mqtt:info 2018-10-25 22:39:06 MQTT publish, topic: 'zigbee2mqtt/0x24xxxxxxxxxxxxxx', payload: '{"presence":true,"linkquality":52,"battery":"9.00","voltage":2800}'
```

battery pulled out at 22:39:07

```
zigbee2mqtt:info 2018-10-25 22:40:46 MQTT publish, topic: 'zigbee2mqtt/0x24xxxxxxxxxxxxxx', payload: '{"presence":false,"linkquality":52,"battery":"9.00","voltage":2800}'
```

works perfectly ! doc of occupancy_timeout says 90 seconds as default value, presence_timeout should probably be the same

- bonus, with a brand new battery inserted :

```
zigbee2mqtt:info 2018-10-25 22:45:20 MQTT publish, topic: 'zigbee2mqtt/0x24xxxxxxxxxxxxxx', payload: '{"presence":true,"linkquality":39,"battery":"66.00","voltage":3000}'
```

3000 = 66%
2800 = 9%
seems weird ? :D 

thanks for your awesome project, and for your time to integrate this device :)

ryanbeaton, 2018-10-26:
Re: "battery inserted" results
That is how Zigbee2MQTT functions with Home Assistant.  The new values (battery) are combined with cached values to create one 'full' message. Zigbee2MQTT must have had false as the last result for the device and combined it with the battery values.

Weird how you still get the "No converter available" message though.

Re: battery percentage
It is probably using the lookup table for the Maxwell Battery - which I personally don't like ;)
https://github.com/Koenkk/zigbee-shepherd-converters/pull/77/files

tb-killa, 2018-10-26:
For the converter message just Insert ``fz.ignore_power_change`` to the ``fromzigbee`` Part.
The battery table isnt good for Devices who only provide 3v battery coin or Block who is maximum. I think this need a rework.

Koenkk, 2018-10-29:
- `WARNING: No converter.....` message will not appear now anymore
- Let's not publish `presence: true` when battery is reported (keep it to the `genIdentify` which is meant for that).
- Beeping will be reported to mqtt now
- Calibrated battery on 100% = 3000, 0% = 2500 

Can you test? If everything is OK I can merge it.

pixeye33, 2018-10-30:
Hello,
just tested your modifications (**after fixing the lint errors mentioned by Travis !**)
works great, and all problems are now fixed :)

just noticed one last thing : when i started zigbee2mqtt i got a presence true mqtt message. Battery was pulled out.
Probably is the last state known being pushed at startup, what bothers me the most is that i have no way to differentiate this message from a real one.
Is there a way to not have this pushed at boot ? (retain is already at false)

After that it should be OK to merge 👍

Koenkk, 2018-10-30:
@pixeye33 currently there is no way to disable this.

But it's hard to end up in this state right? Happens when the `presence: false` countdown is still ticking and you stop zigbee2mqtt. The same issue will currently also occur with presence sensors.

pixeye33, 2018-10-30:
You're right this should not happen that often.

presence: true should really be sent only if the device did as its kind a security device.
push your code like this i'll look if i can make a PR solving this :)

Koenkk, 2018-10-30:
merged thanks!

closed 2018-10-30 by Koenkk