Koenkk/zigbee2mqtt #460: RTCGQ11LM Xiaomi Aqara human body movement and illuminance sensor- unkown - undefined unknown

daswars, 2018-10-08:
`
2018-10-8 21:19:22 - info: Starting zigbee2mqtt version 0.1.6 (commit #c5b9176)
2018-10-8 21:19:22 - info: Coordinator firmware version: '20180815'

2018-10-8 21:19:22 - info: motion_sensor_3 (0x00158d0001de71fb): unkown - undefined unknown (EndDevice)
2018-10-8 21:19:59 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":26}}' of device 'undefined' (0x00158d0001de71fb)
2018-10-8 21:19:59 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'undefined' (0x00158d0001de71fb)
2018-10-8 21:19:59 - debug: Recieved zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":26}}' of device 'undefined' (0x00158d0001de71fb)
2018-10-8 21:19:59 - debug: Recieved zigbee message of type 'devChange' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'undefined' (0x00158d0001de71fb)
2018-10-8 21:21:00 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":58}}' of device 'undefined' (0x00158d0001de71fb)
2018-10-8 21:21:00 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'undefined' (0x00158d0001de71fb)
2018-10-8 21:21:00 - debug: Recieved zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":58}}' of device 'undefined' (0x00158d0001de71fb)
2018-10-8 21:22:03 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":67}}' of device 'undefined' (0x00158d0001de71fb)
`

Koenkk, 2018-10-09:
This happens when the pairing has not fully been completed. Can you try repairing?

daswars, 2018-10-09:
You're absolutely right.
Thank you very much for the quick help.

closed 2018-10-09 by daswars