Koenkk/zigbee2mqtt #461: IKEA TRADFRI wireless dimmer (ICTC-G-1) - no MQTT after nigh of inactivity

MiloshCZ, 2018-10-09:
Hi,
after longer time of inactivity of dimmer (about 12 hours) there is no mqtt message published when I move with sensor.

Log:
```
  serialport:poller received "readable" +2s
  serialport:bindings read +2s
  serialport:unixRead Starting read +2s
  serialport:unixRead Finished read 30 bytes +1ms
  serialport:main binding.read finished +2s
  cc-znp { sof: 254,
  cc-znp   len: 25,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 8,
  cc-znp      srcaddr: 33153,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 42,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 6950846,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 5,
  cc-znp      data: <Buffer 11 60 01 01 c3> },
  cc-znp   fcs: 143,
  cc-znp   csum: 143 } +2s
  serialport:main _read reading +15ms
  serialport:bindings read +18ms
  serialport:unixRead Starting read +17ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 8, srcaddr: 33153, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 42, securityuse: 0, timestamp: 6950846, transseqnumber: 0, len: 5, data: <Buffer 11 60 01 01 c3> } +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +9ms
  serialport:unixRead waiting for readable because of code: EAGAIN +14ms
  serialport:poller Polling for "readable" +34ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +3ms
  serialport:poller received "readable" +296ms
  serialport:bindings read +313ms
  serialport:unixRead Starting read +299ms
  serialport:unixRead Finished read 28 bytes +1ms
  serialport:main binding.read finished +317ms
  cc-znp { sof: 254,
  cc-znp   len: 23,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 8,
  cc-znp      srcaddr: 33153,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 49,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 5880613,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 3,
  cc-znp      data: <Buffer 11 57 07> },
  cc-znp   fcs: 115,
  cc-znp   csum: 115 } +308ms
  serialport:main _read reading +16ms
  serialport:bindings read +19ms
  serialport:unixRead Starting read +18ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 8, srcaddr: 33153, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 49, securityuse: 0, timestamp: 5880613, transseqnumber: 0, len: 3, data: <Buffer 11 57 07> } +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +6ms
  serialport:unixRead waiting for readable because of code: EAGAIN +11ms
  serialport:poller Polling for "readable" +33ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +3ms

```

MiloshCZ, 2018-10-09:
After restart zigbee2mqtt software, everything is OK, but only to next long inactivity time.

Koenkk, 2018-10-15:
I've connected one to my network, will let you know.

Koenkk, 2018-10-17:
For me it's still working after 2 days.

Could you publish your zigbee network map: https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttbridgenetworkmap and did you set `permit_join: false`?

MiloshCZ, 2018-10-18:
With version 0.1.7 is everything OK. I have upgraded yesterday.

Koenkk, 2018-10-18:
OK, can this be closed?

MiloshCZ, 2018-10-18:
Yes, you can close it

closed 2018-10-18 by Koenkk