Koenkk/zigbee2mqtt #462: Do Hue bulbs serve as routers?

orientalsniper, 2018-10-09:
I can't pair newer devices farther away from the CC2531 USB sniffer. I thought all plugged devices function as routers too.

tb-killa, 2018-10-09:
Hi.
Zigbee is told to to support a "Mesh Networking" topology.
There are three sorts of Zigbee radios; Coordinator, Router and End Device.

Coordinator is a hub or bridge, it's where the devices register.

Routers are nodes in the mesh network. They can communicate back to the coordinator and also onwards to other nodes.

End Devices can only talk to a parent device (a router or coordinator), they cannot relay data.

If you in touch with Bulbs then normaly they are routers too (if powered constantly).
But i see that sorts if them working totally different like i first thought they has to be work 
because every producer of them could build own implementation of zigbee.

From your question.. we are working on joining mode by direct point to router adr. but this need some more investigation first by @Koenkk .

orientalsniper, 2018-10-09:
Thanks for replying.

Yea, I have read about the mesh topology in the wiki.

So what you are saying is we can't use Hue bulbs as routers? Because it's going to take me a while to receive extra CC2531 to flash as routers.

ben423423n32j14e, 2018-10-20:
My Hue Lightstrip V2's act as routers (just saying)...

orientalsniper, 2018-10-20:
How do you check if your hue is routing?

ben423423n32j14e, 2018-10-20:
I'm using Zigbee2MQTT Admin Panel for Node-Red

[image]

https://github.com/ben423423n32j14e/zigbee2mqttadminpanel

orientalsniper, 2018-10-20:
I see, I'm gonna have to find a way to install Node-Red in the docker
version of zigbee2mqtt.....

ben423423n32j14e, 2018-10-20:
You don't need to install it in docker, it connects to your MQTT server.

If you install Node-Red anywhere on your PC, import the flow then just configure it to connect to the MQTT server IP.

orientalsniper, 2018-10-20:
Thank you! I'm going to try tonight and see if my bulbs are routing or not.

orientalsniper, 2018-10-21:
[image]

I don't understand, they are running as routers, yet I can't add the Xiaomi sensors in the living room.

ben423423n32j14e, 2018-10-21:
End Devices can't yet join via routers (at all in Zigbee2MQTT), thus they must be in-range of the coordinator to join. This doesn't mean your other devices are not acting as routers, just joining via a router is not currently supported.

closed 2018-12-27 by stale[bot]