Koenkk/zigbee2mqtt #465: Extending Zigbee Range with CC2530 + CC2591

HKC001, 2018-10-09:
Following on from: https://github.com/Koenkk/zigbee2mqtt/issues/52, I was wondering if anyone can provide more specific links detailing the cc2530 + cc2591 setup.

1.  For use with zigbee2mqtt, should I be using the firmware here:http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/, or here: https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices

2. How do you maintain power supply and connection to a Raspberry Pi 3 B+? Is there a shield available, or do we use jumper cables? Is there a link to a product or jumper cabling guide to help achieve this connection?

3. For those who have used the devices here (https://github.com/Koenkk/zigbee2mqtt/issues/52#issuecomment-390380532), how do you normally place them at home? I have 2 CC2531s I plan to use as routers, and the cc2530 + cc2591 as a coordinator on my Pi. I understand that the cc2530 + cc2591 is more sensitive and can 'hear' better, so the CC2531's effective range is increased. But how big is the range increase? Is it only 2 - 3 meters, or more?

lolorc, 2018-10-10:
1) https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591/bin/CC2530_CC2591ZNP-Prod.hex

2) mine is attach to a usb2serial converter, with a long usb cable
    ever looked at https://github.com/Koenkk/zigbee2mqtt/issues/291#issuecomment-424116534

3) my cc2531-cc2590 is in my living room, downstairs, It attach to wall, 2meter high.
    at the moment I have 3 routers downstairs, quite difficult to see how increased is the range.
[image]

I've just replaced the antenna by a 9db wifi one.

don't know about CC2630, i've been looking at cc2538+cc2592 but there's no support for either of them yet

GoSpursGoNL, 2018-10-14:
Hi @lolorc , I see that the firmware you propose if not the one mentioned on the wiki. What's the advantage of the one that you recommend?

Which usb2serial converter are you using?

lolorc, 2018-10-14:
I think it's just matter of updating the wiki :)
this one removes the need for bridging P2_0, P0_6 and P0_4, it doesn't require hardware control.
and if I recall correctly it also has increased buffers.

About powering the cc2530 from the rpi, as far as I remember the rpi is not able to provide a lot of current on its 3.3V output. My usb2serial converter is a FT232BM clone, I power the cc2530 from it with its 5V output, and the serial levels are ttl ones, 3.3V.
So I would connect my cc2530+cc2590 to the 5V output of the rpi. I can't check right now, but my board might have a 3.3V reg because the cc2530 is not supposed to be 5V tolerant.

GoSpursGoNL, 2018-10-18:
Flashed successfully and it's working correctly in combination with the 3.3v from the Pi, thanks.

GoSpursGoNL, 2018-10-21:
The range is still not as good as the Xiaomi Gateway though.

HKC001, 2018-10-25:
This is failing for me, can someone give some advice? My error:

```
  zigbee2mqtt:info 2018-10-25 14:58:05 Starting zigbee2mqtt version 0.1.8 (commit #unknown)
  zigbee2mqtt:info 2018-10-25 14:58:05 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-10-25 14:58:06 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2018-10-25 14:59:06 Starting zigbee-shepherd
  zigbee2mqtt:error 2018-10-25 14:59:06 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-10-25 14:59:06 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-10-25 14:59:06 Failed to start
	{"message":"Error: No such file or directory, cannot open /dev/ttyACM0","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0"
```

My steps:

1. Got [this cc2530+cc2591](https://item.taobao.com/item.htm?spm=a230r.1.14.67.41757d3cwKXgGK&id=19051924425&ns=1&abbucket=19#detail), which looks exactly like the one in https://github.com/Koenkk/zigbee2mqtt/issues/52.

2. I'm using Hass.IO, so I tried disabling BT and switching the UART by adding these lines in config.txt:

```
dtoverlay=pi3-disable-bt
dtoverlay=pi3-miniuart-bt
```

3. Flashed CC2530_CC2591ZNP-Prod.hex from the link above, filesize 681 KB. Using those single pin wires (UART?) connected cc2530+cc2591 to CC Debugger:

```
P2_2 > DC
P2_1 > DD
RST  > RESETn
GND > GND
VCC > 3.3V from Debugger
```

4. Flashed without issues. Then using the same UART wires, connected cc2530+cc2591 to Pi 3B+ (numbered according to [this pinout](https://pinout.xyz/pinout/uart#)

```
P0_2 (RX) > BCM 14 (TX, pin 8)
P0_3 (TX) > BCM 14 (RX, pin 10) 
GND > GND (pin 9)
VCC > 3.3V (pin 1)
```

ptvoinfo, 2018-10-25:
@HKC001 Execute `ls /dev/ttyS*` in the console.
The port name should look like /dev/ttyS1

HKC001, 2018-10-25:
I tried plugging in 2 devices at once, a working cc2531 coordinator and a cc2530 + cc2591 + ptvoinfo's firmware, and now 2 serial connections are showing, but none of them are ttyS* ports.

```
    "data": {
        "serial": [
            "/dev/ttyACM0",
            "/dev/ttyAMA0"
        ],
```

hoangtuit, 2018-10-26:
Hi,
I can't understand the range off CC2531 coordinator. 
My CC2351 in my office room, upstairs. My senors (Xiaomi Aqara temperature sensor WSDCGQ11LM and Xiaomi Aqara door sensor MCCGQ11LM) are in the next room, about 3-4M from the coordinator with 2 walls between. I found that the CC2531 coordinator is failing to receive updates. 
BUT i have another Xiaomi Aqara door sensor, it's in my living room, downstairs, about 8M from the coordinator. The CC2531 coordinator is alway receive updates from it.

So,  I plan to use CC2531 as routers, and the cc2530 + cc2591 as a coordinator. Does this have a better?

ptvoinfo, 2018-10-26:
CC2530 does not have a USB interface. Therefore, you need to connect it to Rx, Tx pins in your SBC (RPi for example). Did you enable the built-in UART? Or do you use an additional UART to USB converter?

HKC001, 2018-10-26:
Ok, got a little further but still no success so far. But this is what I tried:

1. Using this [USB to TTL/232](https://detail.tmall.com/item.htm?spm=a230r.1.14.103.23315e53TwlMPd&id=41455190692&ns=1&abbucket=19).
2. In zigbee2mqtt config, set ```"serial_port": "/dev/ttyUSB0"```
3. Upon starting up, I get a new error:

```
zigbee2mqtt:error 2018-10-26 12:52:36 Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-0.1.8/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/zigbee2mqtt-0.1.8/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```

HKC001, 2018-11-05:
I FINALLY got the my [CC2530 + RFX2401](https://user-images.githubusercontent.com/8360230/40265181-3771ef0e-5b3c-11e8-8fe0-a59375783e41.jpg) (the device on the far right) running with a Pi 3B+ with 25 devices joined. 

From what I gather, setting it up is similar to the CC2530 + CC2591. Here's what I did, if this info helps you, please post here so others know:

1. Soldered according to [this image](https://user-images.githubusercontent.com/8360230/40383446-9aaac210-5e09-11e8-9097-ea2cad19e8cb.jpg) from Kirovilya.

2. Flashed Koenkk's [optimised firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/optimize/coordinator/CC2530_CC2591), ``coordinator version 20181030``.

3. Did NOT add these lines into the Pi config.:
```
dtoverlay=pi3-disable-bt
dtoverlay=pi3-miniuart-bt
```
4. Using zigbee2mqtt-edge, ``zigbee2mqtt version 0.1.8 (commit #2e736bf)``, set ``"rtscts": false`` on the config page.

ryanbeaton, 2018-11-05:
Thanks @HKC001 for that info.

So the firmware for CC2530+CC2591 works with the CC2530+RFX2401?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

HKC001, 2018-11-11:
@ryanbeaton Yes, but I found [issues ](https://github.com/Koenkk/zigbee2mqtt/issues/533#issuecomment-436804651) after running the firmware for a few days.

@thibaultmol I've only tried the CC2531 and CC2530 + RFX2401. If you can solder (or are willing to try, its not that hard), I really suggest you go for the CC2530 + RFX2401. For me, the CC2531's range is too short and keeps dropping devices from the network, causing automations to break every so often. Having unreliable automations is far worse than no automations.

AnhDuc85, 2018-11-27:
anybody tried this version of [CC2530+CC2591](https://www.aliexpress.com/item/CC2530-CC2591-ZigBee-Module-ZA2530-2591-With-PA-Remote/32611383914.html)? It's cheaper than the WeBee modul

lolorc, 2018-11-28:
@AnhDuc85 yes, I'm using it. works well, I'd say it's better than the previous version.

@pixeldoc2000 door sensors don't act as router/repeater.

AnhDuc85, 2018-11-28:
@lolorc what antenna are you using for that module?

lolorc, 2018-11-28:
@AnhDuc85 the one provided with the module. I tried with a huge 9db wifi one, but it wasn't a good idea.

mcsSolutions, 2018-12-20:
I have been reading several threads on alternate interfaces and in particular have interest in the RFX2401 variant.  While I run on Linux, I do my flashing on Windows.  I have used the CC Debugger for flashing the 2531.  I have also used PlatformIO environment and esptool to flash ESP8266 boards.

Looking at the schematics it looks as if RX and TX from USB/serial are routed to the 2530, but would expect some use of RESET to distinguish from normal power up vs. flashing modes.  I have not seen any discussion related to this.

I have seen discussion of wiring 3 pins to ground, but his one offers alternative that does not require it.

My question in particular is the process to flash the RFX2401 variant of this interface.  In particular from a Windows or as a second choice Debian/RPi environment.

Ton1965, 2018-12-20:
Hi @mcsSolutions ,
I am using an RFX2401 stick for about half a year already and it is running great. Essentially it covers all of my 100sq.m apartment where some of the walls are of ferro-concrete.

I used the [Alternative flashing methods](https://koenkk.github.io/zigbee2mqtt/information/alternative_flashing_methods.html) procedure to flash it. Flashing was done under Windows. The drawback is the procedure takes much time (~3hours). I also wired the 3 pins to ground. 

I am in no way an expert in microelectronics so I cannot answer your question about RESET usage. I will be very glad if you find a way of flashing it without soldering since I would like to have a spare one for just in case.

BTW, don't forget to set rtscts to false in zigbee2mqtt configuration if using this stick. It does not work otherwise.

mcsSolutions, 2018-12-21:
The alternate methods link looks to me as if the flashing is actually performed via Arduino micro or ESP8266 micro.  It describes the 10 pin connector that is on the 2531 but is not on my interface that contains RFX2401 so these look to be instructions for the 2531.

The description indicates that "CC_RST" is associated with 5.  The schematic I looked at from the supplier on Alieexpress did not have any programmable pin associated with Reset other than access to the 4 connector pads so I would not expect to be able to program without connection to this pad or with some manual intervention.

How is the Adruino/ESP8266 is connected to the RFX2401 board?  My ESP do not have USB connections.   
 Do I need to buy a specific Arduino product to host the firmware for CCLi?  My zigbee board does not  have breakout for RX/TX but access is via USB / Serial contained on the circuit card normally used for RFX2401 communications.  How is the binary file that is going to be flashed accessed by Arduino/ESP8266?

What specific steps were done to program the RFX2401 product?

Ton1965, 2018-12-21:
There is another link inside the documentation I sent you: [here](https://github.com/kirovilya/CCLib#1-prepare-your-arduino-board). It shows all the necessary connections. Essentially just three pins and ground need to be connected. That link shows picture with some registers but they can be skipped. I did not use them and everything worked ok. 
You need to add CCLib library to your Arduino IDE and flash CC_proxy project to your Arduino. Then you use a Python module to flash the firmware.
That's it.

Koenkk, 2018-12-31:
@danpowell88 thats also possible however that makes the solution unecessary big.

@hoangtuit it depends, in my case the coordinator is not in an optimal place (front of my house). Therefore I've put a CC2530 router in the center of my house. Having the CC2531 as a coordinator is also a easier solution because of the USB port. Thats why I personally prefer solution A.

enomam, 2019-01-29:
I'm about to build one of these, based on the instructions in https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_create_a_cc2530_router.md, but am concerned about the lack of a fuse. Is there one built into the transformer?

GerSant, 2019-03-13:
Someone has the wiring diagram to flash the CC2530 using the Arduino's alternate flashing method?

Thanks in advance!

yozik04, 2019-03-18:
Anybody who has cc2530 + cc2591 please can you check if both antenna pins are shorted to ground for you? I have bought from Ali with GB2530 module in a plastic case and both antenna pins are shorted to the ground.
Oscilloscope of course shows nothing on antenna pins.
i think my unit is defective.

https://www.aliexpress.com/item/RF-TO-USB-CC2530-CC2591-RF-switch-USB-transparent-serial-data-transmission-equipment/1996354384.html?spm=a2g0s.12269583.0.0.42513681WOdOnA

[image]

If I disconnect the antenna header and test on the board it is shorted there.

The unit itself seems to work OK'sh. After penetrating couple stone walls (10m) link quality is around 40. But as I understand I get no benefit from the antenna and RF amplifier in my unit.

enomam, 2019-03-18:
> I'm about to build one of these, based on the instructions in https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_create_a_cc2530_router.md, but am concerned about the lack of a fuse. Is there one built into the transformer?

Current using a USB -> 3.3V adapter 👍

mcsSolutions, 2019-03-24:
@yozik04 
I have the same unit obtained early this year.  If I use ohm meter to measure the center antenna pin and the outside pin it shows a short.  The same if I measure the center pin of the connector mounted on the module.  Without the antenna wire connected the RF signal level is extremely weak with link quality between 1 and 2 over a 10 ft distance.  If I connect the antenna wire then link quality goes up to around 50.  My guess is that the unit is not defective, but active logic enables a non-grounded path to the antenna.

yozik04, 2019-03-25:
@mcsSolutions Thank you. I have talked to a guy more advanced in electronics. He told that it is OK that it shows short. Also oscilloscope won't show any fluctuations as it cannot measure on GHz rate. I asked this because one guy on ali has put in a feedback that he had a short somewhere. I had to check my unit as well =)
With antenna my unit covers whole house. Even diagonally and different floors. It manages to go through 40cm concrete ceiling and around 10m(33ft) in distance through couple walls from light concrete, still shows signal around 30. Close to the unit it measures around 170. I think the link rate also depends on the second device. I currently have only Xiaomi curtain motors.

closed 2019-05-31 by stale[bot]

halfcherry, 2019-08-11:
Will this work?
https://item.taobao.com/item.htm?spm=a1z10.5-c-s.w4002-16785486380.12.c60918aaYpf6dL&id=583299208437