Koenkk/zigbee2mqtt #466: Very small range on CC2530 with RFX2401

sebastiandinu, 2018-10-11:
I flashed a new cc2530 with RXF2401 ( [E18-mdopao v1.1](https://www.aliexpress.com/item/2pcs-lot-Free-shipping-Long-distance-CC2530F256-PA-power-amplifier-2-4G-wireless-module-Internet-of/32632800271.html) )with the latest firmware and have the latest zigbee2mqtt on a hassio rpi2. I connected a aqara temperature/humidity/pressure sensor after 10 minutes of trying but the range is super small. If I have the aqara sensor like next to it it says link quality at about 40, if I move it more than 1.5-2 meters I do not get updates from it even if I press the button on it, and if the sensor is like 50 cm away link quality goes down to 0-10. It looks like the rf amp is somehow not working correctly.  Is there any other firmware I can test ?
The cc2530 has the standard rx/tx/3v3/gnd + p2.0 and p0.4 and p0.6 to gnd.

fred-c1, 2018-10-19:
Looking at the spec of your CC2530 module (E18-MD0PA0_Datasheet_CN_v4.5.pdf), it seem that your module is not using the "standard" interface (=CC2530 GPIOs) to control the power amplifier (RXF2401 in your case).

closed 2018-12-25 by stale[bot]

Nonsense1112, 2019-02-23:
On E18-MD0PA0 the RFX2401 pins are not connected to CC2530, so is not using the RF chip on default.
You must connect pin 23 (TXEN) to pin 12 (P1.1) and pin 24 (RXEN) to pin 9 (P1.4).