Koenkk/zigbee2mqtt #467: Hue motion sensors create periodic occupancy event without actual motion

simonvanderveldt, 2018-10-11:
zigbee2mqtt v 0.1.6.

I'm using several Hue motion sensors, all of them exhibit the same behavior of triggering/publishing an occupancy (motion) event every 10 minutes. For clarity, this happens when there is no actual motion, I've put one of them in a box to test/be sure nothing would trigger it.

Any idea what could be causing this? It makes them effectively useless for motion sensing purpose obviously :)
Would a debug log of this be useful?

```
# Me putting the sensor in a box
zigbee2mqtt:info 2018-10-11 08:35:29 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":28,"occupancy":true,"temperature":18.37,"illuminance":837}'
zigbee2mqtt:info 2018-10-11 08:35:34 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":55,"occupancy":true,"temperature":18.37,"illuminance":423}'
zigbee2mqtt:info 2018-10-11 08:35:39 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":57,"occupancy":true,"temperature":18.37,"illuminance":0}'
zigbee2mqtt:info 2018-10-11 08:35:44 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":57,"occupancy":true,"temperature":18.79,"illuminance":0}'
# Box closed, periodic occupancy reports start here
zigbee2mqtt:info 2018-10-11 08:36:00 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":55,"occupancy":true,"temperature":18.79,"illuminance":0}'
zigbee2mqtt:info 2018-10-11 08:36:02 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":60,"occupancy":true,"temperature":18.79,"illuminance":0}'
zigbee2mqtt:info 2018-10-11 08:46:00 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":60,"occupancy":true,"temperature":19.22,"illuminance":0}'
zigbee2mqtt:info 2018-10-11 08:56:01 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":57,"occupancy":true,"temperature":18.37,"illuminance":0}'
zigbee2mqtt:info 2018-10-11 08:56:03 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":60,"occupancy":true,"temperature":18.37,"illuminance":0}'
```

P.S. Note some/almost all messages are also duplicated, probably because of #360 ?

Koenkk, 2018-10-11:
Thanks, this indeed is a bug in the `generic_occupancy` zigbee-shepherd-converter.

Could you provide an example where the occupancy should be triggered?

simonvanderveldt, 2018-10-11:
This should be from a "clean" trigger of the motion sensor
```
zigbee2mqtt:debug 2018-10-11 17:57:41 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:info 2018-10-11 17:57:41 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":55,"occupancy":true,"temperature":19.64,"illuminance":0}'
zigbee2mqtt:debug 2018-10-11 17:57:41 Recieved zigbee message of type 'devChange' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-11 17:57:43 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:info 2018-10-11 17:57:43 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"battery":100,"linkquality":55,"occupancy":true,"temperature":19.64,"illuminance":0}'
```

tb-killa, 2018-10-11:
I see this on my implementation of the bitron device too.
Its because we trigger by every ``attReport`` .. we need to check if specific value reported and then do occupancy.

tb-killa, 2018-10-11:
From checking the old post who i´m help by implementing this type of device i found the original attribute:
https://github.com/Koenkk/zigbee2mqtt/issues/310#issuecomment-416703966

@Koenkk 
```
zigbee2mqtt:warn 2018-8-28 18:45:45 No converter available for '9290012607' with cid 'msOccupancySensing', type 'attReport' and data '{"cid":"msOccupancySensing","data":{"occupancy":0}}'
```
we have to check if ``occupancy`` return 1 and then to the timer initialisation.

orientalsniper, 2018-10-15:
Hello, I've got the same issue, my mom's said hallway light (with Hue motion sensor) turns on by itself and I didn't believe her until I checked logbook.

Koenkk, 2018-10-15:
Can you check if this has been fixed in the latest dev branch?

simonvanderveldt, 2018-10-15:
@Koenkk I'm running it now to check. Will report back after it has ran for some time.

I did notice two things which I wanted to check, not sure if it's correct behavior or not:
- Occupancy is being published when a zigbee message without occupancy is received
```
zigbee2mqtt:debug 2018-10-15 19:04:31 Recieved zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:info 2018-10-15 19:04:31 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"occupancy":true,"linkquality":18,"illuminance":0}'
```
- It seems like occupancy is being set to `false` after quiet a long delay (more than 60s in this example)
```
zigbee2mqtt:debug 2018-10-15 19:04:35 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:04:35 Recieved zigbee message of type 'devChange' with data '{"cid":"msOccupancySensing","data":{"occupancy":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:04:37 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:info 2018-10-15 19:05:40 MQTT publish, topic: 'zigbee2mqtt/<deviceid>', payload: '{"illuminance":0,"linkquality":7,"occupancy":false}'
```

Koenkk, 2018-10-15:
- Yes, the whole state is send (this is needed for homeassistant to work properly). If you want to disable this use `cache_state` (https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration)
- After 90 seconds an `occupancy: false` event is generated, this can be configured, see https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration

simonvanderveldt, 2018-10-15:
> * Yes, the whole state is send (this is needed for homeassistant to work properly). If you want to disable this use `cache_state` (https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration)
> * After [...]

Thanks for the info/links. I should've read the wiki better.
Is `occupancy_timeout` meant for devices that don't send a `occupancy: 0` message?

Regarding the dev branch/fix: It seems to be working fine, ran it for over an hour and while there were occupancy messages from the device, no occupancy messages were published :) So from my testing this issue seems to be fixed. Thanks for the quick solution! :+1: 

For reference an excerpt from log from the last hour with zigbee occupancy events that happened, notice the lack of MQTT events:
```
zigbee2mqtt:debug 2018-10-15 19:19:19 Recieved zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2757}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:19:19 Recieved zigbee message of type 'devChange' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2757}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:20:48 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:20:49 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:21:06 Recieved zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":200}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:29:18 Recieved zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2757}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:29:48 Recieved zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2771}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:29:48 Recieved zigbee message of type 'devChange' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2771}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:30:14 Recieved zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:30:48 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:30:50 Recieved zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":0}}' of device 'SML001' (<deviceid>)
zigbee2mqtt:debug 2018-10-15 19:34:17 Recieved zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2785}}' of device 'SML001' (<deviceid>)
```

Koenkk, 2018-10-16:
`occupancy_timeout` is used for all occupancy sensors. 

If set to 0 no `occupancy: false` will be send.

If set to e.g. 10 an `occupancy: false` will be send after no `occupancy: true` has been send for 10 seconds.

closed 2018-10-16 by Koenkk

allofmex, 2018-12-31:
This fixed solved the problem only partially and is failing in case of continuous motion.
See pull request [https://github.com/Koenkk/zigbee-shepherd-converters/pull/184](https://github.com/Koenkk/zigbee-shepherd-converters/pull/184)