Koenkk/zigbee2mqtt #468: Not showing on HomeAssistant after updating

deten, 2018-10-11:
Hi,

I have some Xiaomi Aqara Temp/Humidity/Pressure sensors I use in my house.  Everything was working fine until I updated my Zigbee2mqtt and now its not working.

I am running Hassbian, followed instructions and everything worked fine.

Update was following the instructions in the wiki, here, item 6. https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge

Zigbmee2mqtt Config:
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM1
devices:
  '0x00158d0002437c56':
    friendly_name: 'Stat 1'
    retain: false
  '0x00158d00023208ae':
    friendly_name: 'Stat 2'
    retain: false
```

HomeAssistant Config:
```
mqtt:
  discovery: true
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'
```

HA Log:
`2018-10-11 15:03:44 WARNING (MainThread) [hbmqtt.mqtt.protocol.handler] BrokerProtocolHandler Unhandled exception in reader coro: IncompleteReadError('0 bytes read on a total of 1 expected bytes',)`

systemctl status zigbee2mqtt.service
```
pi@hassbian:/opt/zigbee2mqtt $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-10-11 14:08:20 UTC; 53min ago
 Main PID: 500 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─500 npm
           ├─539 sh -c node index.js
           └─540 node index.js

Oct 11 14:09:44 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:09:44 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.4,"linkquality":23,"humidity":58.28,"pressu
Oct 11 14:17:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:17:20 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.28,"linkquality":34,"humidity":57.41,"press
Oct 11 14:17:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:17:20 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.25,"linkquality":34,"humidity":57.37,"press
Oct 11 14:40:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:40:20 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.3,"linkquality":36,"humidity":58.28,"pressu
Oct 11 14:40:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:40:20 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.3,"linkquality":34,"humidity":58.15,"pressu
Oct 11 14:51:46 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:51:46 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.11,"linkquality":23,"humidity":57.37,"press
Oct 11 14:51:46 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:51:46 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.11,"linkquality":26,"humidity":56.76,"press
Oct 11 14:51:46 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:51:46 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.11,"linkquality":23,"humidity":56.76,"press
Oct 11 14:56:24 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:56:24 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.3,"linkquality":39,"humidity":58.15,"pressu
Oct 11 14:56:24 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:56:24 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.13,"linkquality":39,"humidity":58.05,"press
lines 1-19/19 (END)
```

sudo journalctl -u zigbee2mqtt.service -f
```
pi@hassbian:/opt/zigbee2mqtt $ sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at Thu 2016-11-03 17:16:42 UTC. --
Oct 11 14:09:44 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:09:44 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.4,"linkquality":23,"humidity":58.28,"pressure":1008,"battery":"98.33","voltage":2995}'
Oct 11 14:17:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:17:20 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.28,"linkquality":34,"humidity":57.41,"pressure":1008,"battery":"100.00","voltage":3015}'
Oct 11 14:17:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:17:20 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.25,"linkquality":34,"humidity":57.37,"pressure":1008.53,"battery":"100.00","voltage":3015}'
Oct 11 14:40:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:40:20 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.3,"linkquality":36,"humidity":58.28,"pressure":1008,"battery":"98.33","voltage":2995}'
Oct 11 14:40:20 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:40:20 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.3,"linkquality":34,"humidity":58.15,"pressure":1008,"battery":"98.33","voltage":2995}'
Oct 11 14:51:46 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:51:46 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.11,"linkquality":23,"humidity":57.37,"pressure":1008.53,"battery":"100.00","voltage":3015}'
Oct 11 14:51:46 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:51:46 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.11,"linkquality":26,"humidity":56.76,"pressure":1008.53,"battery":"100.00","voltage":3015}'
Oct 11 14:51:46 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:51:46 MQTT publish, topic: 'zigbee2mqtt/Stat 2', payload: '{"temperature":23.11,"linkquality":23,"humidity":56.76,"pressure":1008,"battery":"100.00","voltage":3015}'
Oct 11 14:56:24 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:56:24 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.3,"linkquality":39,"humidity":58.15,"pressure":1008,"battery":"98.33","voltage":2995}'
Oct 11 14:56:24 hassbian npm[500]:   zigbee2mqtt:info 2018-10-11 14:56:24 MQTT publish, topic: 'zigbee2mqtt/Stat 1', payload: '{"temperature":23.13,"linkquality":39,"humidity":58.05,"pressure":1008.34,"battery":"98.33","voltage":2995}'

```

Koenkk, 2018-10-12:
Whats the problem? Are the values not shown in home assistant? (as I see they are being published by zigbee2mqtt).

faronov, 2018-10-12:
@deten what version HA you use?
I also upgrade HA to 0.79.3 version and any ZigBee sensors is gone
something broke around mqtt in this HA version

SalkinDK, 2018-10-12:
> @deten what version HA you use?
> I also upgrade HA to 0.79.3 version and any ZigBee sensors is gone
> something broke around mqtt in this HA version

i had the exact same problem after updating to 0.79.3. i had some lights but the most of my lights and sensors disapearred.
I ended up setting the homeassistant setting to false, and then adding all the lights and sensor manuel instead of using the discovery setting. I now have no problem at all.
So i think it is a bug in the discovery module.

deten, 2018-10-13:
@faronov I had that version, today I updated and its still not working...

deten, 2018-10-13:
@SalkinDK 

Can you elaborate a bit? What did setting homeassistant to false do?

SalkinDK, 2018-10-13:
> @SalkinDK
> 
> Can you elaborate a bit? What did setting homeassistant to false do?

Setting the homeassistant setting to false, disables the discovery way of doing things.
So instead of ziggbee2mqtt sending the config mqtt payload, and the sensors automatically appearing in homeassistant. You have to add and define the sensors yourself. Read more here, below the "**In case you dont want to use Home Assistant MQTT discovery you can use the configuration below**.":
https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant

deten, 2018-10-14:
I installed Mosquitto and this fixed one of my issues.  Other issue is still there and I openend a new issue here:

https://github.com/Koenkk/zigbee2mqtt/issues/474

faronov, 2018-10-16:
> Setting the homeassistant setting to false, disables the discovery way of doing things.
> So instead of ziggbee2mqtt sending the config mqtt payload, and the sensors automatically appearing in [...]

I tried to add manually, but this not helped me
I remove hassio, install plain distribution , and downgrade HA to 0.70.3 version
it back to work

Bug seems around MQTT server implementation, that switched to something new
now version HA 0.79.3 just don't see internal mqtt server

faronov, 2018-10-16:
but this bug for HA , not for zigbe2mqtt

tweaker, 2018-10-17:
0.1.7 publishes json not always with temperature and humidity noticed.
I have the same problem.
How to fix the HA yet I do not know. Rolled back to 0.1.3

Koenkk, 2018-10-18:
@tweaker this issue has been fixed in https://github.com/Koenkk/zigbee2mqtt/issues/485 which is available in the dev branch.

closed 2018-12-24 by stale[bot]