Koenkk/zigbee2mqtt #469: Switch listed in HA as sensor (should be binary_sensor)

avolmensky, 2018-10-12:
I have a Xiaomi WXKG01LM Smart Switch, when I pair it to zigbee2mqtt it comes up in Home Assistant as a sensor, while the door switch and motion sensors appear as binary_sensor. The value of that sensor is whatever the last press of that button was (eg, single, double)

The problem I have is whenever I make a change in the home assistant web interface (eg, renaming a sensors name/entity_id) HA fires out the value of the sensor and will trigger an automation to just say toggle a light.

Is there any way to get this device to act like a binary_sensor? This button is a little different than just say a door sensor as you can double tap it. @ciotlosm is this the problem that https://github.com/home-assistant/home-assistant/pull/14957 is trying to fix?

mario-g0, 2018-10-18:
I'm also losing my mind with this same switch... being detected as a sensor, I can't make any automation work...

WagnerBeccard, 2018-10-18:
Use the message as a trigger 

  - alias: single_p
    trigger:
      platform: mqtt
      topic: 'zigbee2mqtt/xiaomibutton_p'
    condition:
      condition: template
      value_template: "{{ 'single' == trigger.payload_json.click }}"
    action:
    - service: scene.turn_on
      entity_id: scene.vr_hell

avolmensky, 2018-10-19:
@WagnerBeccard The problem with using the message as a trigger, is when you make certain changes in Home Assistant (eg Changing the entity_id/name on any other sensor/binary_sensor from the web interface) home assistant will transmit that same message again (even though the button wasn't pushed) and trigger the action again. I get a lot of false button presses when using the button as a sensor

Koenkk, 2018-10-19:
The homeassistant sensor shouldn't be used to respond to button clicks. You should instead rely on the MQTT messages (as @WagnerBeccard  mentions and which is also described on the wiki: https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant).

The binary sensor cannot be used because buttons can have multiple values (e.g. `single`, `double`, `triple`, etc.), and even if it could, it won't solve anything.

avolmensky, 2018-10-20:
No worries. Sorry missed the part of using the MQTT message as a trigger, not the sensor value.

closed 2018-10-20 by avolmensky