Koenkk/zigbee2mqtt #473: New Aqara wireless mini switch

kaelsaes, 2018-10-13:
Okt 13 17:36:28 Raspberry-MQTT npm[2498]:   zigbee2mqtt:warn 2018-10-13 17:36:28 Device with modelID 'lumi.remote.b1acn01
Okt 13 17:36:28 Raspberry-MQTT npm[2498]:   zigbee2mqtt:warn 2018-10-13 17:36:28 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

Koenkk, 2018-10-13:
Seems its already supported, could you update to the latest dev branch? https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L82

kaelsaes, 2018-10-13:
@Koenkk 

Thanks.

How can I update the dev branch?

Is my devices.js overwritten with the update?

Koenkk, 2018-10-15:
That can be done by checking out the dev branch (https://github.com/koenkk/zigbee2mqtt/tree/dev) and then: `rm -rf node_modules && npm install`. Your `devices.js` will be overwritten.

closed 2018-10-17 by kaelsaes