Koenkk/zigbee2mqtt #474: Cannot seem to pair Xiaomi WSDCGQ11LM Sensor (temp/humid/press) -  zigbee2mqtt:warn Message without device!

deten, 2018-10-14:
Hi Folks,

I cannot seem to add my third sensor.  I checked the configs and everything seems to be fine.

When I pair the device by holding down the button until the blue light blinks, I get a warning in the NPM log:

`  zigbee2mqtt:warn 2018-10-13 19:33:04 Message without device!`

I am not sure what to do, I have tried dozens of things and cant seem to get it working.  I appreciate any help!

Zigbmee2mqtt Config:

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: xxxxx
  password: xxxxx
serial:
  port: /dev/ttyACM1
devices:
  '0x00158d0002437c56':
    friendly_name: 'Stat 1'
    retain: false
  '0x00158d00023208ae':
    friendly_name: 'Stat 2'
    retain: false
```

HomeAssistant Config:

```
mqtt:
  discovery: true
  broker: 192.168.1.xx
  port: xx
  client_id: home-assistant-1
  username: xxx
  password: xxx
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'
```

FutureCow, 2018-10-14:
Try paring it again(and again) had the same problem, after a while it worked.

deten, 2018-10-14:
I keep getting the following over and over again:

Cannot get the Node Descriptor of the Device. (Error: Timed out after 10000 ms)

Koenkk, 2018-10-14:
This happens when the pairing procedure is not correctly followed, see https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi

deten, 2018-10-14:
Hrmm,

I have tried it about 80 different ways now.  Tried pulling battery and various other things ive read.  Removing unknown devices from database.db and trying again.

Honestly I am at a loss.

Any further info that may help?

SalkinDK, 2018-10-15:
> Hrmm,
> 
> I have tried it about 80 different ways now. Tried pulling battery and various other things ive read. Removing unknown devices from database.db and trying again.
> 
> Honestly I am at a loss.
> 
> Any further info that may help?

A far fetch, but have you tried restarting the zigbee2mqtt add-on?
I had a similar problem with some ikea bulbs and got the same error (Cannot get the Node Descriptor of the Device. (Error: Timed out after 10000 ms)) When i restartet the add-on the paired instantly!

lolorc, 2018-10-15:
have you tried removing the device from zigbee2mqtt with the mqtt command and also from the database.db file ?

deten, 2018-10-15:
@SalkinDK 

I am using Hassbian, so I dont have an addon, but I have restarted my rbpi and the individual services running.

deten, 2018-10-15:
@lolorc 

Not familiar with that, I will look into it, if you know yourself please post here so I can try it!

faronov, 2018-10-16:
> I keep getting the following over and over again:
> 
> Cannot get the Node Descriptor of the Device. (Error: Timed out after 10000 ms)

Same problem for the motion sensor, and for temperature/humidity 
You just need press join button every 2-3 second and keep device closely to coordinator
and everything will be fine
if not, try update battery
that always help me

lolorc, 2018-10-16:
@deten 

using unix cmds,

to remove the device from zigbee2mqtt, example:
```
mosquito_pub -h server -t zigbee2mqtt/bridge/config/remove -m '{"friendly_name": "FRIENDLY_NAME"}'
```
to remove from database.db, an example here https://github.com/Koenkk/zigbee2mqtt/issues/350#issuecomment-425871870

deten, 2018-10-18:
Okay, I figured it out.  The short of it is I should be ashamed.  I had 3 of these sensors, 2 were paired and hanging in bedrooms.  The third one wouldnt pair (WTF!).  But it WAS paired and I had mixed up 2 of them.

So all this work has been for naught because they have been working fine.  I tried to pair the one upstairs hanging which I thought was working and it registered right away.

Issue Closed.

closed 2018-10-18 by deten