Koenkk/zigbee2mqtt #475: Osram Plug

kaelsaes, 2018-10-14:
I have paired successfully two new Osram Plugs.

When I restart zigbee2mqtt these messages appear in the log:

Okt 14 06:45:54 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:54 MQTT publish, topic: 'zigbee2mqtt/ze_Osram_Plug_1', payload: '{"state":"OFF","linkquality":84}'
Okt 14 06:45:54 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:54 MQTT publish, topic: 'zigbee2mqtt/ar_Osram_Plug_2', payload: '{"state":"OFF","linkquality":7}'
Okt 14 06:45:54 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:54 MQTT publish, topic: 'zigbee2mqtt/te_Xiaomi_Water_8', payload: '{"battery":"100.00","voltage":3015,"linkquality":0,"water_leak":false}'
Okt 14 06:45:54 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:54 MQTT publish, topic: 'zigbee2mqtt/Xiaomi_Button_4', payload: '{"battery":"100.00","voltage":3065,"linkquality":73}'
Okt 14 06:45:55 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:55 Succesfully configured ze_Osram_Plug_1 0x7cb03eaa00ad27f6
Okt 14 06:45:55 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:55 Succesfully configured ar_Osram_Plug_2 0x7cb03eaa00ad468b
Okt 14 06:45:55 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:55 MQTT publish, topic: 'zigbee2mqtt/ar_Osram_Plug_2', payload: '{"state":"OFF","linkquality":5}'
Okt 14 06:45:55 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:55 MQTT publish, topic: 'zigbee2mqtt/ze_Osram_Plug_1', payload: '{"state":"OFF","linkquality":84}'
Okt 14 06:45:56 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:56 Device incoming...
Okt 14 06:45:56 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:45:56 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
Okt 14 06:46:04 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:46:04 Device incoming...
Okt 14 06:46:04 Raspberry-MQTT npm[3309]:   zigbee2mqtt:info 2018-10-14 06:46:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'

Why do the messages "Succesfully configured..." appear?
And why do the messages "Device incoming..." appear?

Sometimes after the "Device incoming..." messages an error message "Node descriptor" appears.

All other devices don't create such messages at zigbee2mqtt restart.

Koenkk, 2018-10-14:
This device needs to be configured, otherwise it doesn't reports its on/off state when toggling it via the button.

kaelsaes, 2018-10-15:
How can it be configured?

closed 2018-10-17 by kaelsaes