Koenkk/zigbee2mqtt #479: Can't pair Ikea Tradfri LED1623G12

gabe0815, 2018-10-14:
Although it is a supported device and I know it works for some people, I'm not able to pair any of these devices. I tried the master as well as the dev branch (commit #98302be) and I've reflashed the z-stack firmware. Xiaomi Mija temp sensors and buttons paired without issues. 

Factory resetting the bulbs did not solve the problem.

Any hints?

Koenkk, 2018-10-15:
Do you keep it close (a few cms) to the coordinator after factory reset?

gabe0815, 2018-10-15:
I've placed the bulb < 5cm from the zigbee-sniffer (CC2531).

Koenkk, 2018-10-15:
Can you move the sniffer and bulb to a different location to make sure it doesn't connect to another zigbee network?

bmachuletz, 2018-10-15:
Set the bridge to pairing off and pairing on again. I wrote a little Windows Forms Tool to act as a Mqtt-Client. But and other Mqtt-Client will also do that trick.

gabe0815, 2018-10-15:
I setup iobroker with iobroker.zigbee adapter (0.7.5) and I'm still not able to pair any of my 3 Tradfri bulbs. My Xiaomi temp sensor (WSDCGQ01LM) pairs on first try... Is there anyway to troubleshoot this further?

bmachuletz, 2018-10-15:
My temp sensors are pairing with no Problems too. To pair the TRÅDFRI bulb set pairing of via mqtt and set pairing on again. This should do the thing.

bmachuletz, 2018-10-15:
> I switched zigbee2mqtt/bridge/config/permit_join from true/false several times (resetting the bulbs) without success.

Do you have the IKEA Gateway powered in. Please       try pairing with IKEA Gateway switched off.

gabe0815, 2018-10-15:
I do not own a Tradfri gateway. The only other zigbee devices I have are Xiaomi sensors and switches (no gateway neither).

gabe0815, 2018-10-18:
I finally managed to pair the Ikea Tradfri LED1623G12. It turns out, I was never able to perform a factory reset. Carlessly flicking on and off the bulb never worked for me. 
To properly reset the bulbs, the off-time must be longer (~2 s) than the on time (<1 s). The distance to the coordinator is not super critical (1.5 m in my case) and other devices (Tradfri and Xiaomi) can be on and paired at the same time. 
Thanks for your help and this great project!

closed 2018-10-18 by gabe0815