Koenkk/zigbee2mqtt #481: Share your real life experience with zigbee2mqtt and +++15 devices with several routers

large, 2018-10-15:
I have tested the coordinator firmware for quite some while and are happy with it in small scale.

In my living room the owner before us mounted 48(!) GU10 spots in the kitchen/livingroom spread on 4 dimmers (analog shit). 

I'm considering if I can buy 48xGU10 IKEA Trådfri ( https://www.ikea.com/us/en/catalog/products/60365266/ ) and pair them up to my coordinator CC2531 one by one (to not reach the 15 continuous active units limit) and install them in their final socket. As I have seen other Trådfri bulbs they are acting as "routers".

So in theory if 1 or more bulbs are reaching the coordinator all others will pair together with another router and then forward the coordinator packets between them (as a mesh network).

Before I go into this journey I want to hear your experience.
So I know if I'm the right path or not...

ben423423n32j14e, 2018-10-15:
I have 17 devices (8 of which are routers) joined to my CC2531, most of the routers are Philips Hue Lightstrip V2's. So far I have not seen any deterioration in the performance of the network, from my perspective its still running flawlessly.

Not as many devices as you are talking about though.

I'm thinking of doing similar (I've got 31 spot lights in an apartment), however I'd highly recommend waiting until Zigbee2MQTT Z-Stack 3.0 firmware is released and see if the team make a recommendation to use it only with upgraded hardware.

I don't think it's too far away, see: https://github.com/Koenkk/zigbee2mqtt/issues/211

lolorc, 2018-10-15:
I'm running zigbee2mqtt with 29 devices, 4 of them are osram plug also actings as router.
At the moment it's a hassle to pair the devices. I'm curious and patient enough but I wouldn't recommend zigbee2mqtt to everyone yet.

There're still very weird issues like coordinator refusing to re-use its panId. in that case you have to repair all the devices.
It's still not ok to pair through router, so you basically have to move each sensor next to the coordinator and try to pair it again.
Comparing to other zigbee coordinators, the pairing process is not as straightforward as clicking the pairing button. Most of the time the pairing process has to be tried again and again.

Last time I had to repair my 30ish devices, it took me half a day...

concerning z-stack 3.0, I'm not that confident, If you look at the changes done in HalleyAssist repos (zigbee-sheperd, zcl-*, cc-znp, etc...) there're a lot of changes to look at. the forks we are using with zigbee2mqtt have diverged and are not updated very often.

large, 2018-10-15:
Thank you all for the feedback. The WAF is probably not existing for the zigbee2mqtt and after consideration it is a totally "geek" product (love it though).

Now I am thinking of using the Trådfri Gateway and the Home Assistant object to control it: https://www.home-assistant.io/components/tradfri/ that should probably make it more user friendly for a non-tech wife too (app will work)...
Since HA can use automation groups, then it will probably expand 10-blub-group-limit in the gateway by adding 5 groups to one dimmer.

I still though want to hear if anyone have tested performance around +50 units.
And does it matter if it is a router or end-device regarding performance?

WinSCaP, 2018-10-16:
Currently I have over 30 device connected 20 of them are routes (LED bulbs and the IKEA Tradfri GU10 spots). I have about 2 disconnects per week and will move most of them to my Hue bridge, to make sure my hobby is not interfering with easy of use for my wife.

Zigbee2mqtt is awesome to regain privacy in regards to Xiaomi devices. :)

lolorc, 2018-10-16:
i simple firewall rule is also awesome to regain privacy wrt xiaomi ;-)

large, 2018-10-16:
@WinSCaP thanks for sharing. I was afraid that random disconnects could happen, that is whats holds me back atm.

My wife said that she demands a physical switch/dimmer to ensure that she could override any scenarios, moods or whatever "magic" I implement :)
Do you all have a physical "override switch" to ensure usage if everything goes down while you're on a business trip?

lolorc, 2018-10-16:
@large, I know what you mean, and it's not so easy :-)
Here I have a lot of yeelights, most of them replace old bulbs and are still behind real switches. In that case there's no automation/control when the yeelights are not powered.
But that's some kind of override and I'm not to keen to remove yet as it still allows to switch the lights on/off when something is wrong.

and it's becoming quite complex because all those automation now requires that everything works all the time. it must supports power/network outage, everything should come back immediately, the server and gateways must always be on, there can't be any software issue (fsck stuck, /tmp full, unfinished software upgrades, etc...)

arteck, 2018-10-17:
we have for all you visions one solution.. [iobroker ](http://iobroker.de/).. 
as example here my zigbee only..
[image]

I still have a yeelight devices.. Fibaro roller shutter ..Xiaomi .. 

all this is controlable with Alexa .. or Visualisation .. but see for yourself

large, 2018-10-17:
@arteck thank you for sharing 👍 
I was not aware of the ioBroker project and have been reading about it now.

The IOBroker.zigbee ( https://github.com/ioBroker/ioBroker.zigbee ) is a fork of zigbee2mqtt firmware, right?
That means the performance of the Zigbee protocol and number of units should be the same on both fw. So if I understand it right; ioBroker replace the mqtt part; https://www.mysensors.org/controller/iobroker
I thought that MQTT ( http://mqtt.org/ ) was the standard of sharing IoT-info between systems, so what benefits does ioBroker give?

ben423423n32j14e, 2018-10-17:
I have two backups of every dongle sitting in the cupboard and the HA server is a VM running on top of a Proxmox cluster.

I can transfer it between physical hosts, backup, restore and snapshot it. In that way failure of the VM is never really a concern, worst I have to do is move the USB dongles to another host in the cluster.

I've also go an extensive number of "self healing" scripts that fix various known issues with the system.

large, 2018-10-17:
@ben423423n32j14e good tip regarding VMs. I have some experience with ESXi servers, but unfortunately USB devices are not supported, but I guess more simpler VM with KVM could handle it.
So if I understand you have 2 physical servers running equal VMs and have to move dongles when disaster strikes... Do you have a "master" in the cluster and replicate after each change then?

My experience is that hardware lives looong (I have a HP DL380 that never failed me) so my main concern is that 1-2 devices drop out in a weekly/monthly basis because of zigbee problems, that part has to be robust to ensure that things doesn't stop there.

ben423423n32j14e, 2018-10-17:
@large only 1 copy of the VM is running at a time, there are multiple Proxmox hosts in the cluster and a Synology NAS. The VM gets backed up to the NAS daily, if I need to move hosts I can have it restoring to new hardware within seconds. Not to mention the benefit of being able to quickly snapshot the whole vm before upgrading something like zigbee2mqtt, I don't really experience any extended downtime events, I can always just restore or revert a snapshot. Coupled with keeping multiple dongles, eg I've got 3x CC2531 all flashed with the same firmware ready to go, like I said failure isn't really a concern because it's so quick and easy to fix.

I'd suggest downloading and playing with Proxmox, it's very easy to use and works with all my USB devices (Z-Wave, CC2531, RFXCOM etc...).

lolorc, 2018-10-17:
@ben423423n32j14e ever tried to recover ?
because as far as I've experienced, you won't replace your coordinator by another without special ugly tricks.

ben423423n32j14e, 2018-10-17:
@lolorc yes I know about potentially losing the Zigbee network, I was talking about the HA server and control software. Apparently there is a possibility we might be able to backup and restore CC2531 in the future. https://github.com/Koenkk/zigbee2mqtt/issues/282

large, 2018-10-17:
I'm more interested in what solution could benefit my home the most.
And I have come to the same conclusion as @lolorc , it is experimental. It works fine with my +10 objects on my desk, but I don't think it will handle 48++.

So what experience do you have with other solutions, like Conbee, Raspbee or the Phoscon gateway severing data to deCONZ let HA poll data from deCONZ?
It is stated that the Phoscon takes over 200 units: https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/description/?L=1&eID=dam_frontend_push&docID=2695

Conbee costs 73 bucks in my country so is it worth it or do you have other experiences?

lolorc, 2018-10-17:
I'm personally not interested in buying another solution, I already come from a french project called zigate, made around a NXP JN5168 (aqara devices uses JN5169) Unfortunately the project is nearly dead, if it's not it's not lively enough, the firmware is closedsource, that's not helping in any way.
The community around  zigbee2mqtt is quite reactive, I still have vital sensors using the xiaomi gateway, but hey it's not going to last long ! :-)

deconz, as far as I remember, the firmware is also closed, it's a no no for me, especially at this price.

arteck, 2018-10-18:
> 
> 
> @arteck thank you for sharing +1
> I was not aware of the ioBroker project and have been reading about it now.
> 
> The IOBroker.zigbee ( https://github.com/ioBroker/ioBroker.zigbee ) is a fork of zigbee2mqtt firmware, right?
> That means the performance of the Zigbee protocol and number of units should be the same on both fw. So if I understand it right; ioBroker replace the mqtt part; https://www.mysensors.org/controller/iobroker
> I thought that MQTT ( http://mqtt.org/ ) was the standard of sharing IoT-info between systems, so what benefits does ioBroker give?

no its not a fork .. the adapter is from [ilya](https://github.com/kirovilya)..and work stable.. we use only the zigbee-sheperd libs..

but you can use zigbee2mqtt with mqtt adapter.. iobroker combine all (many) deviecs to ONE application

Koenkk, 2018-10-18:
@large I use an Intel Nuc with coreos. 

I use a CC2531 as a coordinator which is in my fuse box. In my living room, where most of my devices are, I have a CC2530 router, most of the devices are connected to this router.

ben423423n32j14e, 2018-10-20:
Funnily enough Zigbee2MQTT is already more reliable than my Z-Wave network (not surprised).

lolorc, 2018-10-25:
as I said  in https://github.com/Koenkk/zigbee2mqtt/issues/398#issuecomment-432999595 my devices were "connected" to the coordinator when there were just a few of them. When I exceeded the 15 devices some of them started to be routed through my routers. I can't explain why.

glentakahashi, 2018-10-27:
I have 40 devices running off of a single CC2531 with my custom fork here: https://github.com/Koenkk/Z-Stack-firmware/issues/10#issuecomment-423723585

It works fine for the most part, mainly having issues when issuing large amounts of commands at once, which will be fixed with group support.

jarrah31, 2018-10-27:
@glentakahashi if we use your coordinator firmware with a router (your linked post indicates there’s room for one router) such as the CC2531 with this firmware 
http://ptvo.info/cc2531-based-router-firmware-136/, how many devices could it support? 
I need a solution that will support 60+ Xiaomi devices...

alinelena, 2018-10-27:
there is an alternative firmware for cc2531 coordinator, 44 devices(or so) check one of the pr for zigbee2mqtt.

my experience in case it helps. 
i have a mixture of ikea gu10 (8) and e17 1000lm (4) bulbs plus various xiaomi sensors (14) (all via zigbee2mqtt) and diy sensors directly into homeassitant via mqtt. 
one of the 1000lm acts as a router for the ikea gu10s. any form of automatation for gu10 i tries up to now seems to be unsatisfactory. gu10 bulbs almost always out of sync. I can speculate is because of the router but do not cite me on that. All the stuff which seems connected to coordinator directly works as expected. 
pairing the ikea stuff turned out to be a job.. ended up building a relay controlled via mqtt to somehow reliable pair them. 
Xiaomi stuff except the motion sensors (aqara ones) paired without issues. 
All this is on a two floors house, 50sqm each. Homeassistant server runs on a bare metal decent hardware 
the good bit is that the old switches can work to override the ha/zigbee2mqtt so keeps everyone happy inside the house.

jasimancas, 2018-11-20:
I have 24 devices and I have problems only with one Xioami model QBKG04LM, do you have this switch?

closed 2019-01-26 by stale[bot]