Koenkk/zigbee2mqtt #482: Determine location on wiki page 'running the bridge'

JanJaapKo, 2018-10-15:
Hi,

I installed the USB stick, but had already a Z-wave stick running, so when I enter `ls -l /dev/ttyACM0` it already returned the correct output before I inserted the sniffer.

So a better procedure would be to check `ls -l /dev/ttyACM0` before inserting the sniffer, see that output is `ls: cannot access '/dev/ttyACM0': No such file or directory` then insert the sniffer, enter again 'ls -l /dev/ttyACM0' and now see the mentioned output. If not then continue with 'ls -l /dev/ttyACM1'

Or maybe there is a more elegant way?

JLeandroFreitas, 2018-10-17:
This is not a issue this is asking for help/knowledge.
Do the following in terminal.
cd /dev/serial/by-id
ls

JanJaapKo, 2018-10-17:
I agree this is not really an issue, but by lack of a forum for a discussion I started it here, hope that's not too bad. 

The mentioned command above also does not lead me (easily) to the serial port that I need to enter in the config.

lolorc, 2018-10-18:
I guess you should find some hints in https://github.com/Koenkk/zigbee2mqtt/issues/463
iow, how to make persistent device names based on the serial number (for instance)

JLeandroFreitas, 2018-10-18:
pi@raspberry:/ $ cd /dev/serial/by-id
pi@raspberry:/dev/serial/by-id $ ls -l
total 0
lrwxrwxrwx 1 root root 13 Oct  2 09:15 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D9DE2E-if00 -> ../../ttyACM0
pi@raspberry:/dev/serial/by-id $ ls -l /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D9DE2E-if00
lrwxrwxrwx 1 root root 13 Oct  2 09:15 /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D9DE2E-if00 # <-- CC2531 on /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D9DE2E-if00

closed 2018-12-24 by stale[bot]