Koenkk/zigbee2mqtt #484: Battery warnings after update 0.1.7

kaelsaes, 2018-10-15:
After update to 0.1.7 the battery levels are wrong.

Koenkk, 2018-10-15:
A lookup is now used to determine the battery voltage (as this is non linear), see https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L14.

Koenkk, 2018-10-16:
Then probably the lookup is wrong, do you know at which voltage the sensors stop working?

kaelsaes, 2018-10-16:
No, I don't know when they stop working.

What do you mean with lookup?

jpmassena, 2018-10-16:
> Then probably the lookup is wrong, do you know at which voltage the sensors stop working?

Can I ask where did you get the data for the lookup values?

evildad, 2018-10-16:
@jpmassena 
https://github.com/Koenkk/zigbee-shepherd-converters/pull/77

closed 2018-12-22 by stale[bot]