Koenkk/zigbee2mqtt #485: WSDCGQ11LM values are getting lost with cache_state: true

Koenkk, 2018-10-15:
Zigbee2mqtt 0.1.7

Log: https://hastebin.com/uvipiliray.js

This sensor supports `humidy`, `temperature` and `pressure` but only 2 values are reported at the same time. It seems that the `cache_state` mechanism doesn't work correctly. 

[image]

hoangtuit, 2018-10-16:
I have same issue

```
Oct 16 14:05:08 hassbian npm[28096]:   zigbee2mqtt:info 2018-10-16 14:05:08 MQTT publish, topic: 'zigbee2mqtt/livingroom_sensor_weather', payload: '{"temperature":32.48,"linkquality":123,"humidity":73.68}'
Oct 16 14:05:08 hassbian npm[28096]:   zigbee2mqtt:info 2018-10-16 14:05:08 MQTT publish, topic: 'zigbee2mqtt/livingroom_sensor_weather', payload: '{"humidity":73.68,"linkquality":123,"pressure":1006}'
Oct 16 14:07:19 hassbian npm[28096]:   zigbee2mqtt:info 2018-10-16 14:07:19 MQTT publish, topic: 'zigbee2mqtt/bedroom_sensor_weather', payload: '{"temperature":30.87,"linkquality":170}'
Oct 16 14:07:19 hassbian npm[28096]:   zigbee2mqtt:info 2018-10-16 14:07:19 MQTT publish, topic: 'zigbee2mqtt/bedroom_sensor_weather', payload: '{"temperature":30.87,"linkquality":168,"humidity":78.57}'
Oct 16 14:07:19 hassbian npm[28096]:   zigbee2mqtt:info 2018-10-16 14:07:19 MQTT publish, topic: 'zigbee2mqtt/bedroom_sensor_weather', payload: '{"humidity":78.57,"linkquality":168,"pressure":1005}'

```

```
2018-10-16 14:05:08 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"temperature":32.48,"linkquality":123}, template: {{ value_json.humidity  | round }})
2018-10-16 14:05:08 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"humidity":73.68,"linkquality":123,"pressure":1006}, template: {{ value_json.temperature  | round(1) }})
2018-10-16 14:07:19 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"temperature":30.87,"linkquality":170}, template: {{ value_json.humidity  | round }})
2018-10-16 14:07:19 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"humidity":78.57,"linkquality":168,"pressure":1005}, template: {{ value_json.temperature  | round(1) }})
```

Ton1965, 2018-10-16:
Having the same problem. Probably I am wrong (never used JS befor) but it looks like commit 03d913d was missing one additional change. Currently it is necessary to change line 563 of controller.js to 'this.state.set(deviceID, messagePayload)'. As it is now cache state is set to the contents of the last zigbee message thus loosing the current cache contents.

Koenkk, 2018-10-16:
@Ton1965 nice catch, this indeed fixes this issue, thanks!

closed 2018-10-16 by Koenkk

hoangtuit, 2018-10-16:
Thanks @Ton1965

mario-g0, 2018-10-18:
so, for the hassio add-on I should write to its developer?

ryanbeaton, 2018-10-19:
No just use the edge version of the add on