Koenkk/zigbee2mqtt #487: Node Red Flow

HannesDi, 2018-10-16:
Not a really Issue...

More a question for existing solution....

Is there any Node-RED flow shows up to control light bulbs with color and brightness slider 
as converting-function or dashboard?

the royal solution (where i would end up) ist a flow converting:
the alexa skill [https://flows.nodered.org/node/node-red-contrib-alexa-home-skill](url) to zigbee2mqtt

ben423423n32j14e, 2018-10-17:
Have a look at the Domoticz Philips Hue Lightstrip V2 flow I wrote, also look at my git for the Zigbee2MQTT Node-Red Admin Dashboard.

https://github.com/ben423423n32j14e/zigbee2mqttflowdevices/tree/master/devices/Philips%20Hue%20Lightstrip%20Plus%20V2

HannesDi, 2018-10-17:
All right, thank you for the links ill take a look, the adminpanel i allready used since first day i have started ;-) great work

closed 2018-12-24 by stale[bot]