Koenkk/zigbee2mqtt #488: Tradfri issues after 0.1.7 upsate

aptonline, 2018-10-16:
Just updated to latest build and now receiving the following error but just when I’m controlling a Tradfri bulb:

'zigbee2mqtt:error 2018-10-16 17:43:58 Zigbee publish to 'address', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.'

Koenkk, 2018-10-16:
I doubt this is due to the 0.1.7 update. This error means that the bulb is not connected to the network (which can solve itself after some time) and probably happend because you had to reset zigbee2mqtt. 

Is problem still present?

thinkl33t, 2018-10-16:
I'm getting this same error after today's update (on -dev) with all my tradfri bulbs.  Other zigbee devices are fine.

thinkl33t, 2018-10-17:
Still happening here, i've pulled some bits from the docker container log:

Bulb is connected and online (from a graphviz export):
```
  "0x90fd9ffffed42631" [label="{0x90fd9ffffed42631|Router|IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (LED1545G12)|online}"];
  "0x90fd9ffffed42631" -> "0x00124b00120233ba" [label="0"]
```

Ask for it to be turned off:
```
zigbee2mqtt:info 2018-10-17 18:30:24 Zigbee publish to '0x90fd9ffffed42631', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2018-10-17 18:30:24 Zigbee publish to '0x90fd9ffffed42631', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
```

running latest dev container:
```
42633c47b66c        koenkk/zigbee2mqtt:latest-dev          "./run.sh"               2 hours ago         Up 2 hours                                   zigbee2mqtt
```

thinkl33t, 2018-10-17:
I've removed, deleted from the database, and factory reset the bulbs.  2 have rejoined and are now working happily, the third now isn't able to re-join the network.

Koenkk, 2018-10-18:
@thinkl33t it may take a few tries. (experienced the same before)

thinkl33t, 2018-10-20:
I've now managed to get the third one to rejoin, and all are working again

akudaikon, 2018-10-23:
I had this issue too. Seems like all of my Tradfri bulb's mac addresses changes after the update. Before, they started with 90fd9f (like the @thinkl33t). Now, they start with 000b57. For example, one of my bulbs changed from 90fd9ffffe20bf0a to 000b57fffeeb9325 after rejoining them after the update.

ODINServ, 2018-11-02:
Seems related to #362 

I had the same issue today, the lights (2) suppose to go on at sunset, found out after 1 hour they were not.
* MQTT topic (since it is retained) reflected the correct state of the bulb (OFF)
* Send off (Since i turn off my software switch in openhab) and ON, and the light went on again.

I'm not sure which version i run, files within the docker container have a time-stamp of 23 September.