Koenkk/zigbee2mqtt #489: Connecting a cc2530 to a Raspberry PI with usb to ttl

tobang, 2018-10-16:
Hi,
Help wanted. I have bought a cc2530:

https://www.aliexpress.com/item/CC2530-Zigbee-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-24MHz/32767470382.html?spm=a2g0s.9042311.0.0.72684c4drdWMjj

And a USB to TTL:
https://www.aliexpress.com/item/Aluminum-shell-CP2102-USB-2-0-to-TTL-UART-Module-6Pin-Serial-Converter-STC-Replace-FT232/32673467746.html?spm=a2g0s.9042311.0.0.72684c4drdWMjj

I have flashed the cc2530 with this firmware:
http://ptvo.info/wp-content/uploads/2018/09/cc2530_1.2.2a.44539_firmware.zip

My idea was to connect the cc2530 to the USB to TTL device and then plug it into my Raspberry PI. But I cannot get it working. My connection between the cc2530 and the usb to ttl is:
TX = P03
RX = P02
GND = GND
3v3 = VCC
But I am a bit confused, because I have read that P20 should be connected to GND also. But on the cc2530 board I have two GND. One beside the VCC and one beside P20. 
Maybe its completely wrong to connect the cc2530 this way or have I just connected the pins the wrong way?
Or maybe the serial port  it is configured wrong in the configuration.yaml
I also have a working cc2531 and this is the configuration
serial:
  port: /dev/ttyACM0
But I suspect that it should be different with the usb to ttl
So, could some please point me in the right direction. Maybe there is a tutorial somewhere?
Thanks
Torben

ptvoinfo, 2018-10-17:
@tobang 1. My firmware does not use hardware flow control and ignores P02.
2. I would recommend connecting your CC2530 to RPi directly (https://pinout.xyz/pinout/pin8_gpio14#, GPIO8,10,14), without the converter. Then the port name will look like /dev/ttyS1

jwillmer, 2018-10-20:
How about a wiki entry after the setup works? Could be interesting for others as well.

RodBr, 2018-11-05:
Yes, a wiki entry would be great!  I'm trying the same thing with no success, and no new ports are turning up after I wire the CC2530 in as recommended.

RodBr, 2018-11-14:
I managed to get this working.  It was quite straightforward using this firmware from @dzungpv.
[https://raw.githubusercontent.com/dzungpv/Z-Stack-firmware/master/coordinator/CC2530/bin/CC2530ZNP-Prod.hex](url)
The USB TTL made it very easy to set the wiring.

giejay, 2019-01-11:
> @tobang 1. My firmware does not use hardware flow control and ignores P02.
> 2. I would recommend connecting your CC2530 to RPi directly (https://pinout.xyz/pinout/pin8_gpio14#, GPIO8,10,14), without the converter. Then the port name will look like /dev/ttyS1

And which pins on the cc2530 should be connected to 8,10 and 14?

ptvoinfo, 2019-01-11:
@giejay 
Rpi 8 -> P03
Rpi 10 -> P02 
Rpi 14 -> GND

JLFN, 2019-02-02:
@ptvoinfo  What about 3.3v ?  Rpi 1 -> VCC

ptvoinfo, 2019-02-03:
@JLFN This is possible. No problems, but it depends on your CC2530 board. Some boards use 5v only.

giejay, 2019-02-08:
So, first I flashed the coordinator firmware to my CC2530 and then I've attached it to my RPI through:

Rpi 8 -> P03
Rpi 10 -> P02
Rpi 14 -> GND
Rpi 1 -> VCC

When starting zigbee2mqtt Im getting the following error:

```

  serialport:binding:auto-detect loading LinuxBinding +0ms
  zigbee2mqtt:debug 2019-2-8 15:04:56 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/app/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-2-8 15:04:56 Loaded state from file /app/data/state.json
  zigbee2mqtt:info 2019-2-8 15:04:57 Starting zigbee2mqtt version 0.2.0 (commit #b0d3c2f)
  zigbee2mqtt:info 2019-2-8 15:04:57 Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  serialport:main opening path: /dev/ttyAMA0 +0ms
  serialport:bindings open +0ms
  serialport:main _read queueing _read for after open +18ms
  serialport:poller Creating poller +0ms
  serialport:main opened path: /dev/ttyAMA0 +21ms
  cc-znp The serialport /dev/ttyAMA0 is opened. +94ms
  serialport:main _read reading +11ms
  serialport:bindings read +49ms
  serialport:unixRead Starting read +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +22ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +16ms
  serialport:main _write 8 bytes of data +77ms
  serialport:bindings write 8 bytes +77ms
  serialport:unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +103ms
  serialport:poller Polling for "readable" +128ms
  serialport:unixWrite write returned null 8 +31ms
  serialport:unixWrite wrote 8 bytes +1ms
  serialport:unixWrite Finished writing 8 bytes +1ms
  serialport:main binding.write write finished +46ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +4ms
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +26ms
  zigbee2mqtt:info 2019-2-8 15:05:00 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  serialport:main #flush +3s
  serialport:bindings flush +3s
  serialport:main binding.flush finished +10ms
  serialport:main #close +3ms
  serialport:bindings close +13ms
  serialport:poller Stopping poller +3s
  serialport:poller Destroying poller +4ms
  serialport:main binding.read error { Error: Canceled
    at Poller.emitCanceled (/app/node_modules/serialport/lib/bindings/poller.js:102:17)
    at Poller.stop (/app/node_modules/serialport/lib/bindings/poller.js:92:10)
    at close.then (/app/node_modules/serialport/lib/bindings/linux.js:49:21)
    at <anonymous> canceled: true } +12ms
  serialport:main _read queueing _read for after open +3ms
  serialport:main binding.close finished +3ms
  cc-znp The serialport /dev/ttyAMA0 is closed. +54ms
^C  zigbee2mqtt:debug 2019-2-8 15:05:02 Saving state to file /app/data/state.json
  zigbee2mqtt:error 2019-2-8 15:05:02 Not connected to MQTT server!
  zigbee2mqtt:error 2019-2-8 15:05:02 Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
  zigbee-shepherd zigbee-shepherd is stopping. +2s
  zigbee-shepherd zigbee-shepherd is stopped. +6ms
  zigbee2mqtt:info 2019-2-8 15:05:02 zigbee-shepherd stopped
```

It should be mounted at AMA0 right? Any way to test that?

JLFN, 2019-02-08:
@giejay   
```md
#I’m running raspbian on a RPi. I use a dedicated user named pi which is added to group dialout to allow access to the device (or you can use root).
# Device can also be connect to with an  3.3v usb TTL or direct to Raspberry Pi TX,RX,3V3,GND. Do not connect to 5V it will destory the cc2530
#cc2530 = P02 -> Raspberry Pi = TX    = PIN 10 
#cc2530 = P03 -> Raspberry Pi = RX    = PIN 8
#cc2530 = GND -> Raspberry Pi = GND   = PIN 14
#cc2530 = VCC -> Raspberry  Pi = 3.3V = PIN 1

#Running with Raspberry pi TX and RX
#Disabling on-board Bluetooth for using RX/TX

#Open /boot/config.txt file.
sudo nano /boot/config.txt

#Add the following lines to the end of the file:
# Disable Bluetooth
dtoverlay=pi3-disable-bt

#Disable related services.
sudo systemctl disable hciuart.service
sudo systemctl disable bluealsa.service
sudo systemctl disable bluetooth.service
#Reboot is needed
sudo reboot
```

See more https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/

giejay, 2019-02-08:
Thats exactly what I all did. First on a rasbperry pi 3, and then I tried on a RPI2. Im running it in docker so I dont think its a user access rights issue. Still the error remains:(

JLFN, 2019-02-08:
@giejay  

You need to enable serial wiith raspi-config or

JLFN, 2019-02-08:
@giejay Do you have a esp8266 laying around?

giejay, 2019-02-08:
Unfortunately I already did that and that didn't help much. I am using Hypriot instead of Raspbian so I will first flash the latest raspbian on my rpi3 and follow the guide from the link you sent. I don't have a esp:( 

Thanks for the help!

giejay, 2019-02-08:
Did a reinstall of the rpi3, now with raspbian. Followed the guide to the letter, hooked up my cc2530 and still the same error:

```
 zigbee2mqtt:debug 2/8/2019, 7:41:40 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11],"precfgkey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]},"dbPath":"/home/pi/zigbee/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2/8/2019, 7:41:40 PM Loaded state from file /home/pi/zigbee/data/state.json
  zigbee2mqtt:info 2/8/2019, 7:41:40 PM Starting zigbee2mqtt version 1.1.1 (commit #a497cf0)
  zigbee2mqtt:info 2/8/2019, 7:41:40 PM Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  serialport:main opening path: /dev/serial0 +0ms
  serialport:bindings open +0ms
  serialport:main _read queueing _read for after open +5ms
  serialport:poller Creating poller +0ms
  serialport:main opened path: /dev/serial0 +6ms
  cc-znp The serialport /dev/serial0 is opened. +0ms
  serialport:main _read reading +4ms
  serialport:bindings read +15ms
  serialport:unixRead Starting read +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +0ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +9ms
  serialport:main _write 8 bytes of data +23ms
  serialport:bindings write 8 bytes +22ms
  serialport:unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +27ms
  serialport:poller Polling for "readable" +34ms
  serialport:unixWrite write returned null 8 +6ms
  serialport:unixWrite wrote 8 bytes +0ms
  serialport:unixWrite Finished writing 8 bytes +0ms
  serialport:main binding.write write finished +9ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/home/pi/zigbee/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:277:13)
    at CcZnp.emit (events.js:189:13)
    at Timeout.<anonymous> (/home/pi/zigbee/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10) +0ms
  zigbee2mqtt:info 2/8/2019, 7:41:43 PM Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  serialport:main #flush +3s
  serialport:bindings flush +3s
  serialport:main binding.flush finished +4ms
  serialport:main #close +2ms
  serialport:bindings close +6ms
  serialport:poller Stopping poller +3s
  serialport:poller Destroying poller +2ms
  serialport:main binding.read error { Error: Canceled
    at Poller.emitCanceled (/home/pi/zigbee/node_modules/serialport/lib/bindings/poller.js:102:17)
    at Poller.stop (/home/pi/zigbee/node_modules/serialport/lib/bindings/poller.js:92:10)
    at close.then (/home/pi/zigbee/node_modules/serialport/lib/bindings/linux.js:49:21) canceled: true } +5ms
  serialport:main _read queueing _read for after open +2ms
  serialport:main binding.close finished +1ms
  cc-znp The serialport /dev/serial0 is closed. +35ms
```

To test if it wasn't a docker permission issue on the AMA0 port, I installed zigbee2mqtt through npm on the pi itself. No luck unfortunately.

nob0dy80, 2019-02-12:
cc2530 = P02 -> Raspberry Pi = TX    = PIN 10 
cc2530 = P03 -> Raspberry Pi = RX    = PIN 8

This is wrong ! just swap pin 8 and 10 ! at least its on raspberry 3b+ like this.
now its working fine for me too.

giejay, 2019-02-12:
I followed the instructions by @ptvoinfo. So you are saying, connect p03 with rpi10 and p02 with 8 and everything will be ok? I will try tonight

nob0dy80, 2019-02-12:
> I followed the instructions by @ptvoinfo. So you are saying, connect p03 with rpi10 and p02 with 8 and everything will be ok? I will try tonight

You got it. I'm pretty sure, it will work for you too.
Drop a feedback after testing!

giejay, 2019-02-12:
I have checked a lot of forums and issues and I assumed it had something to do with me flashing the incorrect firmware or the pi3 changing the uart stuff. Never doubted the pin layout that was posted in this thread. But I should have, it's all working now! Thank you very much, I almost gave up.

nob0dy80, 2019-02-12:
well. at least you didnt look close enough..i found the solution in THIS git..
you're welcome...glad its working now
not posting much,only if i'm sure i can help..and your problem was very clear in the logs.. (again, why you ever thought over permissions?!

enjoy your zigbee stick now.. :-)

nob0dy80, 2019-02-12:
So finally to close this issue..the correct pin layout to connect a raspi3b+ over uart is:
raspi pin1 -> VCC
raspi pin8 -> P02
raspi pin9 -> Ground (or any other GND pin.)
raspi pin10 -> P03

giejay, 2019-02-13:
Well, thinking it could be something with permissions is definitively not that weird. At least in previous versions of ubuntu you had to add yourself to the dialout group to prevent permission denied errors when trying to read /dev/ttyACM0. Link: https://askubuntu.com/questions/58119/changing-permissions-on-serial-port

closed 2019-04-21 by stale[bot]

ChumKiu, 2019-05-17:
Hi all, there are some update on this subject, I've also bought some CC2530 modules (I'm waiting to got them) in order to setup some testing. Thank. Regards.

JLFN, 2019-05-17:
@ChumKiu  ITs all in the Docs/wiki.

hosein682, 2020-06-04:
Hi guys, 
I don't know how I can flash CC2530 UART module on Raspberry Pi 4. Can anyone please help me with it? 
I have connected the module to UART pins of raspberry pi and I don't have USB to TTL. is it possible or do I need a CC debugger or something?

ptvoinfo, 2020-06-05:
@hosein682 Please, look here
https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html
There are special pins for flashing

hosein682, 2020-06-07:
Thanks in advance dear @ptvoinfo. It really helped.