Koenkk/zigbee2mqtt #495: Battery level and voltage on RTCGQ01LM

ghiglie, 2018-10-18:
Hi there! I'm a Home Assistant noob user; I've started using a template in  to manage battery statuses.
I've just found that on Aqara and Mija motion / human body sensors I have this data on illumination sensor data, and only on that - so the ones that haven't it, they won't show it event there's the correct data in MQTT JSON. Is there a way to "move" it from "illumination" sensor to "presence" one?

Thanks,
Marco

Koenkk, 2018-10-18:
Could you post the related part of your homeassistant configuration?

Offtopic: I would recommend you to use home assistant discovery: https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant

ghiglie, 2018-10-18:
Hi Koenkk! It's MQTT autodiscovery that generates that configuratio, I haven't written a single line of code. I can't catch the config messagese, but I'll add a screenshot ASAP.

ghiglie, 2018-10-18:
[image]
[image]

ghiglie, 2018-10-18:
Don't give attention to the loss of data, it was my wife cleaning home... :D
That's what I have on HA with autodiscovery, but the two PIR that haven't the illuminance sensor haven't got the voltage/battery data on HA - there are instead in JSON.

Koenkk, 2018-10-19:
Could you post your zigbee2mqtt log?

This may also be due to https://github.com/Koenkk/zigbee2mqtt/issues/485 which will be fixed in zigbee2mqtt 0.1.8.

ghiglie, 2018-10-21:
I'm sorry, maybe I'm too late, just upgraded to 0.1.8. My "request" is different from what you mean.
For example, in HA, my AQARA sensors that have illuminance are defined like this:
[image]
(still dunno why battery and voltage are missing on one of them!)
but "normal" mija pirs are like this:
[image]
same for door sensor, they are missing the Battery/Voltage data that is in MQTT message:
[image]

Koenkk, 2018-10-21:
The battery will only be visible after the sensor reported it, this can take some time (a few hours).

ghiglie, 2018-10-27:
Ok, I'm really missing something! I've redefined the sensor in HA, as in Wiki and adding linkquality:

```
homeassistant:
  customize:
    binary_sensor:
      - platform: "mqtt"
        state_topic: "zigbee2mqtt/0x00158d00023f3532"
        availability_topic: "zigbee2mqtt/bridge/state"
        payload_on: true
        payload_off: false
        value_template: "{{ value_json.occupancy }}"
        device_class: "motion"
        json_attributes: 
          - battery
          - voltage
          - linkquality

```
So I expect to see the added attributes on the sensor, but I can only see " value_json.battery " as battery attribute value. With MQTTfx I sniffed the messages for autodiscovery, but they all look correct, and also the occupancy sensors should have the battery/voltage attributes... 
Where am I doing bad? :(

Koenkk, 2018-10-30:
Could you provide your zigbee2mqtt log? And which HA version are you running?

Koenkk, 2018-10-31:
Everything looks OK from the zigbee2mqtt side. Can you open an issue for this @ home assistant?

ghiglie, 2018-11-01:
Sorry for late reply. I'll surely do, something is acting strange on my install. This works:
```
sensor:
  - platform: mqtt
    name: "Mija PIR 1"
    state_topic: "zigbee2mqtt/0x00158d00023f3532"
    value_template: "{{ value_json.battery }}"
    json_attributes:
      - voltage
    retain: false
    device_class: battery
```
And shows correctly the values.

iwwwwwwi, 2018-12-06:
It appears that only mqtt sensor component has the json_attributes feature implemented in haas so the automatically generated config example here for binary_sensors with additional json_attributes is a bit misleading.

Koenkk, 2018-12-07:
@iwwwwwwi this has been fixed in https://github.com/Koenkk/zigbee2mqtt/issues/605 (currently in the dev branch)

closed 2019-02-12 by stale[bot]