Koenkk/zigbee2mqtt #497: Need guidance to add support for Stelpro products

carldebilly, 2018-10-18:
## Context
I'm currently in the process to switch to Home Assistant and I'm using _zigbee2mqtt_ for my zigbee devices. I'm in Canada and here a lot of thermostats are _Line Voltate Thermostats_, meaning the power current is flowing in the thermostat itself. It seems there's only one company in the world building _Smart Home compatible_ thermostats like this:  [Stelpro](https://www.stelpro.com/en-CA/smart-home). I have many of them in my home. Some of them are zigbee other are z-wave... but let's talk about the zigbee models here.

## Outdoor temperature
They integrates pretty well as _Heating Thermostat_ and _Temperature Sensor_ in the zigbee network. But they also have a very nice feature: **they are able to display the outdoor temperature** and I want my new setup to support this.

To acheive that, they implemented a custom zigbee MFG cluster to _push_ the outdoor temperature to the device. We can see they published the [implementation of this for SmartThings](https://github.com/stelpro/Ki-ZigBee-Thermostat/blob/master/devicetypes/stelpro/stelpro-ki-zigbee-thermostat.src/stelpro-ki-zigbee-thermostat.groovy#L334).

## Implementation into Zigbee2Mqtt
Now, I need guidance to add this support into Zigbee2Mqtt. I think the best way would be to add a new [`toZigbee` converter](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js) to set the outdoor temperature and listen to a MQTT topic for the temperature to push.

Where I'm not sure is how to setup the MQTT topic listening and where to define it. Should I _hard code_ a topic for this? Is it better to have one topic per device and have something elsewhere to "push" the outdoor temperature to each of the thermostats on the network, separately instead of a shared topic?

Thanks for your help.

Koenkk, 2018-10-19:
Try adding something like:

```js
    stelpro_outdoor_temperature: {
        key: 'outdoor_temperature',
        attr: [],
        convert: (value, message) => {
            // magic here
        },
    },
```

This converter will be triggered once a message to `zigbee2mqtt/[DEVICEID]/set` with payload 

```js
{
  "outdoor_temperature": 10,
}
```

is send.

carldebilly, 2018-10-31:
I have a problem pairing the Stelpro. While pairing, I'm getting the following messages in the log file:

```
zigbee2mqtt:info 2018-10-31 01:11:34 New device with address 0xf8f005fffff78d7c connected!
zigbee2mqtt:info 2018-10-31 01:11:34 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0xf8f005fffff78d7c"}'
zigbee2mqtt:warn 2018-10-31 01:17:56 Message without device!

(...after a reboot)
zigbee2mqtt:info 2018-10-31 01:46:30 0xf8f005fffff78d7c (0xf8f005fffff78d7c): unkown - undefined unknown (Router)

```

When checking in the database.db, I can see the following:
```
{
    "id": 8,
    "type": "Router",
    "ieeeAddr": "0xf8f005fffff78d7c",
    "nwkAddr": 18326,
    "manufId": 4485,
    "manufName": "Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "powerSource": "Mains (single phase)",
    "epList": [
        25
    ],
    "status": "online",
    "joinTime": 1540950799,
    "endpoints": {
        "25": {
            "profId": 260,
            "epId": 25,
            "devId": 769,
            "inClusterList": [
                0,
                3,
                4,
                513,
                516
            ],
            "outClusterList": [
                1026
            ],
            "clusters": {
                "genBasic": {
                    "dir": {
                        "value": 1
                    },
                    "attrs": {
                        "zclVersion": 1,
                        "appVersion": 104,
                        "stackVersion": 4,
                        "hwVersion": 1,
                        "manufacturerName": "Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
                        "modelId": "STZB402+\u0000\u0007\ufffd\u0000\ufffd\u0000\ufffd\ufffd\u0000\ufffdJ\u0004\u0000\ufffd\ufffd\u0000\ufffd\u0000\u0000\u0000\ufffd\u0000\u0000",
                        "dateCode": "20170628 00062\u0000",
                        "powerSource": 1
                    }
                },
                "genIdentify": {
                    "dir": {
                        "value": 1
                    },
                    "attrs": {
                        "identifyTime": 172,
                        "identifyCommissionState": 0
                    }
                },
                "genGroups": {
                    "dir": {
                        "value": 1
                    },
                    "attrs": {
                        "nameSupport": 0
                    }
                },
                "hvacThermostat": {
                    "dir": {
                        "value": 1
                    },
                    "attrs": {
                        "16385": -32768,
                        "16412": 4,
                        "localTemp": 2100,
                        "outdoorTemp": -32768,
                        "ocupancy": 1,
                        "absMinHeatSetpointLimit": 500,
                        "absMaxHeatSetpointLimit": 3000,
                        "absMinCoolSetpointLimit": 700,
                        "absMaxCoolSetpointLimit": 4000,
                        "pIHeatingDemand": 0,
                        "systemTypeConfig": 0,
                        "localTemperatureCalibration": 0,
                        "occupiedCoolingSetpoint": 3500,
                        "occupiedHeatingSetpoint": 2100,
                        "unoccupiedCoolingSetpoint": 4000,
                        "unoccupiedHeatingSetpoint": 1700,
                        "minHeatSetpointLimit": 500,
                        "maxHeatSetpointLimit": 3000,
                        "minCoolSetpointLimit": 700,
                        "maxCoolSetpointLimit": 4000,
                        "minSetpointDeadBand": 25,
                        "remoteSensing": 2,
                        "ctrlSeqeOfOper": 2,
                        "systemMode": 4
                    }
                },
                "hvacUserInterfaceCfg": {
                    "dir": {
                        "value": 1
                    },
                    "attrs": {
                        "tempDisplayMode": 0,
                        "keypadLockout": 0,
                        "programmingVisibility": 0
                    }
                },
                "msTemperatureMeasurement": {
                    "dir": {
                        "value": 2
                    },
                    "attrs": {}
                }
            }
        }
    }
```
Obviously the device is well described there...

What is the missing piece ? I suspect the `\u0000` in the names are problematic.

bobzer, 2019-01-03:
Hi, 
I'm also trying to manage my stelpro with zigbee2mqtt, but can't even attach the device
How did you do to pair your stelpro to zigbee2mqtt ?

did you succeed in getting the outdoor temperature ?

Thank you

goldriver2, 2019-01-08:
@carldebilly 

Hi Carl, have you been able to integrate your stelpro thermostat ?

closed 2019-03-16 by stale[bot]

bobzer, 2019-03-19:
maybe this can help ? 
https://github.com/stelpro/maestro-thermostat/blob/master/devicetypes/stelpro/stelpro-maestro-thermostat.src/stelpro-maestro-thermostat.groovy

it's the file that make the integration with smart things