Koenkk/zigbee2mqtt #498: aqara cube not working

tarbax, 2018-10-18:
i just paired the cube and i get this error:
```  zigbee2mqtt:info 2018-10-18 16:25:14 New device with address 0x00158d00027d8f95 connected!
  zigbee2mqtt:info 2018-10-18 16:25:14 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d00027d8f95"}'
  zigbee2mqtt:warn 2018-10-18 16:25:14 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-10-18 16:25:14 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices```

Koenkk, 2018-10-19:
Could you provide your startup log?

tarbax, 2018-10-21:
```
[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
[2018-10-21T18:17:23.016Z] PM2 log: Launching in no daemon mode
[2018-10-21T18:17:23.106Z] PM2 log: Starting execution sequence in -fork mode- for app name:npm id:0
[2018-10-21T18:17:23.114Z] PM2 log: App name:npm id:0 online
> zigbee2mqtt@0.1.6 start /zigbee2mqtt-0.1.6
> node index.js
  zigbee2mqtt:info 2018-10-21 18:17:23 Logging to directory: '/share/zigbee2mqtt/log/2018-10-21.18-17-23'
  zigbee2mqtt:info 2018-10-21 18:17:24 Starting zigbee2mqtt version 0.1.6 (commit #unknown)
  zigbee2mqtt:info 2018-10-21 18:17:24 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-10-21 18:17:24 zigbee-shepherd started
  zigbee2mqtt:info 2018-10-21 18:17:24 Coordinator firmware version: '20180815'
  zigbee2mqtt:info 2018-10-21 18:17:24 Currently 8 devices are joined:
  zigbee2mqtt:info 2018-10-21 18:17:24 0x00158d0002386762 (0x00158d0002386762): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-21 18:17:24 0x00158d00023284b5 (0x00158d00023284b5): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-21 18:17:24 0x00158d0002340205 (0x00158d0002340205): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
  zigbee2mqtt:info 2018-10-21 18:17:24 0x001788010302ca10 (0x001788010302ca10): 8718696548738 - Philips Hue white ambiance E27 (Router)
  zigbee2mqtt:info 2018-10-21 18:17:24 0x0017880103cbae9b (0x0017880103cbae9b): 324131092621 - Philips Hue dimmer switch (EndDevice)
  zigbee2mqtt:info 2018-10-21 18:17:24 0x00158d00024e50e0 (0x00158d00024e50e0): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-21 18:17:24 0x00158d000252acea (0x00158d000252acea): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-10-21 18:17:24 0x00158d00027d8f95 (0x00158d00027d8f95): unkown - undefined unknown (EndDevice)
  zigbee2mqtt:info 2018-10-21 18:17:24 Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 2018-10-21 18:17:24 zigbee-shepherd ready
  zigbee2mqtt:info 2018-10-21 18:17:24 Connected to MQTT server
```

groovepack, 2018-10-21:
I also got these days my MFKZQ01LM and it does not work - maybe my logs help this helps
I'm running latest-dev inside docker.

2018-10-21 16:24:10 - info: Connecting with device...
2018-10-21 16:24:10 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:11 - info: Connecting with device...
2018-10-21 16:24:11 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:12 - info: Connecting with device...
2018-10-21 16:24:12 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:12 - info: Connecting with device...
2018-10-21 16:24:12 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:13 - info: Connecting with device...
2018-10-21 16:24:13 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:14 - info: Connecting with device...
2018-10-21 16:24:14 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:15 - info: Connecting with device...
2018-10-21 16:24:15 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:15 - info: Connecting with device...
2018-10-21 16:24:15 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:16 - info: Connecting with device...
2018-10-21 16:24:16 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:17 - info: Connecting with device...
2018-10-21 16:24:17 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:18 - info: Connecting with device...
2018-10-21 16:24:18 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:18 - info: Connecting with device...
2018-10-21 16:24:18 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:19 - info: Connecting with device...
2018-10-21 16:24:19 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:19 - info: Connecting with device...
2018-10-21 16:24:19 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-10-21 16:24:19 - info: Device incoming...
2018-10-21 16:24:19 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-10-21 16:24:19 - info: New device with address 0x00158d000275cc0e connected!
2018-10-21 16:24:19 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d000275cc0e"}'

config
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.10.11.55:1883'
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___........-if00
devices:
  '0x90fd9ffffe3e4b7b':
    friendly_name: Lamp1
    retain: false
  '0x00158d0002136919':
    friendly_name: Switch1
    retain: false
  '0x00158d0001ef612e':
    friendly_name: Switch2
    retain: false
  '0x00158d00029c09df':
    friendly_name: Motion1
    retain: false
    occupancy_timeout: 90
  '0x00158d00020b50c5':
    friendly_name: BadSensor
    retain: false
  '0x00158d0002090a8d':
    friendly_name: WohnzimmerSensor
    retain: false
  '0x00158d000275cc0e':
    friendly_name: '0x00158d000275cc0e'
    retain: false

startup log
[... 2 lines cut ...]
2018-10-21 16:39:15 - info: Starting zigbee2mqtt version 0.1.7 (commit #0aab565)
2018-10-21 16:39:15 - info: Starting zigbee-shepherd
2018-10-21 16:39:16 - info: zigbee-shepherd started
2018-10-21 16:39:16 - info: Coordinator firmware version: '20180815'
2018-10-21 16:39:16 - info: Currently 6 devices are joined:
2018-10-21 16:39:16 - info: Lamp1 (0x90fd9ffffe3e4b7b): LED1624G9 - IKEA TRADFRI LED bulb E27 600 lumen, dimmable, color, opal white (Router)
2018-10-21 16:39:16 - info: Switch1 (0x00158d0002136919): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
2018-10-21 16:39:16 - info: Switch2 (0x00158d0001ef612e): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
2018-10-21 16:39:16 - info: Motion1 (0x00158d00029c09df): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-10-21 16:39:16 - info: BadSensor (0x00158d00020b50c5): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-10-21 16:39:16 - info: WohnzimmerSensor (0x00158d0002090a8d): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-10-21 16:39:16 - warn: `permit_join` set to  `true` in configuration.yaml.
2018-10-21 16:39:16 - warn: Allowing new devices to join.
2018-10-21 16:39:16 - warn: Set `permit_join` to `false` once you joined all devices.
2018-10-21 16:39:16 - info: Zigbee: allowing new devices to join.
2018-10-21 16:39:16 - info: Connecting to MQTT server at mqtt://10.10.11.55:1883
[... cut ...]
2018-10-21 16:39:16 - info: MQTT publish, topic: 'zigbee2mqtt/Motion1', payload: '{"battery":"100.00","voltage":3165,"linkquality":84,"illuminance":126,"occupancy":false}'
2018-10-21 16:39:16 - info: MQTT publish, topic: 'zigbee2mqtt/BadSensor', payload: '{"humidity":60.73,"linkquality":39,"temperature":22.49,"battery":"42.00","voltage":2985}'
2018-10-21 16:39:16 - info: MQTT publish, topic: 'zigbee2mqtt/WohnzimmerSensor', payload: '{"temperature":24.48,"humidity":46.07,"linkquality":23,"battery":"42.00","voltage":2985}'

Koenkk, 2018-10-22:
@groovepack after connecting, does it still not appear in the connected devices list on startup?

groovepack, 2018-10-22:
The Logs and copies are done in that order. After connecting it appeared as device but no action are received. Then on restart it does not appear in the connected devices list.

Koenkk, 2018-10-22:
Can you post your `database.db`?

groovepack, 2018-10-22:
[Uploading database.db.zip…]()

here it is ...

Koenkk, 2018-10-23:
It is very strange that the device does not appear in your `database.db`, did you change the permissions of this file? Could you provide the complete log when trying to pair?

lociii, 2018-10-25:
Same here.

Reported on startup as
```zigbee2mqtt:info 2018-10-25 19:54:46 0x00158d00028f8637 (0x00158d00028f8637): unkown - undefined unknown (EndDevice)```

When connecting
```
zigbee2mqtt:debug 2018-10-25 19:55:19 Received zigbee message of type 'devChange' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device '0x656275635f726f736e65732e696d756c' (0x00158d00028f8637)
zigbee2mqtt:warn 2018-10-25 19:55:19 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
zigbee2mqtt:warn 2018-10-25 19:55:19 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

in database.db
```{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d00028f8637","nwkAddr":52610,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"0x656275635f726f736e65732e696d756c","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25],"outClusterList":[0,3,4,5,18,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"O7hf0G5PQcVAB8dR"}```

Running latest git on Raspberry Pi with CC2530

Koenkk, 2018-10-26:
@lociii this was a bug in an older version of zigbee2mqtt. To fix:
- stop zigbee2mqtt
- remove all entries of `0x656275635f726f736e65732e696d756c` from `database.db`
- start zigbee2mqtt
- pair

ryanbeaton, 2018-10-26:
PS: https://github.com/Koenkk/zigbee2mqtt/issues/388#issuecomment-424858076

groovepack, 2018-10-27:
So, now made a new connect. Upfront i did a restart of zigbee2mqtt. Now the device is listed in the database.db but no events are received or recognized ... :-/

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2521742/log.txt)

[database_db.txt](https://github.com/Koenkk/zigbee2mqtt/files/2521743/database_db.txt)

[configuration_yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/2521744/configuration_yaml.txt)

tarbax, 2018-10-27:
my cube is not connecting anymore nothing comes into the log

groovepack, 2018-10-29:
@tarbax did you ever get it working? Meaning have you ever had messages from the cube in MQTT?

tarbax, 2018-10-29:
i never had anything in MQTT showing from the cube. but it was pairing with errors. now i cant even pair anymore

lociii, 2018-10-29:
> @lociii this was a bug in an older version of zigbee2mqtt. To fix:
> 
> * stop zigbee2mqtt
> * remove all entries of `0x656275635f726f736e65732e696d756c` from `database.db`
> * start zigbee2mqtt
> * pair

- stopped the service
- updated repo 97cc56b691dd4e1dc9e95be0d22a4c8be3037291 
- removed database.db and state.json
- removed all devices from configuration.yaml

database entry
```
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d00028f8637","nwkAddr":2748,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"0x656275635f726f736e65732e696d756c","epList":[1,2,3],"status":"online","joinTime":1540841383,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25],"outClusterList":[0,3,4,5,18,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"vR7J7Zgk0dD4SsRi"}
```

still listed as unsupported.

```
zigbee2mqtt:warn 2018-10-29 19:33:12 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
```

tarbax, 2018-10-29:
i run hassio and the plugin provided is at version 0.1.6

tarbax, 2018-11-06:
zigbee2mqtt:error 2018-11-6 22:44:26 Cannot get the Node Descriptor of the Device: 0x00158d00027d8f95 (Error: Timed out after 10000 ms)

tarbax, 2018-11-07:
```
  zigbee2mqtt:warn 2018-11-7 17:31:59 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-11-7 17:31:59 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

closed 2018-11-07 by tarbax

lociii, 2018-11-07:
Why closed without comment?

Koenkk, 2018-11-07:
@lociii after update, did you do a `rm -rf node_modules && npm install`?

tarbax, 2018-11-07:
@lociii @Koenkk 
oh my fault i did not enter the comment :-/
after the reinstall i did yesterday, i discovered that i had a faulty database file.
and once i updated from 0.16 to 0.18 and tried again the problem was solved.
the cube is connected without any errors and evry function works.

lociii, 2018-12-04:
@Koenkk @tarbax 
running 0.2.0 now.
cleaned my config and my database, removed node modules and reinstalled them.
now the cube doesn't even show up when i try to connect it.
really weird.