Koenkk/zigbee2mqtt #499: Example automation in Wiki not working

mario-g0, 2018-10-18:
I'm trying to set up an automation in Home Assistant using the example in the Wiki: https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant

but it just doesn't work... has anyone event tested it?

carldebilly, 2018-10-18:
You should give more details about your installation or logs...  because I think most of the users of `zigbee2mqtt` are successfully using it with Home Assistant.

What do you mean by _it just doesn't work_?

1. Did you check the logs for error details? You can also enable debug mode <https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug>.
2. Do you see the connection on your MQTT broker?
3. Do you see any MQTT traffic?

WagnerBeccard, 2018-10-18:
The settings there are the setups for e.g.  sensor not an automation.

Example for a round Xiaomi button

  - alias: single
    trigger:
      platform: mqtt
      topic: 'zigbee2mqtt/xiaomibutton'
    condition:
      condition: template
      value_template: "{{ 'single' == trigger.payload_json.click }}"
    action:
    - service: scene.turn_on
      entity_id: scene.vr_1

You use the linked siteto look for the condition templates.

closed 2018-12-25 by stale[bot]

caffeineflo, 2019-01-20:
I think he is right, the syntax in the wiki example is broken

Koenkk, 2019-01-21:
@caffeineflo could you provide your zigbee2mqtt logging and your automation configuration? The example on the wiki works fine in my case.

caffeineflo, 2019-01-21:
@Koenkk I don't think there's really much you can do. If you paste the yaml into your configuration the indentation gets screwed up due to the way it's displayed through HTML.

The only way of making this better that I can think of, is keeping the automation content in a separate file and link to the raw file on GH. That would prevent that issue for most people

vpomax, 2019-01-23:
I have the same problem, my automation:

  - alias: Switch Yeelight
    trigger:
      platform: mqtt
      topic: 'zigbee2mqtt/sensor.0x00158d0001593043_click'
    condition:
      condition: template
      value_template: "{{ 'single' == trigger.payload_json.click }}"
    action:
      service: light.toggle
      entity_id: light.yeelightw1

it do not works...

in the log of zigbee2mqtt I found:

`zigbee2mqtt:info 2019-1-23 09:56:32 MQTT publish: topic 'zigbee2mqtt/0x00158d0001593043', payload '{"battery":100,"voltage":3032,"linkquality":28,"last_seen":"2019-01-23T08:56:32.381Z","click":"single"}'`

but nothing happens

vpomax, 2019-01-23:
I have just did it but unsuccesfull...

sehraf, 2019-01-23:
Mhh... Can you check your home assistant log book to see if the automation gets triggered?

vpomax, 2019-01-23:
NO!
in logbook I found only.... "0x00158d0001593043_click changed to"
without the autmation I called "Switch Yeelight"

why it do not start the automation?

vpomax, 2019-01-23:
Temporarily I solved in another way... not using template mqtt but using template: state

Since my Hassio discover xiaomo switch and call it "sensor.0x00158d0001593043_click"
I created a new automation...

```
  - alias: Turn_on_off_yeelight
    trigger:
      platform: state
      entity_id: sensor.0x00158d0001593043_click
      to: 'single'
    action:
      service: light.toggle
      entity_id: light.yeelightw1
```

and it works fine!
Then is the automation based on mqtt that don't works

sehraf, 2019-01-23:
I tried that already, too. The downside of this is that this only works if the state actually changes. When you press the button, wait and press it again this triggered the automation only once instead of twice for me.

sehraf, 2019-01-23:
This is my config with mqtt:
```yaml
- alias: Xiaomi - TV Schrank
  hide_entity: true
  initial_state: 'on'
  trigger:
    - platform: mqtt
      topic: zigbee2mqtt/wohnzimmer.switch
  condition:
    - condition: and
      conditions:
      - condition: template
        value_template: '{{ trigger.payload_json.click is defined }}'
      - condition: template
        value_template: '{{ "single" in trigger.payload_json.click }}'
  action:
    - service: switch.toggle
      entity_id: switch.wohnzimmer_light_tv_vitrine_switch
```
The topic is `zigbee2mqtt/<friendlyName>` as configured in `data/configuration.yaml`