Koenkk/zigbee2mqtt #500: Updated from 0.13 to 0.17. Now messages in the log contains partial information.

CypherMK, 2018-10-18:
I have updated from 0.13 to 0.17. I have these devices connected:

```
2018-10-18 20:06:41 - info: Temperatuur (0x00158d0002325166): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-10-18 20:06:41 - info: Bewegingsmelder (0x00158d00024e56f8): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-10-18 20:06:41 - info: Kubus (0x00158d00027bb184): MFKZQ01LM - Xiaomi Mi smart home cube (EndDevice)
2018-10-18 20:06:41 - info: Ikea1 (0x000b57fffed92c7a): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
2018-10-18 20:06:41 - info: Ikea2 (0x90fd9ffffe73e37c): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
2018-10-18 20:06:41 - info: Ikea3 (0x000b57fffed8037e): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)

```
But since the update, they give me these (sometimes incomplete log entries, examples:

```
2018-10-18 20:51:42 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":false,"linkquality":42}'
2018-10-18 20:53:49 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":false,"linkquality":42,"illuminance":3}'
2018-10-18 20:53:49 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"illuminance":3,"linkquality":42,"occupancy":true}'
2018-10-18 20:55:19 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":false,"linkquality":42}'
2018-10-18 20:55:46 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":false,"linkquality":42,"illuminance":3}'
2018-10-18 20:55:46 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"illuminance":3,"linkquality":42,"occupancy":true}'
2018-10-18 20:57:16 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":false,"linkquality":42}'
2018-10-18 20:57:37 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":false,"linkquality":47,"illuminance":3}'
2018-10-18 20:57:37 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"illuminance":3,"linkquality":47,"occupancy":true}'
2018-10-18 20:57:42 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":true,"linkquality":47,"illuminance":3}'
2018-10-18 20:57:42 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"illuminance":3,"linkquality":47,"occupancy":true}'
2018-10-18 20:58:55 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":true,"linkquality":55,"illuminance":3}'
2018-10-18 20:58:55 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"illuminance":3,"linkquality":49,"occupancy":true}'
2018-10-18 20:59:12 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":true,"linkquality":47,"illuminance":4}'
2018-10-18 20:59:12 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"illuminance":4,"linkquality":47,"occupancy":true}'
2018-10-18 20:59:23 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"occupancy":true,"linkquality":49,"illuminance":3}'
2018-10-18 20:59:23 - info: MQTT publish, topic: 'zigbee2mqtt/Bewegingsmelder', payload: '{"illuminance":3,"linkquality":47,"occupancy":true}'
```

Or:
```
2018-10-18 20:36:04 - info: MQTT publish, topic: 'zigbee2mqtt/Temperatuur', payload: '{"pressure":1019,"linkquality":13,"battery":"42.00","voltage":2985}'
2018-10-18 20:36:04 - info: MQTT publish, topic: 'zigbee2mqtt/Temperatuur', payload: '{"battery":"42.00","voltage":2985,"linkquality":13,"temperature":21.83,"humidity":58.11,"pressure":1020.03}'
```

So sometimes it gives me a complete message, like it did in 0.13. But most of the time it is partial information. Also the battery level seems off now. It was 100% before, and no 42%.

So what is causing this issue?

rugtra, 2018-10-18:
same here:-( Partial Info...

Koenkk, 2018-10-19:
This is the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/485 (fixed in dev branch)

closed 2018-10-19 by Koenkk