Koenkk/zigbee2mqtt #502: Messages not containing all information

zerofruchtshake, 2018-10-19:
I had to reinstall everything on my raspberry due to a problem (not related to zigbee2mqtt) and now my Xiaomi Sensors do not publish all available information - see logs below. Any idea how to fix this?

`Oct 19 12:58:47 raspberrypi npm[2742]:   zigbee2mqtt:info 2018-10-19 12:58:47 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002855f41', payload: '{"state":"OFF","power":0,"voltage":221,"consumption":0,"temperature":25,"linkquality":113}'
Oct 19 12:58:49 raspberrypi npm[2742]:   zigbee2mqtt:info 2018-10-19 12:58:49 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002855f41', payload: '{"state":"ON","linkquality":99}'
Oct 19 12:58:52 raspberrypi npm[2742]:   zigbee2mqtt:info 2018-10-19 12:58:52 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002855f41', payload: '{"state":"OFF","linkquality":110}'`

`Oct 19 13:00:51 raspberrypi npm[2742]:   zigbee2mqtt:info 2018-10-19 13:00:51 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002320326', payload: '{"temperature":29.17,"linkquality":147,"humidity":53.64}'
Oct 19 13:00:51 raspberrypi npm[2742]:   zigbee2mqtt:info 2018-10-19 13:00:51 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002320326', payload: '{"humidity":53.64,"linkquality":147,"pressure":997}'`

cjsteward, 2018-10-19:
I had the same issue, I had to revert back to 1.6.

zerofruchtshake, 2018-10-19:
i'm quite new to raspberry and linux, how do i revert to an old version?

ryanbeaton, 2018-10-19:
https://github.com/Koenkk/zigbee2mqtt/issues/500

https://github.com/Koenkk/zigbee2mqtt/issues/485

zerofruchtshake, 2018-10-19:
Thank you! I tried the fix from #485 and it seems to work better but perfect - the battery information is still missing.

Koenkk, 2018-10-19:
@zerofruchtshake it takes some time before then sensor reports it's battery.

closed 2018-10-20 by zerofruchtshake