Koenkk/zigbee2mqtt #504: linkquality value range

jpmassena, 2018-10-19:
Hey guys!

Quick question: Is there a min-max range for linkquality values? I've seen 0 and 19x so I'm guessing it's not a percentage.

I'd like to have a signal strength meter in Home Assistant and for that I need the min-max values.

Thnaks for this great project!

Koenkk, 2018-10-19:
To be honest, I don't know. But I think you can figure this out easily.

Max: hold device close to CC2531
Min: check at which linkquality a device stop working. 

I'm also interested in these values, as we can use them to publish % instead of these vague linkquality vaues.

jpmassena, 2018-10-20:
The thing about the values is that they don't seem very consistent:
If I open/close the sensor touching the dongle, I get values from 195 to 215;
If I open/close it 3 feet away, it ranges from 100 to 150; etc

closed 2018-12-26 by stale[bot]

h4nc, 2019-01-01:
I have two cc2530 routers linked to my cc2531 coordinator. 

In home assistant I mostly get the linkquality value 0 for my routers. 

Current states/values in home assistant
Router1 (distance 10m, 2 walls),  0 linkquality, "connected", RSSI: 15 
Router2 (distance 20m, 3 walls), 0 linkquality, "disconnected", RSSI: 1

I'm really confused about these values. 
My aqara sensors around router 2 only work when I leave the router plugged in. When I unplug it the devices will not send the stated. This means they are connected threw the router, although I have RSSI:1, "disconnected" and zero linkquality.

Could someone explain those values?
The device must be connected to why does it show disconnected? 0 linkquality also doesn't look that good. 
For me it seems like those values are totally useless .