Koenkk/zigbee2mqtt #507: support  Osram Smart+ Motion Sensor?

salopette, 2018-10-20:
Hello, is Osram Smart + Motion Sensor supported?

https://www.osram-lamps.de/ekatalog/smart-home/smart-home-komponenten/smart-motion-sensor/index.jsp

Koenkk, 2018-10-21:
No, but can be if you have the device and follow: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

nebman, 2018-10-22:
I recently got one, but it failed while pairing. Tried several factory resets and button presses but kept getting the following error msg:

> zigbee2mqtt:error 22/10/2018, 21:13:51 Cannot get the Node Descriptor of the Device: 0x000d6f000fffffff (Error: Timed out after 10000 ms)

The sensor blinks blue (searching network) followed by green every other time (indicating successful pairing) and then keeps repeating this sequence after some seconds. 

Which DEBUG parameters would be helpful to debug?

nebman, 2018-10-23:
I created a log file with DEBUG=cc-znp, would this be helpful? @Koenkk   
Lines start when pluggin in the battery in the motion sensor and ends at the error message.

[Link to Logfile](https://gist.github.com/nebman/fbb10c1dc2f0197e9777c9a7d91a1f99)

salopette, 2018-10-23:
Okt 23 15:49:22 raspberrypi npm[602]:   zigbee2mqtt:debug 2018-10-23 15:49:22 Check online 0x00124b0014d9dafa 0x00124b0014d9dafa
Okt 23 15:49:36 raspberrypi npm[602]:   zigbee2mqtt:debug 2018-10-23 15:49:36 Saving state to file /opt/zigbee2mqtt/data/state.json
Okt 23 15:49:42 raspberrypi npm[602]:   zigbee2mqtt:error 2018-10-23 15:49:42 Cannot get the Node Descriptor of the Device: 0x000d6f000f71110c (Error: Timed out after 10000 ms)

 Saving state to file /opt/zigbee2mqtt/data/state.json

{
    "0x00124b0014d9dafa": {
        "description": "0/0x00124B0014D9D6F9",
        "type": "COORD",
        "rssi": 48,
        "linkquality": 47
    },
    "0x00158d0001bc53c4": {
        "battery": "99.00",
        "voltage": 3022,
        "linkquality": 84
    },
    "0x00158d0001f9efde": {
        "battery": "99.00",
        "voltage": 3025,
        "linkquality": 63
    }

tb-killa, 2018-10-25:
@salopette please provide full debug logfile via ``DEBUG=* npm start`` and do fresh pairing.
It could be that you have first of all remove your actually device entry from your ``database.db`` file.

for pairing please refer to this: https://images-eu.ssl-images-amazon.com/images/I/C18QWX-dFxS.pdf

``
Reset the Motion Sensor: Press and hold the marked button for longer than 10 seconds
``

it could be that you have to keep the device on wakeup by press the button e.g. every second on pairing process themselve. please test again and report.

nebman, 2018-10-27:
> it could be that you have to keep the device on wakeup by press the button e.g. every second on pairing process themselve. please test again and report.

I have tried the button-press-keep-awake method. but the device doesnt look like it goes to sleep. it starts repairing after some seconds, well before we get the 10s timeout message in the log file

dcsg, 2018-11-01:
Any news on this?

dcsg, 2018-11-01:
Was able to use `DEBUG=*` this is what I got:

```
  serialport:poller Polling for "readable" +19ms
  serialport:poller received "readable" +83ms
  serialport:bindings read +89ms
  serialport:unixRead Starting read +86ms
  serialport:unixRead Finished read 18 bytes +1ms
  serialport:main binding.read finished +93ms
  cc-znp { sof: 254,
  cc-znp   len: 13,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'endDeviceAnnceInd',
  cc-znp   payload:
  cc-znp    { srcaddr: 11649,
  cc-znp      nwkaddr: 11649,
  cc-znp      ieeeaddr: '0x000d6f000e0597b7',
  cc-znp      capabilities: 128 },
  cc-znp   fcs: 64,
  cc-znp   csum: 64 } +97ms
  serialport:main _read reading +10ms
  serialport:bindings read +14ms
  serialport:unixRead Starting read +13ms
  cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 11649, nwkaddr: 11649, ieeeaddr: '0x000d6f000e0597b7', capabilities: 128 } +6ms
  zigbee-shepherd spinlock: true [] +2ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +6ms
  serialport:poller Polling for "readable" +23ms
  serialport:poller received "readable" +4s
  serialport:bindings read +4s
  serialport:unixRead Starting read +4s
  serialport:unixRead Finished read 18 bytes +1ms
  serialport:main binding.read finished +4s
  cc-znp { sof: 254,
  cc-znp   len: 13,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'leaveInd',
  cc-znp   payload:
  cc-znp    { srcaddr: 11649,
  cc-znp      extaddr: '0x000d6f000e0597b7',
  cc-znp      request: 0,
  cc-znp      removechildren: 0,
  cc-znp      rejoin: 0 },
  cc-znp   fcs: 100,
  cc-znp   csum: 100 } +4s
  serialport:main _read reading +5ms
  serialport:bindings read +6ms
  serialport:unixRead Starting read +6ms
  cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 11649, extaddr: '0x000d6f000e0597b7', request: 0, removechildren: 0, rejoin: 0 } +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +10ms
  zigbee2mqtt:error 2018-11-1 22:46:21 Not connected to MQTT server!
  zigbee2mqtt:error 2018-11-1 22:46:26 Cannot get the Node Descriptor of the Device: 0x000d6f000e0597b7 (Error: Timed out after 10000 ms)
```

dcsg, 2018-11-05:
@Koenkk anything I can do to help?

Koenkk, 2018-11-09:
This device probably requires an enroll request (as explained here: https://github.com/zigbeer/zigbee-shepherd/issues/8). As I don't have the device, it is hard for me to work further on this.

ToTXR4Y, 2018-11-30:
What do you need to get it working? Is a ssh session to my server useful to get it working? Or a teamviewer session?

Koenkk, 2018-11-30:
@ToTXR4Y can you check if joining works in the: https://github.com/Koenkk/zigbee2mqtt/tree/osram_motion branch?

nebman, 2018-11-30:
@Koenkk looks like it's connected! 😃 

```
  zigbee2mqtt:info 30/11/2018, 19:31:44 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 30/11/2018, 19:31:44 New device with address 0x000d6f000f710e13 connected!
  zigbee2mqtt:info 30/11/2018, 19:31:44 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f000f710e13"}'
  zigbee2mqtt:warn 30/11/2018, 19:31:44 Device with modelID 'Motion Sensor-A' is not supported.
  zigbee2mqtt:warn 30/11/2018, 19:31:44 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 30/11/2018, 19:31:44 Received zigbee message of type 'devStatus' with data '"online"' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 19:31:44 Device with modelID 'Motion Sensor-A' is not supported.
  zigbee2mqtt:warn 30/11/2018, 19:31:44 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

nebman, 2018-11-30:
Tried to write a converter but I couldn't get it to trigger any kind of message...  
Also the device seems to need a repairing after software restart.

Koenkk, 2018-11-30:
Can you try with

```js
    {
        zigbeeModel: ['Motion Sensor-A'],
        model: 'Motion Sensor-A',
        vendor: 'OSRAM',
        description: 'Motion detector',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
             execute(device, actions, callback, 1000);
        },
    },
```

nebman, 2018-11-30:
awesome 😄 I am now working on the converters

```
  zigbee2mqtt:debug 30/11/2018, 20:58:55 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":38}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 20:58:55 No converter available for 'AC01353010G' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":38}'
  zigbee2mqtt:warn 30/11/2018, 20:58:55 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 30/11/2018, 20:58:55 Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":13,"zoneStatus":38,"iasCieAddr":"0x00124b0018ed3c2e","zoneId":23}}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 20:58:55 No converter available for 'AC01353010G' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":13,"zoneStatus":38,"iasCieAddr":"0x00124b0018ed3c2e","zoneId":23}}'
  zigbee2mqtt:warn 30/11/2018, 20:58:55 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

  zigbee2mqtt:debug 30/11/2018, 21:05:20 Received zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2465}}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 21:05:20 No converter available for 'AC01353010G' with cid 'msTemperatureMeasurement', type 'attReport' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2465}}'
  zigbee2mqtt:warn 30/11/2018, 21:05:20 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 30/11/2018, 21:05:20 Received zigbee message of type 'devChange' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2465}}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 21:05:20 No converter available for 'AC01353010G' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2465}}'
  zigbee2mqtt:warn 30/11/2018, 21:05:20 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

  zigbee2mqtt:error 30/11/2018, 21:08:41 Failed to configure MOTION-OSRAM 0x000d6f000f710e13
  zigbee2mqtt:debug 30/11/2018, 21:08:43 Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":13,"zoneStatus":38,"iasCieAddr":"0x00124b0018ed3c2e","zoneId":23}}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 21:08:43 No converter available for 'AC01353010G' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":13,"zoneStatus":38,"iasCieAddr":"0x00124b0018ed3c2e","zoneId":23}}'
  zigbee2mqtt:warn 30/11/2018, 21:08:43 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

  zigbee2mqtt:debug 30/11/2018, 21:11:31 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":36}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 21:11:31 No converter available for 'AC01353010G' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":36}'
  zigbee2mqtt:warn 30/11/2018, 21:11:31 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 30/11/2018, 21:12:03 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":38}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:warn 30/11/2018, 21:12:03 No converter available for 'AC01353010G' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":38}'
  zigbee2mqtt:warn 30/11/2018, 21:12:03 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

Koenkk, 2018-11-30:
Great! 😄

nebman, 2018-11-30:
```
  zigbee2mqtt:info 30/11/2018, 22:20:19 Mount the cieApp, epId: 11
  zigbee2mqtt:error 30/11/2018, 22:20:24 Failed to configure MOTION-OSRAM 0x000d6f000f710e13
  zigbee2mqtt:debug 30/11/2018, 22:20:26 Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":13,"zoneStatus":38,"iasCieAddr":"0x00124b0018ed3c2e","zoneId":23}}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
  zigbee2mqtt:info 30/11/2018, 22:20:26 MQTT publish: topic 'zigbee2mqtt/MOTION-OSRAM', payload '{"occupancy":true,"battery":"OK","linkquality":110,"temperature":23.62,"tamper":true}'
```

As of now I got this working. Still showing "Failed to configure" despite working.... 🤔   
Just wondering... why does it retain the temperature property when it should be reported from another converter?

Also: I don't know what the tampering flag does... When you press the button for longer than 2 sec it will flip to false but switches back to true immediately after releasing the button. Shouldn't it be the other way round? 😅

Koenkk, 2018-11-30:
Can you provide the log with `DEBUG=* npm start`? That will show why the configure failed..

nebman, 2018-12-01:
(So many loglines with 12 devices in this network 😅)

I could isolate it to the following lines, if you need more just let me know
```
  zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: request unsuccess: 112' +0ms
  zigbee2mqtt:error 01/12/2018, 10:43:23 Failed to configure MOTION-OSRAM 0x000d6f000f710e13
```

Koenkk, 2018-12-01:
Can you check again with the updated https://github.com/Koenkk/zigbee2mqtt/tree/osram_motion branch. Please pair the device again.

nebman, 2018-12-01:
Fantastic work @Koenkk!

```
  zigbee2mqtt:info 01/12/2018, 15:31:48 Mounted the cie app, (epId {epId})
  zigbee2mqtt:info 01/12/2018, 15:31:51 MQTT publish: topic 'zigbee2mqtt/MOTION-OSRAM', payload '{"occupancy":true,"tamper":true,"battery_low":false,"linkquality":105}'
  zigbee2mqtt:info 01/12/2018, 15:32:00 MQTT publish: topic 'zigbee2mqtt/MOTION-OSRAM', payload '{"occupancy":true,"tamper":true,"battery_low":false,"linkquality":105}'
  zigbee2mqtt:info 01/12/2018, 15:32:01 MQTT publish: topic 'zigbee2mqtt/MOTION-OSRAM', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":105}'
  zigbee2mqtt:info 01/12/2018, 15:32:02 MQTT publish: topic 'zigbee2mqtt/MOTION-OSRAM', payload '{"occupancy":true,"tamper":true,"battery_low":false,"linkquality":105}'
  zigbee2mqtt:info 01/12/2018, 15:32:03 MQTT publish: topic 'zigbee2mqtt/MOTION-OSRAM', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":105}'
  zigbee2mqtt:info 01/12/2018, 15:32:03 Succesfully configured MOTION-OSRAM 0x000d6f000f710e13
  zigbee2mqtt:info 01/12/2018, 15:32:03 MQTT publish: topic 'zigbee2mqtt/MOTION-OSRAM', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":105,"temperature":23.79}'
```

After removing and (somehow) pairing the device again, I had to restart zigbee2mqtt before receiving any messages. But now it looks very good.

Should I send you a pull request to zigbee-shepherd-converters? 

In the devices.js file I had two additional line for temperature reporting, copied those off another device at first and I'm not sure the are necessary? But they dont generate any error message either...

```
configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
            ];
            execute(device, actions, callback);
        },
```

Koenkk, 2018-12-01:
Looks good, you can keep the bind and report for `msTemperatureMeasurement`.

What does tamper/battery_low mean? How do you get to these messages? For the battery I would prefer a % (to keep it the same as other devices)

nebman, 2018-12-01:
See https://www.nxp.com/docs/en/user-guide/JN-UG-3077.pdf page 459 has a table of flags for `b16ZoneStatus` where I took the information from. `ZoneStatus `is the only property that gets transmitted. The device just sends a bit for "Battery OK" or "Battery Low", same for the "Tampered" status (which I observed just indicates a longer button press on this device?!).

ToTXR4Y, 2018-12-01:
Tamper should be an "opening" detection of the device itself.

nebman, 2018-12-01:
> Tamper should be an "opening" detection of the device itself.

Can't really see any switch or detection mechanism on the inside @ToTXR4Y

Koenkk, 2018-12-01:
Does battery reporting work with:

```js
    {
        zigbeeModel: ['Motion Sensor-A'],
        model: 'Motion Sensor-A',
        vendor: 'OSRAM',
        description: 'Motion detector',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb),
            ];
             execute(device, actions, callback, 1000);
        },
    },
```

nebman, 2018-12-01:
After trying about an hour to re-pair the motion sensor, I am now back at 
```
  zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: request unsuccess: 112' +0ms
  zigbee2mqtt:error 01/12/2018, 20:20:45 Failed to configure 0x000d6f000f710e13 0x000d6f000f710e13
```

I dont know... this thing is slowly driving me insane and I need a break now ☹️ At least messages are arriving (but no battery status), so I will post the converters and other can try their luck too 😅

nebman, 2018-12-04:
Just caught this genPowerCfg reports, but wont seem worthwhile to really add, looks like just another low battery warning?

```
Dec 04 20:35:35 rpi-home npm[19950]:   zigbee2mqtt:debug 04/12/2018, 20:35:35 Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryAlarmState":0}}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
Dec 04 20:35:35 rpi-home npm[19950]:   zigbee2mqtt:warn 04/12/2018, 20:35:35 No converter available for 'AC01353010G' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryAlarmState":0}}'
Dec 04 20:35:35 rpi-home npm[19950]:   zigbee2mqtt:warn 04/12/2018, 20:35:35 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
Dec 04 20:35:35 rpi-home npm[19950]:   zigbee2mqtt:debug 04/12/2018, 20:35:35 Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryAlarmState":0}}' of device 'Motion Sensor-A' (0x000d6f000f710e13)
Dec 04 20:35:35 rpi-home npm[19950]:   zigbee2mqtt:warn 04/12/2018, 20:35:35 No converter available for 'AC01353010G' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryAlarmState":0}}'
```

Koenkk, 2018-12-05:
Doesn't look important indeed, we can ignore this message by adding an `ignore` converter.

nebman, 2018-12-05:
I can add that and push it. But I will watch the logs a bit more just in case there are also other uncaught messages.

dcsg, 2018-12-10:
What is missing here in order to get released?

Koenkk, 2018-12-10:
It's supported in the dev branch now, will be part of the next release.

closed 2018-12-10 by Koenkk

Morphy99, 2019-01-05:
I'm using the Hass.io Add-on zigbee2mqtt-edge and I get this error still:

> Cannot get the Node Descriptor of the Device: 0x000d6f000e056cfd (Error: Timed out after 10000 ms)

johnnyletrois, 2019-01-05:
@Morphy99 it should work even with the error. Try it out.

Morphy99, 2019-01-05:
Hmm so it appears to be added but not doing anything. I also have a temp sensor in devices as well!? Powering on and off just shows 

> zigbee2mqtt:warn 2019-1-5 10:27:58 Message without device!

All I have in state attributes is

> {
  "friendly_name": "PIR Sensor",
  "device_class": "motion"
}

Can't see any mqtt messages in the log either. 

I've set an automation up with this but nothing either:

```
trigger:
  - entity_id: binary_sensor.0x000d6f000e056cfd_occupancy
    platform: state
    to: 'on'
  action:
    service: homeassistant.turn_on
    entity_id: light.spot1_light_2

```

dreimer1986, 2019-01-06:
Second the mentioned behavior in NOT edge, but most recent stable on most recent Hass.IO on firmware 24122018:
`  zigbee2mqtt:info 2019-1-6 19:28:28 MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f000e059aae/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/Motion","availability_topic":"zigbee2mqtt/bridge/state","name":"Motion_occupancy","unique_id":"0x000d6f000e059aae_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000e059aae","name":"Motion","sw_version":"Zigbee2mqtt 1.0.1","model":"SMART+ Motion Sensor (AC01353010G)","manufacturer":"OSRAM"}}'
  zigbee2mqtt:info 2019-1-6 19:28:28 MQTT publish: topic 'homeassistant/sensor/0x000d6f000e059aae/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/Motion","availability_topic":"zigbee2mqtt/bridge/state","name":"Motion_temperature","unique_id":"0x000d6f000e059aae_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000e059aae","name":"Motion","sw_version":"Zigbee2mqtt 1.0.1","model":"SMART+ Motion Sensor (AC01353010G)","manufacturer":"OSRAM"}}'
  zigbee2mqtt:error 2019-1-6 19:28:32 Failed to configure Motion 0x000d6f000e059aae`

nebman, 2019-01-08:
@Morphy99 maybe it's not paired correctly? Try repairing it. 

One note though, I _think_ the instructions given in the manual are not correct, holding the button for >10sec did nothing for me.   
I guess to fully reset the device you can try holding the button while inserting the battery. The status light will change to some kind of red or purple IIRC

Morphy99, 2019-01-08:
Ok I removed the device by sending the `zigbee2mqtt/bridge/config/remove` topic and after a while after repairing this happened:

> zigbee2mqtt:warn 2019-1-8 19:34:24 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:34:34 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:34:38 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:34:40 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:34:41 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:13 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:15 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:17 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:35 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:39 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:43 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:45 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:35:47 Message without device!
  zigbee2mqtt:warn 2019-1-8 19:36:00 Message without device!
  zigbee2mqtt:info 2019-1-8 19:36:00 Device incoming...
  zigbee2mqtt:info 2019-1-8 19:36:00 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-1-8 19:36:00 MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f000e056cfd/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x000d6f000e056cfd","availability_topic":"zigbee2mqtt/bridge/state","name":"0x000d6f000e056cfd_occupancy","unique_id":"0x000d6f000e056cfd_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000e056cfd","name":"0x000d6f000e056cfd","sw_version":"Zigbee2mqtt 1.0.1","model":"SMART+ Motion Sensor (AC01353010G)","manufacturer":"OSRAM"}}'
  zigbee2mqtt:info 2019-1-8 19:36:00 MQTT publish: topic 'homeassistant/sensor/0x000d6f000e056cfd/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x000d6f000e056cfd","availability_topic":"zigbee2mqtt/bridge/state","name":"0x000d6f000e056cfd_temperature","unique_id":"0x000d6f000e056cfd_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000d6f000e056cfd","name":"0x000d6f000e056cfd","sw_version":"Zigbee2mqtt 1.0.1","model":"SMART+ Motion Sensor (AC01353010G)","manufacturer":"OSRAM"}}'
  zigbee2mqtt:info 2019-1-8 19:36:02 MQTT publish: topic 'zigbee2mqtt/0x000d6f000e056cfd', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":60}'

dreimer1986, 2019-01-09:
Yeah, pairing the motion sensor is a real pain. Took me about 30 minutes to finally get it working. Now it runs fine and gives back motion and temperature. (And thus one thing more than before via Osram Gateway linked to Hass.IO)

Koenkk, 2019-01-26:
To all: we need some help in order to verify a change which could have impact on this device. 

Could you check when changing https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zapp/cie.js#L12 to 

```js
cieClusters.init('ssIasZone', 'dir', {value: 1});
```

this device still works? Please re-pair after making this change.

wossix, 2019-02-02:
@Koenkk,
I did some first tests. But it doesn't look good. After changing ssIasZone to 1. I have tried to join a sensor multiple times - with no success.

Received: Timed out after 10000;

Cheers,
Ben

Koenkk, 2019-02-02:
@wossix thanks for testing, when changing it back, does it work again?

wossix, 2019-02-02:
@Koenkk, yes.

allofmex, 2019-03-15:
> pairing the motion sensor is a real pain. Took me about 30 minutes to finally get it working.

@dreimer1986 How did you managed to pair? I can't make it.  Starting from fresh plugged cc2531, fresh shepherd instance (ioBroker in my case). Broker only says "Cannot get the Node Descriptor of the Device (Timed out after 10000ms)"

Did some sniffing:
```
Time | Source | Destination | Protocol |   | Info
-- | -- | -- | -- | -- | --
11:03:31,305639 | 00:0d:6f:00:0e:63:a0:b9 | 0x0000 | IEEE 802.15.4 |   | Association Request, RFD
11:03:31,506668 | 00:0d:6f:00:0e:63:a0:b9 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:31,508722 | 00:12:4b:00:19:36:77:dc | 00:0d:6f:00:0e:63:a0:b9 | IEEE 802.15.4 |   | Association Response, PAN: 0x1a62 Addr: 0x9f01
11:03:31,519884 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:31,714720 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:31,908602 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:32,108259 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:32,303200 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:32,497145 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:32,499075 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:32,501427 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:32,503212 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:32,516579 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:32,522316 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:32,527085 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:32,531532 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:40,125088 | 00:0d:6f:00:0e:63:a0:b9 | 0x0000 | IEEE 802.15.4 |   | Association Request, RFD
11:03:40,322863 | 00:0d:6f:00:0e:63:a0:b9 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:40,325542 | 00:12:4b:00:19:36:77:dc | 00:0d:6f:00:0e:63:a0:b9 | IEEE 802.15.4 |   | Association Response, PAN: 0x1a62 Addr: 0x9f01
11:03:40,342245 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:40,533247 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:40,669232 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:40,870101 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:41,063767 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:41,065671 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:41,068023 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:41,070088 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:41,107430 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:41,112218 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:41,117955 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:41,122115 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:41,258413 | 0x9f01 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:41,263054 | 0x0000 | 0x9f01 | ZigBee |   | Transport Key
11:03:41,328883 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,334531 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,348335 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,354538 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,362992 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,366576 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,370352 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,375946 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,399387 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,402507 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,408534 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,412571 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,422562 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,455791 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,463409 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,851761 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,857260 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,868297 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,873726 | 0x9f01 | Broadcast | ZigBee ZDP |   | Device Announcement, Nwk Addr: 0x9f01, Ext Addr: Ember_00:0e:63:a0:b9
11:03:41,902561 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
11:03:41,914480 | 0x9f01 | Broadcast | ZigBee ZDP |   | Match Descriptor Request, Nwk Addr: 0xfffd, Profile: 0x0104
[... cut ...]
11:03:44,352533 | 0x9f01 | 0x0000 | ZigBee ZDP |   | Active Endpoint Response, Nwk Addr: 0x9f01, Status: Success
11:03:45,138328 | 0x9f01 | 0x0000 | ZigBee |   | Leave
```
Then it starts again...
```
11:03:52,519752 | 00:0d:6f:00:0e:63:a0:b9 | 0x0000 | IEEE 802.15.4 |   | Association Request, RFD
11:03:52,718920 | 00:0d:6f:00:0e:63:a0:b9 | 0x0000 | IEEE 802.15.4 |   | Data Request
11:03:52,721532 | 00:12:4b:00:19:36:77:dc | 00:0d:6f:00:0e:63:a0:b9 | IEEE 802.15.4 |   | Association Response, PAN: 0x1a62 Addr: 0x3856
```

Anyone an idea why it does not pair?

dreimer1986, 2019-03-17:
I removed any instance to be found from .db and repaired and it ... just worked then? RePair is a mix of Pushing the button while pluggin in the battery and some fast pushing of the button, I got it by luck...

salopette, 2019-04-10:
How does it look with the sensor? supported devices are listed with.
But does not work.

{"id":21,"type":"EndDevice","ieeeAddr":"0x000d6f000f7109a2","nwkAddr":12943,"manufId":4174,"epList":[1,2],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{}},"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,1,3,1030,2821],"outClusterList":[3],"clusters":{}}},"_id":"JsDtgd6tedXIn12N"}

fademirci, 2019-04-14:
> How does it look with the sensor? supported devices are listed with.
> But does not work. [...]

Hi,
can confirm .... still not working !

Any News?

kind regards
FD

salopette, 2019-04-14:
Yesterday I completely deleted the zigbee2mqtt folder and set everything up again. And the motion detector was immediately recognized correctly

schnub69, 2019-07-17:
> Yesterday I completely deleted the zigbee2mqtt folder and set everything up again. And the motion detector was immediately recognized correctly

I got only iobroker.zigbee installed, tried to reinstall ist, deleted the zigbee-shepherd-converters folder like in the referred issue. flashed my cc2531 with the latest prod image available. still get "cannot get node descriptor" :( is ther another folder i can delete? or db files? to try this?

gabrieleantonini, 2019-07-20:
Same here. 
I managed to pair the motion sensor correctly but I can only see periodic `temperature `and `linkquality` messages. 
I'm running `zigbee2mqtt` v1.5.1 in a docker container

```

dreimer1986, 2019-07-20:
Retry again and again until it finally works. This sensor is a bitch to pair. I had to do so last week, too after I killed my SD card and lost some stuff. Result: I tinkered around... 30 minutes? I removed all entries of the sensor from db while addin inactive and then restart it. At that time the sensor was/has to be  without battery, then push the button and put the battery back in. Green LED lighting is a good sign, but as I said... took ages here to pair it properly. Seems like the sensor does strange stuff with the original gateay to be that easily paired. (But many OSRAM devices do funny things, like the lightify 4 way switch...)

gabrieleantonini, 2019-07-20:
Hi @dreimer1986 , thanks for your reply.
I've tried again a couple of times with the same following result:

```

  zigbee2mqtt:info 7/20/2019, 1:42:40 PM New device 'Motion Sensor-A' with address 0x000d6f000fcd2bda connected!

  zigbee2mqtt:info 7/20/2019, 1:42:40 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f000fcd2bda","meta":{"modelID":"Motion Sensor-A"}}'

  zigbee2mqtt:info 7/20/2019, 1:43:28 PM MQTT publish: topic 'zigbee2mqtt/0x000d6f000fcd2bda', payload '{"temperature":28.12,"linkquality":99}'

  zigbee2mqtt:warn 7/20/2019, 1:45:42 PM Failed to configure 0x000d6f000fcd2bda (0x000d6f000fcd2bda) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)

  zigbee2mqtt:warn 7/20/2019, 1:45:48 PM Failed to configure 0x000d6f000fcd2bda (0x000d6f000fcd2bda) ('Error: request unsuccess: 112') (attempt #2)
```

I'm going to try to enable more logging to try to get as much information as possile

mabi, 2019-08-29:
I have a similar issue as @gabrieleantonini :

zigbee2mqtt version 1.5.1 (commit #ac3b924)
Coordinator firmware version: '20190315'
max_stability

```
8/29/2019, 7:10:52 AM - info: Successfully reenabled joining
8/29/2019, 7:12:05 AM - info: New device 'Motion Sensor-A' with address 0x00redacted000000 connected!
8/29/2019, 7:12:06 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00redacted000000","meta":{"modelID":"Motion Sensor-A"}}'
8/29/2019, 7:12:10 AM - warn: Failed to configure 0x00redacted000000 (0x00redacted000000) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
8/29/2019, 7:12:10 AM - warn: This can be ignored if the device is working properly
8/29/2019, 7:12:19 AM - warn: Failed to configure 0x00redacted000000 (0x00redacted000000) ('Error: request unsuccess: 112') (attempt #2)
8/29/2019, 7:12:19 AM - warn: This can be ignored if the device is working properly
8/29/2019, 7:13:05 AM - warn: Failed to configure 0x00redacted000000 (0x00redacted000000) ('Error: request unsuccess: 112') (attempt #3)
8/29/2019, 7:13:05 AM - warn: This can be ignored if the device is working properly
8/29/2019, 7:13:07 AM - info: MQTT publish: topic 'zigbee2mqtt/0x00redacted000000', payload '{"temperature":24.68,"linkquality":42}'
8/29/2019, 7:13:32 AM - info: Successfully reenabled joining

```
Any suggestion?
Thank you.