Koenkk/zigbee2mqtt #508: Unable to pair Xiaomi devices through router

Jordi1990, 2018-10-20:
I'm using multiple Ikea bulbs and control outlets which act as router and CC2530 with RFX2401 as the coordinator. Since i've reached the 15 devices limit I seem to be unable to pair xiaomi aqara sensors. No pair messages appear in the log. I've had the same issue with IKEA control outlets but I managed to pair them when moving them close to a IKEA light bulb(router).

When I look at the network map most xiaomi devices are connected through routers to the coordinator, it seems like it's the pairing that doesn't work through IKEA routers.

Koenkk, 2018-10-21:
@ptvoinfo is my assumption right that xiaomi devices only connect when the router has a specific manufacturer ID?

@Jordi1990 I would recommend adding a CC2530 router https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices

ptvoinfo, 2018-10-21:
@Koenkk I think it is possible to create any firmware, where a router will retransmit only specific packets. I don't have IKEA routers to test.

Jordi1990, 2018-10-21:
@Koenkk Thanks for the tip, I'll add some of those routers to my network.

closed 2018-12-27 by stale[bot]