Koenkk/zigbee2mqtt #509: DZX-ZI01 Flat Motion Sensor

ben423423n32j14e, 2018-10-20:
Been looking for a flat Zigbee motion sensor to stick on walls for a while now. Ordered a couple of these: DZX-ZI01

[Link 1](https://www.aliexpress.com/item/free-ship-zigbee-wrieless-pir-detector-smart-homeProtection-Body-Motion-Sensors-Match-with-Gateway-APP-Control/32849282265.html)

[Link 2](https://www.aliexpress.com/item/Battery-Powered-Antitheft-Home-Alarm-Security-Phone-Controlled-Zigbee-PIR-Motion-Sensor/32948546382.html)

Spec is Zigbee HA 1.2, should be easy to add support.

ryanbeaton, 2018-10-21:
Look ok. Larger than the Xiaomi and they are only 90 degrees compared to the Xiaomi 120, because they are flat I suppose. Quite expensive too. 

First link store has low percentage at 94.7. Best to stick to 98 or above.

I found another listing that called them a ceiling motion detector so I guess that’s why I keep seeing it say to install them at 2.1m

http://www.heimantech.com/product/93.html

ryanbeaton, 2018-10-21:
Thanks for the reply. Yeah I noticed your issues. You have me confused with the replace Z-Wave though. Z-wave is better only more expensive. If you already own it why change?

ben423423n32j14e, 2018-10-21:
I get delays controlling devices (eg it could take 5 seconds for a roof light to turn on), sometimes motion sensors fail to report in when motion is detected, the battery life for some sensors is far less than ideal.

I've tried different controllers Aeotec Z-Stick Gen 2, Gen 5, Razberry, ZME UZB1, never been completely happy with the system.

I can't receive regular watts usage from all my devices without overloading the Z-Wave network due to it's tiny bandwidth. I see a lot of device timeouts despite having put 3x Aeotec Range extenders around the house.

I'm gradually replacing all Z-Wave devices with Zigbee equivalents.

ryanbeaton, 2018-10-21:
Wow that’s the complete opposite to what I usually hear. Will be very interested how you go 🙂

Koenkk, 2018-10-21:
@ben423423n32j14e good to read :smile:

ben423423n32j14e, 2018-10-22:
Also RE Z-Wave I should mention the different frequencies, I live in Australia and there is a frequency just for Australia and New Zealand. We are very limited in regards to devices / sensors available in this frequency region.

With Zigbee, if it's battery operated I can purchase any sensor I like from around the world, not to mention the price of sensors, eg a Z-Wave motion sensor vs Xiaomi Zigbee motion sensor, the Z-Wave is about 5x more expensive. Thus I am a huge fan of Zigbee increasing the range of devices I can buy.

ryanbeaton, 2018-10-22:
I feel your pain, I am in AU (Perth). A bought an Aeotec Z-Stick Gen 5 but have never bought any Z-wave devices due to cost. I bought 20 odd Xiaomi devices after seeing this project in the Home Assistant forum.

ryanbeaton, 2018-10-27:
Did you consider this one? Looks the same.

ZigBee Infrared Motion Sensor Detector for Wireless Smart Home Automation and Sec... https://www.amazon.com.au/dp/B078PL871W/ref=cm_sw_r_sms_c_api_uOh1BbH7AQYH9

ben423423n32j14e, 2018-11-16:
@Koenkk @ryanbeaton I now have these sensors.

The good news is they appear technically identical and identify with the same Zigbee name "PIR_TPV16". both sensors (branded and unbranded) identify as brand name Heiman in database.db.

One of the sensors is branded and the other is not, see photos:

[image]
[image]
[image]

They are reasonably small and in my opinion good looking sensors although I need some assistance pairing them with Zigbee2MQTT.

1) Terminal pairing console log:

`  zigbee2mqtt:debug 2018-11-17 01:25:52 Received zigbee message of type 'devInterview' with data '"0x005043c9a339b85f"'
  zigbee2mqtt:info 2018-11-17 01:25:52 Connecting with device...
  zigbee2mqtt:info 2018-11-17 01:25:52 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-11-17 01:26:00 Received zigbee message of type 'devInterview' with data '"0x005043c9a339b85f"'
  zigbee2mqtt:info 2018-11-17 01:26:00 Connecting with device...
  zigbee2mqtt:info 2018-11-17 01:26:00 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-11-17 01:26:03 Received zigbee message of type 'devInterview' with data '"0x005043c9a339b85f"'
  zigbee2mqtt:info 2018-11-17 01:26:03 Connecting with device...
  zigbee2mqtt:info 2018-11-17 01:26:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-11-17 01:26:06 Received zigbee message of type 'devInterview' with data '"0x005043c9a339b85f"'
  zigbee2mqtt:info 2018-11-17 01:26:06 Connecting with device...
  zigbee2mqtt:info 2018-11-17 01:26:06 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-11-17 01:26:07 Received zigbee message of type 'devInterview' with data '"0x005043c9a339b85f"'
  zigbee2mqtt:info 2018-11-17 01:26:07 Connecting with device...
  zigbee2mqtt:info 2018-11-17 01:26:07 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-11-17 01:26:10 Received zigbee message of type 'devInterview' with data '"0x005043c9a339b85f"'
  zigbee2mqtt:info 2018-11-17 01:26:10 Connecting with device...
  zigbee2mqtt:info 2018-11-17 01:26:10 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-11-17 01:26:10 Received zigbee message of type 'devIncoming' with data '"0x005043c9a339b85f"' of device 'PIR_TPV16' (0x005043c9a339b85f)
  zigbee2mqtt:info 2018-11-17 01:26:10 Device incoming...
  zigbee2mqtt:info 2018-11-17 01:26:10 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2018-11-17 01:26:10 New device with address 0x005043c9a339b85f connected!
  zigbee2mqtt:info 2018-11-17 01:26:10 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x005043c9a339b85f"}'
  zigbee2mqtt:warn 2018-11-17 01:26:10 Device with modelID 'PIR_TPV16' is not supported.
  zigbee2mqtt:warn 2018-11-17 01:26:10 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-11-17 01:26:10 Received zigbee message of type 'devStatus' with data '"online"' of device 'PIR_TPV16' (0x005043c9a339b85f)`

2) devices.js entry

`    // Heiman
    {
        zigbeeModel: ['PIR_TPV16'],
        model: 'HS1MS-M',
        vendor: 'Heiman',
        description: 'Smart Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
    },`

Here is the database.db entry:

`{"id":5,"type":"EndDevice","ieeeAddr":"0x005043c9a3313e8d","nwkAddr":38975,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"PIR_TPV16","epList":[1],"status":"online","joinTime":1542378488,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"PIR_TPV16","dateCode":"2017.4.7","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":0,"mainsFrequency":0,"mainsAlarmMask":0,"mainsVoltMinThres":0,"mainsVoltMaxThres":65535,"mainsVoltageDwellTripPoint":0,"batteryVoltage":30,"batteryPercentageRemaining":200,"batteryManufacturer":"","batterySize":255,"batteryAHrRating":0,"batteryQuantity":0,"batteryRatedVoltage":30,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryVoltThres1":0,"batteryVoltThres2":0,"batteryVoltThres3":0,"batteryPercentMinThres":40,"batteryPercentThres1":0,"batteryPercentThres2":0,"batteryPercentThres3":0,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":13,"zoneStatus":37,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}}}}},"_id":"kjsUi85p4N2B9dt5"}`

No data shows up in terminal when triggering motion, where to go from here?

Koenkk, 2018-11-16:
As this is an ias device it needs to be enrolled. Can you try with (may require repairing):

```js
{
    zigbeeModel: ['PIR_TPV16'],
    model: 'HS1MS-M',
    vendor: 'Heiman',
    description: 'Smart Motion Sensor',
    supports: 'occupancy',
    fromZigbee: [],
    toZigbee: [],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const device = shepherd.find(ieeeAddr, 1);
        const actions = [
            (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
            (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
            (cb) => device.functional('ssIasZone', 'enrollRsp', {
                enrollrspcode: 1,
                zoneid: 23
            }, cb),
        ];

        execute(device, actions, callback);
    },
},
```

ben423423n32j14e, 2018-11-17:
@Koenkk done, terminal output is as follows, I'm not seeing any output from triggering motion on the sensor:

`  zigbee2mqtt:debug 2018-11-17 18:41:22 Received zigbee message of type 'devLeaving' with data '"0x005043c9a3313e8d"'
  zigbee2mqtt:warn 2018-11-17 18:41:22 Message without device!
  zigbee2mqtt:debug 2018-11-17 18:42:27 Received zigbee message of type 'devInterview' with data '"0x005043c9a3313e8d"'
  zigbee2mqtt:debug 2018-11-17 18:42:34 Received zigbee message of type 'devInterview' with data '"0x005043c9a3313e8d"'
  zigbee2mqtt:debug 2018-11-17 18:42:37 Received zigbee message of type 'devInterview' with data '"0x005043c9a3313e8d"'
  zigbee2mqtt:debug 2018-11-17 18:42:39 Received zigbee message of type 'devInterview' with data '"0x005043c9a3313e8d"'
  zigbee2mqtt:debug 2018-11-17 18:42:42 Received zigbee message of type 'devInterview' with data '"0x005043c9a3313e8d"'
  zigbee2mqtt:debug 2018-11-17 18:42:44 Received zigbee message of type 'devInterview' with data '"0x005043c9a3313e8d"'
  zigbee2mqtt:debug 2018-11-17 18:42:44 Received zigbee message of type 'devIncoming' with data '"0x005043c9a3313e8d"' of device 'PIR_TPV16' (0x005043c9a3313e8d)
  zigbee2mqtt:info 2018-11-17 18:42:44 Device incoming...
  zigbee2mqtt:info 2018-11-17 18:42:44 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 2018-11-17 18:42:44 Received zigbee message of type 'devStatus' with data '"online"' of device 'PIR_TPV16' (0x005043c9a3313e8d)`

ben423423n32j14e, 2018-11-17:
@Koenkk I no longer see the failed to configure message if that helps.

Koenkk, 2018-11-17:
But do you see the successfully configured message?

ben423423n32j14e, 2018-11-17:
no

Koenkk, 2018-11-19:
Can you post your log? One of the messages should be shown.

closed 2019-01-25 by stale[bot]

whyzermans, 2019-05-10:
Guys, I have the same issue here. Anyone managed to solve this? On the back of the sensor there is the tamper button. If I press it, there is action triggered and logged, but on the motion itself there is no action triggered. So currently is useless for me :(