Koenkk/zigbee2mqtt #513: TRADFRI LED1545G12 with unicode characters in model name

ekinmeroglu, 2018-10-20:
Hi,

I've got a LED1545G12 with model name "TRADFRI bulb E27 WS�opal 980lm" (please see the char between WS and opal) and I got the famous "TRADFRI bulb E27 WS�opal 980lm' is not supported." error on the logs.
I've checked with a hex editor, the character is 0xEFBFBD which seems to translate to \uFFFD in UTF-8. I've added the model name to devices.js as 'TRADFRI bulb E27 WS\uFFFDopal 980lm' , and everything just started to work. 

Is this an encoding problem on my system, or a non-standard bulb or firmware? I'm using hass.io with zigbee2mqtt add-on v1.7. The issue is present on both 1.5 & 1.7

Thanks,

ekinmeroglu, 2018-10-21:
Tried with 1.8, the issue is still here.

Koenkk, 2018-10-21:
Could you:
- stop zigbee2mqtt
- remove all entries of the device from `database.db`
- start zigbee2mqtt
- pair again

Koenkk, 2018-10-30:
Ok then we should add it, could you provide your `database.db`?

thoj, 2018-10-31:
here is mine:

https://gist.github.com/thoj/2c42db8cdc6bdcd0bf32c46a3aa8651c

ekinmeroglu, 2018-10-31:
And mine:

https://gist.github.com/ekinmeroglu/1703b459cbbf9396b2f5340eff9af6b4

Koenkk, 2018-10-31:
Ok thanks, will be supported in the next version.

closed 2019-01-06 by stale[bot]