Koenkk/zigbee2mqtt #515: SmartThings Motion Sensor Slim Design

ben423423n32j14e, 2018-10-21:
Can't see this sensor on the supported device list, I've got one on order.

https://www.amazon.com/gp/product/B07F8ZHBLS

Koenkk, 2018-10-21:
Good! Once you receive it follow: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices 

Let me know when you need additional help

ben423423n32j14e, 2018-11-16:
@Koenkk I have this sensor and have done the following:

1) Paired it with Zigbee2MQTT, below is debug output of the pairing process:

`  zigbee2mqtt:debug 2018-11-10 15:57:21 Received zigbee message of type 'devInterview' with data '"0x286d970001017d19"'
  zigbee2mqtt:debug 2018-11-10 15:57:33 Received zigbee message of type 'devInterview' with data '"0x286d970001017d19"'
  zigbee2mqtt:debug 2018-11-10 15:57:34 Received zigbee message of type 'devInterview' with data '"0x286d970001017d19"'
  zigbee2mqtt:debug 2018-11-10 15:57:34 Received zigbee message of type 'devInterview' with data '"0x286d970001017d19"'
  zigbee2mqtt:debug 2018-11-10 15:57:35 Received zigbee message of type 'devInterview' with data '"0x286d970001017d19"'
  zigbee2mqtt:debug 2018-11-10 15:57:36 Received zigbee message of type 'devInterview' with data '"0x286d970001017d19"'
  zigbee2mqtt:debug 2018-11-10 15:57:37 Received zigbee message of type 'devInterview' with data '"0x286d970001017d19"'
  zigbee2mqtt:debug 2018-11-10 15:57:37 Received zigbee message of type 'devIncoming' with data '"0x286d970001017d19"' of device 'motion' (0x286d970001017d19)
  zigbee2mqtt:info 2018-11-10 15:57:37 Device incoming...
  zigbee2mqtt:info 2018-11-10 15:57:37 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2018-11-10 15:57:37 Device with modelID 'motion' is not supported.
  zigbee2mqtt:warn 2018-11-10 15:57:37 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-11-10 15:57:37 Received zigbee message of type 'devStatus' with data '"online"' of device 'motion' (0x286d970001017d19)`

2) Added entry to devices.js

`    {
        zigbeeModel: ['motion'],
        model: 'IM6001-MTP01',
        vendor: 'SmartThings',
        description: 'Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [],
        toZigbee: [],
    },`

After adding the entry to devices.js (and before) I get no terminal output when triggering motion on the sensor. Where to from here?

Koenkk, 2018-11-16:
As this is an ias device it needs to be enrolled. Can you try with (may require repairing):

```js
{
    zigbeeModel: ['motion'],
    model: 'IM6001-MTP01',
    vendor: 'SmartThings',
    description: 'Motion Sensor',
    supports: 'occupancy',
    fromZigbee: [],
    toZigbee: [],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const device = shepherd.find(ieeeAddr, 1);
        const actions = [
            (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
            (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
            (cb) => device.functional('ssIasZone', 'enrollRsp', {
                enrollrspcode: 1,
                zoneid: 23
            }, cb),
        ];

        execute(device, actions, callback);
    },
},
```

closed 2019-01-23 by stale[bot]