Koenkk/zigbee2mqtt #516: CREE Connected bulbs - device undefined

r1cht3r, 2018-10-21:
Not able to add support for CREE bulbs due to `device 'undefined'`. Here's the debug log from the attempt to join: https://www.hastebin.com/raw/ubodefesam

If I restart the service, the bulb shows up as `unkown - undefined unknown (Router)`, log: https://www.hastebin.com/raw/isitaxiyed

- I am running latest version of zigbee2mqtt from addon hassio-zigbee2mqtt
- Running latest firmware version for CC2531 from https://github.com/Koenkk/Z-Stack-firmware
- Am able to add Philips Hue dimmer/bulbs
- I've tried this with multiple CREE bulbs, I have quite a few, all respond the same
- Removing/re-adding had same result

Would love to get these working as they are the majority of my zigbee devices, will gladly provide any additional info. Thanks!

Koenkk, 2018-10-21:
This happens when the device isn't paired correctly, how did you pair them?

r1cht3r, 2018-10-22:
Running `zigbee2mqtt/bridge/networkmap` with `graphviz` as payload, I do see the CREE bulb online:
`"0xe20db9fffe010860" [label="{0xe20db9fffe010860|Router|CREE                            |online}"];`

In spite of this, I never get any `modelID` to add support with.

Koenkk, 2018-10-22:
Somehow it doens't pair properly, can you provide the log when running with `DEBUG=* npm start`.

r1cht3r, 2018-10-22:
Here is the debug log from running `DEBUG=* npm start`: https://www.hastebin.com/raw/ponojusoju

Koenkk, 2018-10-23:
@kirovilya could it be possible that the modelID is not parsed correctly from the `AF:incomingMsg`?

kirovilya, 2018-10-23:
from log:
`<Buffer 18 01 01 04 00 00 42 20 43 52 45 45 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 07 00 00 30 01> }`
It is manuf name (attr 4 https://github.com/zigbeer/zigbee-shepherd/blob/b536bcdc4fe89b6433c58a8ba09672c88b0002c6/lib/components/event_handlers.js#L123)
18 01 01 - header
04 - attr manufName
00 00 - status
42 - data type (string)
20 - length (32)
.... - "CREE"

`<Buffer 18 04 01 05 00 00 42 20 43 6f 6e 6e 65 63 74 65 64 20 41 2d 31 39 20 36 30 57 20 45 71 75 69 76 61 6c 65 6e 74 20 20 20 06 00 00 42 10 32 30 31 35 30 ... > },`
It is model name (attr 5 https://github.com/zigbeer/zigbee-shepherd/blob/b536bcdc4fe89b6433c58a8ba09672c88b0002c6/lib/components/event_handlers.js#L124)
18 04 01 - header
05 - attr modelId
00 00 - status
42 - data type (string)
20 - length (32)
.... - "Connected A-19 60W Equivalent"

But why not parsed ... no ideas yet

kirovilya, 2018-10-23:
In our code, we expect the response to the attribute request to come as one. https://github.com/zigbeer/zigbee-shepherd/blob/b536bcdc4fe89b6433c58a8ba09672c88b0002c6/lib/components/event_handlers.js#L155
But in the log I see that there are several answers coming and that a message is displayed in the log without a product:
`zigbee-shepherd Identified Device: { manufacturer: CREE                            , product: undefined }`

dnaphreak, 2018-11-07:
> @HalleyAssist build own logic to retry the process:
> https://github.com/HalleyAssist/zigbee-shepherd/blob/master/lib/components/controller.js#L890
@tb-killa How would I use this to update my zigbee2mqtt?

DJTerentjev, 2018-12-10:
I have same issue with GLEDOPTO GL-S-007Z (bulb with GU10 i found).  Bulb paired, but  Device with modelID 'undefined' is not supported.  
I am running latest version of zigbee2mqtt on Hassbian and latest CC2531 firmware.
Xiaomi ( WXKG01LM) working fine. 

Wish to make it works, it is only one zigbee bulb with GU10 i found.
Thank you!

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2664240/log.txt)

eddeandres, 2019-01-22:
Just in case this applies to anyone else I had the same message appearing when trying to pair Xiaomi sensors 'Device with modelID 'undefined' is not supported'. I have a load of Hive light bulbs and Xiaomi sensors. If I paired the Hive lightbulbs first so they were acting as routers I couldn't pair the Xiaomi sensors (door/window and occupancy) however, the temperature/humidity sensors did pair. As soon as I removed the hive bulbs from the setup all of the Xiaomi sensors paired successfully.

closed 2019-03-30 by stale[bot]

dpicts, 2019-08-04:
was this ever fixed?  I'm getting the same issue when I pair the cree bulbs... it gives this message:
`zigbee2mqtt:warn 8/3/2019, 7:45:52 PM Device with modelID 'Connected A-19 60W Equivalent   ' is not supported.
  zigbee2mqtt:warn 8/3/2019, 7:45:52 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Is there anyway to get this device added? (I see that ZHA supports it)

Koenkk, 2019-08-04:
@dpicts looks like you have managed to pair it correctly, I expect that it will work with the configuration below:

```yaml
{
    zigbeeModel: ['Connected A-19 60W Equivalent '],
    model: 'Connected A-19 60W Equivalent ',
    vendor: 'CREE',
    description: 'Connected bulb',
    extend: generic.light_onoff_brightness,
},
```

(https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)

r1cht3r, 2019-08-05:
@dpicts @Koenkk I am the original issue creator and seeing this fresh activity, I tried again with a newer CC2531 firmware/fresh zigbee2mqtt checkout with a CREE bulb and I was able to get one paired correctly. I had to re-run the `2 seconds off, 1 second off, 4 times` pairing method a few times and it stopped showing  `modelID 'undefined'`. I had a slightly different model string with more spaces at the end, fairly old CREE Connected bulbs,  so here's the a slightly modified version of Koenkk's config I used:

```
    {
        zigbeeModel: ['Connected A-19 60W Equivalent   '],
        model: 'Connected A-19 60W Equivalent   ',
        vendor: 'CREE',
        description: 'Connected bulb',
        extend: generic.light_onoff_brightness,
    },
```
Won't be surprised if I find variation in the others I have, but glad to see it's working! I would like to get away from ZHA in homeassistant.