Koenkk/zigbee2mqtt #517: Error while starting zigbee-shepherd

ThiemeNL, 2018-10-21:
Hi,

I'm stuck on starting the bridge. Hope somebody can help.

Version information:

- Raspbian GNU/Linux 9 (stretch)
- npm 6.4.1
- node v8.12.0
- latest: CC2531ZNP-Prod.hex
- zigbee2mqtt version 0.1.8 (commit #97cc56b)

```
pi@raspberrypi:/opt/zigbee2mqtt $ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Oct 21 18:54 /dev/ttyACM0
```

```
pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=* npm start

> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-10-21 19:07:56 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-10-21.19-07-56'
  zigbee2mqtt:debug 2018-10-21 19:07:56 Removing old log directory '/opt/zigbee2mqtt/data/log/2018-10-20.14-06-48'
  serialport:binding:auto-detect loading LinuxBinding +0ms
  zigbee2mqtt:debug 2018-10-21 19:07:57 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2018-10-21 19:07:57 Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 2018-10-21 19:07:57 Starting zigbee2mqtt version 0.1.8 (commit #97cc56b)
  zigbee2mqtt:info 2018-10-21 19:07:57 Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  serialport:main opening path: /dev/ttyACM0 +0ms
  serialport:bindings open +0ms
  serialport:main _read queueing _read for after open +10ms
  serialport:poller Creating poller +0ms
  serialport:main opened path: /dev/ttyACM0 +21ms
  cc-znp The serialport /dev/ttyACM0 is opened. +67ms
  serialport:main _read reading +6ms
  serialport:bindings read +37ms
  serialport:unixRead Starting read +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +14ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +11ms
  serialport:main _write 8 bytes of data +51ms
  serialport:bindings write 8 bytes +51ms
  serialport:unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +59ms
  serialport:poller Polling for "readable" +73ms
  serialport:unixWrite write returned null 8 +11ms
  serialport:unixWrite wrote 8 bytes +0ms
  serialport:unixWrite Finished writing 8 bytes +1ms
  serialport:main binding.write write finished +19ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +2ms
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +14ms
  zigbee2mqtt:info 2018-10-21 19:08:00 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  serialport:main #flush +3s
  serialport:bindings flush +3s
  serialport:main binding.flush finished +6ms
  serialport:main #close +2ms
  serialport:bindings close +8ms
  serialport:poller Stopping poller +3s
  serialport:poller Destroying poller +2ms
  serialport:main binding.read error { Error: Canceled
    at Poller.emitCanceled (/opt/zigbee2mqtt/node_modules/serialport/lib/bindings/poller.js:102:17)
    at Poller.stop (/opt/zigbee2mqtt/node_modules/serialport/lib/bindings/poller.js:92:10)
    at close.then (/opt/zigbee2mqtt/node_modules/serialport/lib/bindings/linux.js:49:21)
    at <anonymous> canceled: true } +8ms
  serialport:main _read queueing _read for after open +2ms
  serialport:main binding.close finished +2ms
  cc-znp The serialport /dev/ttyACM0 is closed. +35ms
  zigbee2mqtt:info 2018-10-21 19:09:00 Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +1m
  serialport:main opening path: /dev/ttyACM0 +1m
  serialport:bindings open +1m
  serialport:main _read queueing _read for after open +2ms
  serialport:poller Creating poller +1m
  serialport:main opened path: /dev/ttyACM0 +5s
  cc-znp The serialport /dev/ttyACM0 is opened. +5s
  serialport:main _read reading +2ms
  serialport:bindings read +5s
  serialport:unixRead Starting read +1m
  zigbee-shepherd:request REQ --> SYS:osalNvRead +5ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +2ms
  serialport:main _write 8 bytes of data +10ms
  serialport:bindings write 8 bytes +10ms
  serialport:unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +1m
  serialport:unixRead waiting for readable because of code: EAGAIN +12ms
  serialport:poller Polling for "readable" +18ms
  serialport:unixWrite write returned null 8 +3ms
  serialport:unixWrite wrote 8 bytes +1ms
  serialport:unixWrite Finished writing 8 bytes +0ms
  serialport:main binding.write write finished +8ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +0ms
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +4ms
  zigbee2mqtt:error 2018-10-21 19:09:08 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-10-21 19:09:08 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-10-21 19:09:09 Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
^C  zigbee2mqtt:debug 2018-10-21 19:09:51 Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:error 2018-10-21 19:09:51 Not connected to MQTT server!
  zigbee2mqtt:error 2018-10-21 19:09:51 Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
  zigbee-shepherd zigbee-shepherd is stopping. +43s
  zigbee-shepherd zigbee-shepherd is stopped. +4ms
  zigbee2mqtt:info 2018-10-21 19:09:51 zigbee-shepherd stopped
```

When I plugin the CC2531 into the USB a green light is on. But after running the start command, the light is off.

Koenkk, 2018-10-21:
Can you try replugging the stick, pressing the button on the stick which is closest to the USB, green light should turn off; now try starting zigbee2mqtt.

ThiemeNL, 2018-10-21:
Thanks for your help. I re-plugged the stick and after pressing the button the green light was off. But still I have the same issues starting zigbee2mqtt: https://pastebin.com/raw/4vnaYR2P

ThiemeNL, 2018-10-21:
```
pi@raspberrypi:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Oct 21 21:23 usb-RFXCOM_RFXtrx433_A11Y8WZZ-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root 13 Oct 21 21:11 usb-Texas_Instruments_TI_CC2531_USB_CDC___0XD0B5C2ED5B7DFFFF-if00 -> ../../ttyACM0
```

Koenkk, 2018-10-22:
looks good, does it work when starting with `sudo npm start`?

ThiemeNL, 2018-10-22:
Same issue: https://pastebin.com/dfPGgY1c
I re-plugged the stick, then pressed the reset button. Than I ran the sudo npm start.

Should I try an older version? And which version should you advise?

ThiemeNL, 2018-10-22:
Hi Koenkk, that would be difficult. 
I think I'm not the only one running it on a raspberry with stretch.
Is it possible to run the script on a Windows machine?

Stephan296, 2018-10-22:
I have the same problem in Hassio after a update from the Hassio zigbee2mqtt plugin at Ubuntu 18.04.

lolorc, 2018-10-22:
i'm running zigbee2mqtt with raspbian/stretch, it works ok with both cc2531 and cc2530.
is it possible something else is using or trying to use /dev/ttyACM0 ?
lsof /dev/ttyACM0 ?

ThiemeNL, 2018-10-22:
@lolorc that sounds great. Which version are you using commit #97cc56b? 
I not aware of any other process that is using the /dev/ttyACM0

pedroOliveiraptc, 2019-01-21:
Hello , I am running the same problem :/ 
Someone find the problem and the solution ?

pedroOliveiraptc, 2019-01-21:
@ThiemeNL  
Hello , I am running the same problem :/
Someone find the problem and the solution ?

ThiemeNL, 2019-01-25:
hi @pedroOliveiraptc I still have no solution.

I'm planning to migrate to a newer Raspberry Pi, since I'm now using an old Raspberry Pi Model 2B.
Which Raspberry version do you have?

Ulli2k, 2019-01-25:
same issue on nuc with ubuntu server and on notebook with ubuntu desktop

curryhoden, 2019-02-10:
Same issue here on Rasp v2 
installed on iobroker but the log shows,
we got a timeout on zigbee_shepard

deisi, 2019-02-16:
what does the error actually mean? I had it before, and could fix it with re-plugging, but now its there since 5 or six attempts.

sanderdatema, 2019-02-22:
And since a few days the same here as well, happening after a few hours running on two different RPi3s and two different sticks. Only a reboot works (for another few hours).

Tried latest `master` and `dev` branch.

deisi, 2019-02-22:
So I think I know what the error message means. It means that z2m cant talk to your stick. So its essentially a communication isse and thus means:
- Stick is not connected/wrong path (al though there is an additional warning for this case)
- Stick is broken
- Stick responds with garbage
- No permission to talk to stick
- Kernel drivers not loaded correctly > Stick responds with gibberish

In any case it is a problem with the stick and not z2m. So there are couple of things you can try:

- Check permissions. Can the user you use to run z2m communicate with the stick
- Unplug and replug the stick (check `dmesg` if there is something happening)
- Reboot 
  Helps with kernel related issues
- Reflash the stick

I would suggest to follow the steps in this order. As they go go from potentially harmless to beeing more dangerous.
In particular the refhlashing of the stick needs some planning. I know there is a firmware that you can define the `pan_id` with and keep it there, such that you don't need to repair all your network on a re flash. However if you don't checkout the right firmware version, this will happen.

sanderdatema, 2019-02-22:
> Check permissions. Can the user you use to run z2m communicate with the stick
It works fine for the first few hours and then it freezes somehow. Nothing in the logs. When I restart Zigbee2MQTT, I then get the error.

> Unplug and replug the stick (check `dmesg` if there is something happening)
Seeing quite a few `cdc_acm 1-1.2:1.0: failed to set dtr/rts` errors, on both RPis.

> Reboot Helps with kernel related issues
Reboot always helps, but after a few hours the error is back.

> Reflash the stick
I will try that and return with results.

deisi, 2019-02-22:
Okay, how about electric issues? Is your stick connected to the USB port of the raspberry directly? If so, do you have some usb extension cable? Basically the first thing that is written here:
https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

sanderdatema, 2019-02-22:
TBH, no, no extension cable. But I don't see how that would make the CC2531 fail, the extension cable is all about range.

deisi, 2019-02-22:
Well, the thing is, that the 2.4GHz module kind of backfires some signals between usb of your pi and the stick. 
I don't know the details, but it is an electrical issue. So I can imagine, that it might also make the stick send arbitrary signals to your pi that then in turn lead to a crashing of either the stick or something on the pi.

Koenkk, 2019-02-22:
@sanderdatema do you have `modemmanager` installed? (see #805)

sanderdatema, 2019-02-22:
> @sanderdatema do you have `modemmanager` installed? (see #805)

No, I don't.

I just flashed the sticks a couple of times and used a cable between it and the RPi. It went fine for a while, devices pairing, etc. Until it froze again, no error. But when I restarted Zigbee2MQTT I got the error again.

I can't for the life of me find out what is causing this. :(

nob0dy80, 2019-02-25:
same problem here..went to bed with a working zigbee network and waking up with:

`

zigbee.0 | 2019-02-25 07:17:57.711 | info | Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
-- | -- | -- | --
zigbee.0 | 2019-02-25 07:17:49.617 | info | Queue is: true
zigbee.0 | 2019-02-25 07:17:49.617 | info | Start on port: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0019366835-if00 with panID 6754 channel 11
zigbee.0 | 2019-02-25 07:17:49.617 | info | starting. Version 0.9.0 in /opt/iobroker/node_modules/iobroker.zigbee, node: v8.15.0

`

Koenkk, 2019-02-25:
@sanderdatema so your problem was fixed? Which other process was hijacking your usb?

deisi, 2019-02-25:
@nob0dy80 you went through all the trouble shooting already?

weegum60, 2019-03-11:
Hi guys, 
By any chance, did someone find any fix for this issue ?
thank you

nob0dy80, 2019-03-12:
@deisi 
of cause. and as you see there are tons of people having trouble. the guide helps in this case zero.
and as you mentioned last time, it is NOT a electrical issue.
the power is aboute 15mA when transmitting. that makes the outgoing wattage so low, i would say this i nearly impossible. you have this issue with a usb extension (which will increase the distance to the rpi) with same effect.
Things i tried:
- 3 different CC2530
- 1 CC2531
- different connection methods for CC2530 (USB TTL and directly over RPI UART)
- different USB Ports on RPI
- all firmwares on all CC2530/CC2531 available down to 20190109 up to latest 20190223

At least 20190109 worked for 3 days. 20190223 crashes the most time before i'm ready with pairing my devices.

weegum60, 2019-03-12:
Hi,
yesterday night, I flashed the stick. I'm now able to start npm. I don't know how long it will work.
The point now is the temp sensor Aqara are not updating. Still investigating...

weegum60, 2019-03-12:
One hour ago come back at initial state : 

 zigbee2mqtt:info 3/12/2019, 12:32:32 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-12.12-32-32'
  zigbee2mqtt:info 3/12/2019, 12:32:33 PM Starting zigbee2mqtt version 1.1.1 (commit #92d88b6)
  zigbee2mqtt:info 3/12/2019, 12:32:33 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/12/2019, 12:32:33 PM Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 3/12/2019, 12:33:33 PM Starting zigbee-shepherd
  zigbee2mqtt:error 3/12/2019, 12:33:33 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 3/12/2019, 12:33:33 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 3/12/2019, 12:33:33 PM Failed to start
	{"message":"Error Resource temporarily unavailable Cannot lock port","stack":"Error: Error Resource temporarily unavailable Cannot lock port"}

deisi, 2019-03-12:
imo this is a problem with your setup. I don't think there is much we can do. If the stick is not communicating with the kernel, it is not communicating with the kernel.

Maybe the stick has some sort of damage, electrical grid is unstable, usb port on the pc is damaged you name it, I have no idea but as far as I can see we cant do anything about it from here.

nob0dy80, 2019-03-12:
@deisi 
no, no and no.. you should read before answering!
i got 3x cc2530 and 1x cc2531 ..all have the same behavior.
usb voltage is messured as stable (but good idea) ..got a tester for this.

Koenkk, 2019-03-12:
As @deisi mentions, this indeed seem to be problem with the system/setup, a solution e.g. https://github.com/Koenkk/zigbee2mqtt/issues/805.

@nob0dy80 did you also try with a different system? It seems that in your case the stick isn't the problem.

nob0dy80, 2019-03-12:
@Koenkk 
no i didnt, cause i dont have tons of raspberrys at home. :-) But, if you take it logically, it cant be the host.
Yeah ok, maybe there is something with the usb ports, but why is it happening with a cc2530 over uart too (and i got three of it!)
Thought about power fluctuations, but no..absolute stable (messured).

After one month of testing.. 100 reflashes and testing 3 cc2530 and 1 cc2531, i think i'm getting close to the issue many people have.
Let me tell you my thoughts:
It seems like, if the coordinator is connected to an instable router that sometimes isnt reachable (link quality 0-1), the adapter gets stuck instead of rerouting when the commands dont reach the device.
Thats the reason why i asked for this (which is still unanswered by anybody):
https://github.com/Koenkk/Z-Stack-firmware/issues/52

Now, with the same setup, after i placed the coordinator in the middle of my house where every router got at least a quality of 10, i had no further issues till now.
Yeah, maybe a early shoutout but after the stick was often "killed" after an hour (wasnt finished with pairing yet), this is the best shot i got.
If this setup is running in a weak, i'm pretty sure.

i'm using 20190119 cause your new firmware is working on a cc2530 more worse.. that was killed sometimes after minutes.
I see you took out many of the "large network optimizing options" in your patch file, may i ask you why?

Could you have a look the XDATA thing on
http://processors.wiki.ti.com/index.php/Zigbee_Known_Issues_and_Proposed_Fixes (search 0x400)
could this be ANY related?

nob0dy80, 2019-03-12:
@Koenkk 
> EDIT: @nob0dy80 I see you deleted your comment in which you responded to @deisi in, what looked to me, a 'blaming' way. Please stay nice to each other.
Sorry for this! I try always to be, but sometimes...you know..you get aggressive ... cant tell you what time i spend with this thing. But yeah, you are right! Always be nice.

Koenkk, 2019-03-13:
@nob0dy80 interesting findings!

- I found out that some of the default options are good already
- The `XDATA` of our firmwares is even bigger (`0x4D0`)

nob0dy80, 2019-03-13:
@Koenkk 
i'm really interested in fixing this issue with you together, cause many people seem to be affected when i see the issues. Didnt figure out if the most people facing the issue are iobroker users (would be an interisting fact to).

even i'm working with iobroker and not zigbee2mqtt , this is the most valued adapter to me.
your work is really appreciated on this (just to say!) and for sure, you cant hear it any more ... crash crash crash.

Just for my interest.. is there a parameter to specify what the minimum link quality must be before he is trying an other route ?

Koenkk, 2019-03-14:
@nob0dy80 not for the linkquality, but after +- 45 second of unreachable it should find a node.

Note that there could also be stability differences between zigbee2mqtt and iobroker, for users reporting here, please indicate what you are using.

weegum60, 2019-03-14:
Hi guys,
Very interesting topics. It's true I have a device located in the basement with a link quality at 5. 
I will tomorrow delete it from the database (I will have to find how too, I'm not an expert at all :))
Just for you know, I'm using mosquito broker.

Thank you very much for your help guys

yozik04, 2019-03-15:
I have the same problem.
CC2530+CC2591 USB stick flashed as coordinator with 
[CC2530_CC2591ZNP-Prod_20190223](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591/bin/CC2530_CC2591ZNP-Prod_20190223.zip)

No matter what I have tried zigbee2mqtt was not able to talk to the stick. `Error: request timeout at CcZnp`

```
Ubuntu 18.04.2 LTS
```

[Full log](https://pastebin.com/5mBkBGTp)

Initially tried with docker but got the same result on bare metal. Log is from bare metal.

After that tried ioBroker. That connected fine without any problems. Successfully paired a device.

Koenkk, 2019-03-15:
@yozik04 can you post your configuration.yaml?

yozik04, 2019-03-15:
```
$ cat data/configuration.yaml
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyUSB0
```

Koenkk, 2019-03-15:
@yozik04 for the serial section, use:

```yaml
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyUSB0
    rtscts: false
```
(http://www.zigbee2mqtt.io/information/cc_sniffer_devices.html#required-configuration-for-cc2530-coordinator)

yozik04, 2019-03-15:
@Koenkk 

👍 
This is what I expected. RTFM 😅 

This version works:
```
# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyUSB0
advanced:
  rtscts: false
```

Great thanks! My issue solved.

jlo88, 2019-03-16:
Hey guys, 
I'm also facing the same issue. After a while I cannot get a connection to the stick anymore. I reflashed it a few times already and after a few hours it keeps crashing with: 

I use a CC2531 as a coordinater, a bunch of ikea lights as routers, ikea remotes and some xiaomi door/wndow sensors. About 30 device in total 
I'm running Ubuntu 18.10 with Hassio. 
I'm using an extension cable of a few meters and my stick is far away from the machine and the router. 

This is my configuration.yaml: 
advanced:
  log_level: debug
  report: true
devices: devices.yaml
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  password: my_password
  server: 'mqtt://homeassistant'
  user: my_user
permit_join: false
serial:
  port: /dev/ttyACM0

I'm pretty sure last time it crashed this happened when I took some router devices offline. As in switched the power off on those devices. 

I'd like to get this issue sorted out or at least know what I can do to prevent it. Is there any information that I can provide you guys to help or can you maybe give me some tips on where to look?

weegum60, 2019-03-25:
Quick feedback : 

I removed the Xiaomi device located in the basement, where there was a poor signal.
I let the two other devices, and it works now for 10 days without any issue.

It seems a poor signal can be a rootcause of this issue.

Thank you

jirisemecky, 2019-03-30:
I also see this problem on Raspberry pi 1 with docker. I was able to start it before on Ubuntu/intel64 with docker.

closed 2019-06-05 by stale[bot]

tFREYERMUTH, 2019-08-27:
Still having the following errors when I start npm :

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/ziee'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-08-27T12_25_26_382Z-debug.log

ndfred, 2020-07-18:
I had this issue running zigbee2mqtt with a user created with `useradd -r` so it has no shell and no home directory, I solved it by running `sudo usermod -a -G dialout zigbee2mqtt`