Koenkk/zigbee2mqtt #518: zigbee2mqtt/<DEVICE_ID>/get ?

HannesDi, 2018-10-21:
after searching for nearly the complete day i must ask the question:

is there anything like: "zigbee2mqtt/<DEVICE_ID>/get" ?

as counterpart for: "zigbee2mqtt/<DEVICE_ID>/set"

probally i think to complicated but i want you pull the status of a device 

i am using node-red and have set a "mqtt-in node" and topic a single bulb 
with the hope it gives me a lamp status but, nope....

sure it gives me all data when i send a external trigger with homeassistant dashboard like on or off
but i dont want to switch it off and on again to get values like brightness and color etc...

is there any mqtt command i can send to a bulb to get all data without change the lamp status?

at this point it's also not possible to use node-red dashboard with light switches because they do not changing there state cause missing a feedback from mqtt

Koenkk, 2018-10-22:
This is a feature which definitely should be added.

ryanbeaton, 2018-10-23:
Isn't this a dup of https://github.com/Koenkk/zigbee2mqtt/issues/13 ?

sebastianheierhoff, 2018-10-30:
I think, this should already be possible (though ugly) using foundation read (see: https://github.com/Koenkk/zigbee2mqtt/pull/355 and https://github.com/zigbeer/zcl-packet/wiki/6.-Appendix#FoundCmdTbl). Basically you would need a converter with type: 'foundation' and cmd: 'read' in toZigbee.js. This would trigger the device to send the current value of an attribute. From a design perspective, I don't think we need two MQTT topics, the name 'set' is confusing though.

Koenkk, 2018-11-06:
This has been implemented. 

E.g. to read the brightness:
`zigbee2mqtt/DEVICE_ID/get` `{"brightness": ""}`.

Koenkk, 2018-11-07:
@dlashua I know, this is a partial implementation, further changes need to be made in the zigbee-shepherd library to get this fully working.

HannesDi, 2018-11-11:
@dlashua : you are right its not what i need
@Koenkk : i am happy to see you have worked on this, I hope you or another developer also will soon have some spare time to take a look at the zigbee-shepherd library and expand the function

I know very well that we users have no claim to the installation of features or extensions, so I'm glad when somebody working on it and when it takes time, it just takes time. Last but not least, its the result that makes me and other users happy.

sebastianheierhoff, 2018-11-16:
@Koenkk: Should this already be usable? I have added a xiaomi_contact converter to my toZigbee.js:    

```
xiaomi_contact: {
        key: 'contact',
        convert: (value, message, type) => {
            const cid = 'genOnOff';
            const attrId = 'onOff';
            return {
                cid: cid,
                cmd: 'read',
                cmdType: 'foundation',
                zclData: [{attrId: zclId.attr(cid, attrId).value}],
                cfg: cfg.default,
            };
        },
    }
```
Using it via: 
`mosquitto_pub -t 'zigbee2mqtt/0x00158d0001fd55a7/get' -m '{"contact":""}'
`
gives:
`2018-11-16 16:51:16 Zigbee publish to '0x00158d0001fd55a7', genOnOff - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 240. MAC transaction expired.
`
I am running into the same issues using read with the ecozy thermostats. Any clues?

Koenkk, 2018-11-16:
The device is probably sleeping, it should work if you execute it directly after pressing the pairing button

sebastianheierhoff, 2018-11-16:
I don't think that's the problem. The device sends status updates (devChange) right before I try to read the attribute. Any other ideas that I could test?

Koenkk, 2018-11-16:
It could be that this device does not support read commands

Koenkk, 2018-11-25:
@poki123 if the status didn't change, it won't send an update. Probably it's already in sync.

sebastianheierhoff, 2018-11-28:
> @poki123 if the status didn't change, it won't send an update. Probably it's already in sync.

Why is this the default behaviour? I'd assume that if I actively send a message to the 'get' topic, I'll expect a resonse.

sebastianheierhoff, 2018-11-28:
@Koenkk: Any idea when you will have time to do this or could you maybe sketch what needs to be done so someone else could take over?

sebastianheierhoff, 2018-11-28:
```
zigbee2mqtt:debug 2018-11-28 14:59:39 Received zigbee message of type 'attReport' with data '{"cid":"hvacThermostat","data":{"localTemp":1859}}' of device 'Thermostat' (0x70b3d5de00000f1e)
/home/pi/Desktop/Projekte/zigbee2mqtt/lib/extension/deviceReceive.js:56
                return c.cid === cid && c.type === message.type;
                         ^

TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/home/pi/Desktop/Projekte/zigbee2mqtt/lib/extension/deviceReceive.js:56:26)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/home/pi/Desktop/Projekte/zigbee2mqtt/lib/extension/deviceReceive.js:54:52)
    at extensions.filter.forEach (/home/pi/Desktop/Projekte/zigbee2mqtt/lib/controller.js:111:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/home/pi/Desktop/Projekte/zigbee2mqtt/lib/controller.js:111:14)
    at Zigbee.onMessage (/home/pi/Desktop/Projekte/zigbee2mqtt/lib/zigbee.js:155:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/home/pi/Desktop/Projekte/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:159:14)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-11-28T13_59_39_611Z-debug.log
```
Any idea what this might be? I am running the current dev of zigbee2mqtt, sending a read, which seems to be answered correctly by the device.

Koenkk, 2018-12-01:
@sebastianheierhoff is this related to this issue? otherwise can you open a new issue an provide the log there.

sebastianheierhoff, 2018-12-04:
@Koenkk, I am running my ecozy PR (https://github.com/Koenkk/zigbee-shepherd-converters/pull/132), but had the same issues with other devices. I will resolve your review comments for the PR, so that others can test, as well.

rachetfoot, 2018-12-16:
I'm testing this exact change in my fork. I will be happy to make a PR if you can use it.

https://github.com/rachetfoot/zigbee2mqtt/commit/c798ecf94e931d6c627b06ea368d2e6fe7628543

trimethia, 2018-12-20:
@koenkk 
I was hoping to periodically poll the state of my ikea tradfri bulb with this (in order to detect when the bulb gets disconnected/connected after wall switch presses), but publishing to zigbee2mqtt/device_id/get with {"payload": ""} does not trigger any response. Do you know why?

Koenkk, 2018-12-20:
@trimethia only state changes are send. For you use case #534 is more applicable.

rachetfoot, 2018-12-20:
@Koenkk I think it would be useful to return the cached state when /get is published to. This could be enabled/disabled as part of the device configuration, where for example Ikea bulbs do not respond to read, we know we should send cached state instead. 

This should make it easier to synchronize state. Combined with #534, /get would return {state: offline}, or whatever the proper terminology is.

HannesDi, 2019-02-18:
So any news about this? is there a timeline when its planned to make ist useable ?

Koenkk, 2019-02-19:
@HannesDi not yet, but I'm still interested in this feature.

Koenkk, 2019-02-28:
Fully usable in the dev branch now! Device always responds now after a read.

closed 2019-02-28 by Koenkk

snakuzzo, 2019-07-25:
I'm trying to get "contact" of my aqara MCCGQ11LM (door and windows sensors).
Thi is sent by sensor:

```
zigbee2mqtt/0x00158d0002ec0ca0 {"contact":true,"linkquality":94,"battery":100,"voltage":3035}
```

I run this command to get contact

```
mosquitto_pub -t 'zigbee2mqtt/0x00158d0002bfc307/get' -m '{"contact":""}'`
```

but I receive this message...

```
Jul 25 12:39:00 myhost npm[532]:   zigbee2mqtt:error 7/25/2019, 12:39:00 PM No converter available for 'contact' ()
```

Is it possible to get states from this device type ?

Koenkk, 2019-07-25:
@snakuzzo it's not possible, but even if it would it makes no sense because the device is sleeping all most all of the time and thus not answering requests.

snakuzzo, 2019-07-25:
Ok...right!
I was almost sure about this... thank you ;)

alexisayenko, 2019-09-14:
Hi,
I'm using Aqara smart bulb. It sends message with payload `'{"state":"ON","brightness":255,"color_temp":153,"linkquality":44}'`

However, when I try to publish 
`./mosquitto_pub -h 192.168.0.7 -t 'zigbee2mqtt/0x00158d000396ad40/set' -m '{"state":"ON"}'`

I'm getting the following error:

```
Sep 14 04:05:34 raspberrypi npm[413]:   zigbee2mqtt:error 9/14/2019, 4:05:34 AM Zigbee publish to device '0x00158d000396ad40', genOnOff - {state:off} - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Unrecognized command

Sep 14 04:05:34 raspberrypi npm[413]:   zigbee2mqtt:info 9/14/2019, 4:05:34 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: Unrecognized command","meta":{"entity":{"ID":"0x00158d000396ad40","type":"device","friendlyName":"0x00158d000396ad40"},"message":"{State:OFF}"}}'

```

rsalunga29, 2019-10-21:
`zigbee2mqtt/[DEVICE_ID]` and `zigbee2mqtt/[DEVICE_ID]/get` doesn't work whenever I try to subscribe or publish to these topics. Any help?

Device is a Aqara Smart LED Bulb

melonius, 2020-06-15:
It works for me.   I do:  zigbee2mqtt/<DEVICE_ID>/get -m AnyWord 
and the device respond with a publish its status 
I am right, or there is a better system

bachoo786, 2021-03-17:
> It works for me. I do: zigbee2mqtt/<DEVICE_ID>/get -m AnyWord
> and the device respond with a publish with its status
> am I right, or there is a better system?

@melonius can you please share an example? What do you mean by "AnyWord"?

bachoo786, 2021-03-18:
anyone?

bachoo786, 2021-03-18:
> `zigbee2mqtt/[DEVICE_ID]` and `zigbee2mqtt/[DEVICE_ID]/get` doesn't work whenever I try to subscribe or publish to these topics. Any help?
> 
> Device is a Aqara Smart LED Bulb

@rsalunga29 did you resolve this?