Koenkk/zigbee2mqtt #524: Hue + Ambilight connection

regnerus, 2018-10-22:
Philips TV's with Ambilight since 2014 have a setting which allows them to pair with a Hue bridge http://philips-tvconsumercare.kxondemand.com/Portal/en/Faq/All/1645.

I assume that this connection is actually setup via via Wifi instead of Zigbee so this might be the wrong project to ask this, but does anyone know if it is possible to maintain this connection while controlling the Hue lights through Zigbee2MQTT instead of with the Hue Bridge?

tb-killa, 2018-10-25:
from my view this is some sort of lan or wifi communication between tv and the hue bridge who bring zigbee states to the lights.
as we doesn´t emulate the hue bridge and actually only support mqtt like the project name mentioned i would say this is a complete new type of software stuff who is needing.
lets wait what @Koenkk would say.?
maybe you could build something via node-red by yourselve and parse them to mqtt ?
google show me something like this: https://www.npmjs.com/package/ambilight

Koenkk, 2018-10-26:
My guess is that this indeed a non-zigbee protocol, which is out of the scope for this project.

regnerus, 2018-10-26:
@tb-killa thanks, yes I figured that there is some sort of wifi communication between the tv and the bridge.

Thanks for the replies, I will close this issue but definitely report if I am able to get the connection to work via the npm package!

closed 2018-10-26 by regnerus