Koenkk/zigbee2mqtt #526: State of Lumi Magnet Sensor only converted after state change.

Chrischi-, 2018-10-22:
Hi,

after update >=0.1.7 i only get the actual "state" of my Aqara Magnet Door Sensors after 
changing from true to false or vice versa. 

Here you can see that the values for the "state" are also transmitted. 
But Zigbee2MQTT only takes them over in case of a "state change".

_**is this behavior intentional?**_

{
"1":3045, <-- Voltage
"3":30, <-- SoC Temp
"4":17320, <-- ??
"5":72, <-- ??
"6":[0,1], <-- ??
"10":0, <-- ??
"100":0 <-- State (Closed)
}

{
"battery":"100.00",
"voltage":3045,
"linkquality":23
}

```
 zigbee2mqtt:debug 22/10/2018, 14:52:36 Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3045,"3":30,"4":17320,"5":72,"6":[0,1],"10":0,"100":0}}}' of device 'lumi.sensor_magnet.aq2' (0x00158d000232c091)
  zigbee2mqtt:info 22/10/2018, 14:52:36 MQTT publish, topic: 'zigbee2mqtt/Aqara3', payload: '{"battery":"100.00","voltage":3045,"linkquality":23}'
  zigbee2mqtt:debug 22/10/2018, 14:52:36 Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3045,null,30,17320,72,[0,1],null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}' of device 'lumi.sensor_magnet.aq2'
```

side question:
Anyone knows what the meaning of the other values (4, 5, 6, 10)?

Koenkk, 2018-10-22:
The `genBasic` message is ignored, how do you know that `100` is the state?

Chrischi-, 2018-10-22:
> The `genBasic` message is ignored

Of course. ;-)

> how do you know that 100 is the state?

Did a search on GitHub and after that a trial and error with one of the sensors. ^^
However, I am sure that this problem did not occur until 0.1.6.

ptvoinfo, 2018-10-23:
@Koenkk I can confirm that "100" is the state. I'm using this sensor.

Koenkk, 2018-10-23:
@Chrischi- as far as I know, a converter for this message was never in place. Could you make a pr for this?

Chrischi-, 2018-10-23:
Did this out of my head and couldn't test it by now, but i'm curious what happens if i get both messages. `genOnOff` and `genBasic`. 

Duplicate MQTT message?

```
	xiaomi_contact: {
        cid: 'genOnOff',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
           	return {contact: msg.data.data['onOff'] === 0};
        },
        cid: 'genBasic',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
        	return {contact: msg.data.data['65281']['100'] === 0};
		},
	},
```

Should we possibly ignore the `cid: 'genOnOff'`? Because `genBasic` always returns a value. 
However, this leads to a larger number of messages in comparison to `genOnOff` which are 
only transmitted when the state changes.

ptvoinfo, 2018-10-23:
genOnOff sends data immediately after a state change, genBasic sends data rarely.

Chrischi-, 2018-10-23:
>` genOnOff` sends data immediately after a state change, `genBasic` sends data rarely.

@ptvoinfo 
Ok! I thought `genOnOff` will always include `genBasic` when a state changed?!

ptvoinfo, 2018-10-23:
@Chrischi- No, these reports are independent.

Chrischi-, 2018-10-23:
PR opened: https://github.com/Koenkk/zigbee-shepherd-converters/pull/108

Chrischi-, 2018-10-23:
I added a new converter "xiaomi_contact_basic" instead of using the old one as shown above. 
Hope this is OK.