Koenkk/zigbee2mqtt #529: Handle zigbee commands sequentially

Koenkk, 2018-10-23:
As @Skeletorjus mentioned in #481, when sending commands to multiple devices at exactly the same time some commands are not executed.

A practical example of this issue:
I've setup a group of 5 TRADFRI GU10 spots in homeassistant. When I turn the group on, home assistant sends 5 MQTT commands to zigbee2mqtt to turn the 5 bulbs on. It happens quite often that not all bulbs are turned on. However when turning them on separately (manually), everything works fine because there is a delay between the commands.

Zigbee2mqtt should not create a 'burst' of these zigbee commands but instead handle them in a sequential (FIFO) order.

subzero79, 2018-10-23:
Is this ready ? I can give it a try later on. I have 4 osram bulbs on same room that suffer from this.

WebSpider, 2018-10-24:
Tracking this as well, as I have the same issue with a group of 5 Gu10 tradfri's.

Koenkk, 2018-10-24:
Yes, it's in the dev branch. Could you let me know if it fixes the problem?

Skeletorjus, 2018-10-24:
First impressions are good, haven't seen any bulbs not following their group after the update.
Today I discovered to my horror that my CC2531 is running a build dated 20180705, so after a reflash tomorrow I'll do some more testing with bigger groups. 
(I'm hopefult that the newer firmware will also fix the pesky errors I've had with bulbs completely dropping out of the network, fingers crossed!)

WebSpider, 2018-10-24:
I'm trying to figure out how to run the dev branch on hassio. Is it running the -edge package? If so, for some reason mine won't upgrade beyond 0.17

Skeletorjus, 2018-10-24:
Yes, it's the edge package.
You'll have to uninstall and reinstall the addon for it to update.
Uninstalling will not remove the folder containing your configuration, so the only thing you need to do is maybe copy the contents in the config-box for the addon in hass.io to the clipboard, uninstall, reinstall, paste the config back, and start the addon.

subzero79, 2018-10-25:
Seems to work good, now is kind of hard to leave some lights left behind on or off. Takes around 2.5/3 seconds for 4 gu10. I guess this will be improved once we have group support.

Koenkk, 2018-10-25:
Thanks for the feedback all!

In situations where a group of bulbs is one light (e.g. 5 spots above a dinner table) a group should be made, see #15. 

This is for the cases where you, e.g. want to turn off/on all bulbs in your house at once (and thus it doesn't really matter if they respond a few seconds later).

Koenkk, 2018-10-25:
@Skeletorjus it still has to be implemented in zigbee-sheherd but it definitely is something in the pipeline.

Koenkk, 2018-11-16:
Implement in zigbee2mqtt 0.2.0

closed 2018-11-16 by Koenkk

mohoo, 2022-10-19:
Hello,
I have the described problem in version 1.26.0.
I have an Entities Card with all my lamps in Home Assistant with a header toggle. When I want to turn off all lamps using the header toggle, a lot of MQTT /set commands are set. I have already grouped the rooms as much as I can in z2m. However, every 2 or 3 times, some lamps do not go off (or not on), but are incorrectly reported as on or off in MQTT.

If I understand it correctly, a queue was created here to process the commands sequentially. https://github.com/Koenkk/zigbee2mqtt/commit/ddb83cc6e449d2cbebce24e9ced1d3ae4ff7dc0d
I can't find this queue in the current version.