Koenkk/zigbee2mqtt #530: Will this work? 23 devices (Tradfri, Innr and a lot of Xiaomi)

ebivan, 2018-10-24:
Hello there, 
first of all, this is not really an issue, request or enhancement, but I couldn't really find a place for discussion about zigbee2mqtt. So thats why I'm posting here. If this ist not wanted here, please direct me to some place where its appropriate and I'll be on my way.

I'm planing to set up zigbee2mqtt in a (what I now realized) medium sized scale. To get startet I want to connect the following devides to my CC2531:
5x Innr E27 bulbs
3x Tradfri GU10
3x Xiaomi Motion Sensors
3x Xiaomi Window Sensors
3x Xiaomi Cubes
3x Xiaomi Temp/Humidity Sensors
3x eCozy Thermostats (I know this ist not officially supportet at the moment)

Which brings me to a total of 23 devices. When doing some research here, I found out that this is already considered a lot of devices and might not be stable at the current state of the software. Is that true? How are you're experiences with this many devices (and the mentioned devices in particular? 
From what I know so far, at least the Tradfris should act as routers and help to manage the other devices, is that true?

ebivan, 2018-10-24:
Ok, thanks. So for example if i notice issues in a particular room, i can put a cc253x with the router firmware halfway between my zigbee2mqtt gateway and that room and everything should be fine?
The Tradfri bulbs will be installed in the hall approximately in the center of the apartment, so i guess they might provide coverage for the adjacent rooms.
Ok, will order all that stuff now :)

HKC001, 2018-10-28:
I have a mix of 18 XiaoMi/Aqara + Tradfri devices. For me the problem is the cc2531 signal's lack of penetrating power. One thick brick wall with wires inside causes the signal to drop drastically. Your setup looks like you want the bulbs to serve as routers? I haven't had much experience with that (my Tradfri bulbs were just installed), but I hear that bulbs generally make poor routers.

If you run into issues, run a network map and consider adding additional cc2531 routers.

HKC001, 2018-10-30:
@issleiber Adding another cc2531 router should allow you to [increase your network size](https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size).

issleiber, 2018-10-30:
> @issleiber Adding another cc2531 router should allow you to [increase your network size](https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size).

Okay, I will do so, but I had running one 2531 (coord.) and one other 2531 (router) and another Ikea bulb . That means to me, that I was able to use 3 x 15 =45 devices ? ... but currently the pairing of more than 32 devices repeatedly fails ...

Koenkk, 2018-10-30:
@issleiber it doesn't work like that, you only know that you can connect 15 devices directly to the coordinator, you don't know how much you can connect to the cc2531 router and ikea bulb.