Koenkk/zigbee2mqtt #531: Phillips Hue Dimmer stops working periodically (requires re-pairing)

thehaxxa, 2018-10-24:
Hey Folks, 

I have setup the Hue Dimmer and it works really well for a period of a week or so and then drops connection and no longer works, its a brand new device (not a battery issue) and the input is been recorded by zigbee-shepherd. If I re-pair the device it starts to work again, however its become a very frustrating experience. All my Xioami Kit works wonderfully.

I have no idea how to debug this, given its a periodic issue which results in disconnection. 

I have included the debug logs from Zigbee2MQTT:
https://paste.debian.net/plain/1048814

tb-killa, 2018-10-25:
I see this strange behaviour often on my two devices buts its randome timed.
it seems to be that the device lost his state about the bindings and do them again and again.
actually i check if we have to define the right structure of binding and reporting,

thehaxxa, 2018-10-26:
> This could probably be a range issue. When keeping the dimmer close to the CC2531, does it also stop working after a few days?

Possibly, I have bought a router (Xiaomi Power Control) and placed it in between the CC2531 and the Phillips Hue DImmer. I have repaired it and I am now testing the Dimmer. I will report back on findings.

thehaxxa, 2018-10-28:
Well definitely not a range issue. Since using the router, (with link quality consistently above 50), I have experienced another two dropouts. 

If anything the introduction of the router has increased the issue, but more likely I have just noticed it, since monitoring it more carefully. 

To reconnect I have to enter pairing mode and restart zigbee2mqtt. Previous testing with a Phillips Hue hub, I did not experience any issue, so definitely  something software related.

Koenkk, 2018-10-28:
When leaving pairing mode on, do you also experience this?

thehaxxa, 2018-10-28:
Yes, this occurs even if 

> permit_join: true

 is set.

Koenkk, 2018-10-30:
This issue is quite hard to debug, honestly I don't know what to do now..

nxdefiant, 2018-12-23:
Had the same issue on the first day of zigbee2mqtt usage. Above steps worked to reestablish a connection.

RIMperator, 2019-01-08:
I seem to have a similar or maybe even exact the same issue with a Philips Hue Dimmer. My Dimmer basically only works for a few minutes and then stops until it is repaired. What can I do to help isolate the issue?

thehaxxa, 2019-01-10:
> What can I do to help isolate the issue?

Submit your debug output of zigbee-shepherd, (I'm willing to be you'll notice its still receiving commands its just not processing them for some reason, I couldn't isolate why.

chris-jennings, 2019-02-19:
> Submit your debug output of zigbee-shepherd, (I'm expecting its still receiving commands its just not processing them for some reason), I couldn't isolate why.

I just got the same dimmer. Model 324131137411 and it works for a variable amount of time then fails to show up in any Zigbee messages detected by my CC2531. I switched on debug messages and I fail to see any Zigbee packets being detected from the dimmer.

When I restart the zigbee2mqtt service and then repair (all 4 buttons until the green led comes on) then it will work again. I will then see these messages:

`Feb 19 19:40:25 Plex npm[10073]:   zigbee2mqtt:info 2019-2-19 19:40:25 MQTT publish: topic 'homeassistant/0x0017880104afcb62', payload '{"battery":100,"linkquality":65,"action":"down-press","brightness":205}'`

A difficult one to troubleshoot. Perhaps I will need to sniff packets instead when the dimmer is not responding. Do these dimmers have some kind of code rotation or expected sync message? I note that the dimmer tracks the expected brightness internally, I'm not currently using this but I imagine a Hue bridge would sync this to the saved value after a re-pair or reset command.

thehaxxa, 2019-03-07:
I thought I would update this post with the log output from the newest version of zigbee2mqtt:

```
  zigbee2mqtt:info 3/7/2019, 5:46:23 PM MQTT publish: topic 'zigbee2mqtt/Living Room Switch', payload '{"battery":100,"voltage":3055,"linkquality":86,"click":"single"}'
  zigbee2mqtt:info 3/7/2019, 5:46:24 PM MQTT publish: topic 'zigbee2mqtt/Living Room Switch', payload '{"battery":100,"voltage":3055,"linkquality":81,"click":"single"}'
  zigbee2mqtt:warn 3/7/2019, 5:46:35 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOffWithEffect' and data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}'
  zigbee2mqtt:warn 3/7/2019, 5:46:35 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:46:36 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:46:36 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:46:36 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:46:36 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:46:37 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOffWithEffect' and data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}'
  zigbee2mqtt:warn 3/7/2019, 5:46:37 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:46:37 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOffWithEffect' and data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}'
  zigbee2mqtt:warn 3/7/2019, 5:46:37 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:error 3/7/2019, 5:46:40 PM Failed to ping 0x00158d00024f2328
  zigbee2mqtt:warn 3/7/2019, 5:46:41 PM Failed to configure Living Room Light Remote (0x0017880103e44c37) ('Error: Profile: 49246 is not supported.') (attempt #2)
  zigbee2mqtt:warn 3/7/2019, 5:46:41 PM This can be ignored if the device is working properly
  zigbee2mqtt:warn 3/7/2019, 5:46:49 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:46:49 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:46:51 PM Failed to configure Living Room Light Remote (0x0017880103e44c37) ('Error: Profile: 49246 is not supported.') (attempt #3)
  zigbee2mqtt:warn 3/7/2019, 5:46:51 PM This can be ignored if the device is working properly
  zigbee2mqtt:warn 3/7/2019, 5:47:00 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:00 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:04 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:04 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:05 PM No converter available for '324131092621' with cid 'genLevelCtrl', type 'cmdStep' and data '{"cid":"genLevelCtrl","data":{"stepmode":0,"stepsize":30,"transtime":9}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:05 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:05 PM No converter available for '324131092621' with cid 'genLevelCtrl', type 'cmdStep' and data '{"cid":"genLevelCtrl","data":{"stepmode":1,"stepsize":30,"transtime":9}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:05 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:06 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOffWithEffect' and data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:06 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:11 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOffWithEffect' and data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:11 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:13 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:13 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:14 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:14 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:14 PM No converter available for '324131092621' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:14 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
[... cut ...]
  zigbee2mqtt:warn 3/7/2019, 5:47:15 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 3/7/2019, 5:47:15 PM No converter available for '324131092621' with cid 'genLevelCtrl', type 'cmdStep' and data '{"cid":"genLevelCtrl","data":{"stepmode":1,"stepsize":30,"transtime":9}}'
  zigbee2mqtt:warn 3/7/2019, 5:47:15 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

```

I too wonder if there is a code rotation mechanism involved?

chris-jennings, 2019-03-07:
> I too wonder if there is a code rotation mechanism involved?

I would say that fault is not "stops periodically" but instead I think it is related to when the dimmer switch shows to red LED, or when I accidentally trigger a pairing of a nearby Hue light. I don't have any hard evidence except I am now more careful with when I press the Dimmer Switch buttons and not mash buttons when it doesn't respond initially. Now I haven't had to reset the Dimmer Switch for 5 days.

I will try and create the fault and see what are the required conditions.

thehaxxa, 2019-03-07:
> Now I haven't had to reset the Dimmer Switch for 5 days.
> 
> I will try and create the fault and see what are the required conditions.

I hadn't needed to reset (repair) the dimmer for 2 weeks and then it will stop working all of sudden (exactly how you described), and I am careful not to press all buttons to initialize a re-pair. I think you may find the issue appear again in the near future unfortunately.

thehaxxa, 2019-03-15:
I think the issue has been resolved, I have not experienced any disconnects for more than 2 weeks and the latest release has completely reworked the Phillips Hue Remote Controls, so for now I'm going to close this issue. If I experience any problems in the future I may re-open the issue.

thehaxxa, 2019-03-30:
The behavior is sadly still occurring, I need to (infrequently) re-pair my remote and restart zigbee2mqtt service to fix the pairing issue. 🤔

Bastiencc, 2019-04-05:
Hi guys
I'm experencing the same issue @thehaxxa . Only solution to get my remote working again is to remove them, and re-pair. Sometime it's ok for days, sometime it's for hours or minutes :(

@Koenkk you have some ideas about this issues ? 

Tell me if I can help for debug

Bastiencc, 2019-04-05:
If it can help , I get that in log, you can see the `"action":"on-null"` instead of `"action":"on-press" `: 
```
Apr 05 10:28:48 raspberrypi npm[515]:   zigbee2mqtt:error 4/5/2019, 10:28:48 AM Cannot get the Node Descriptor of the Device: 0x0017880104af512e (Error: Timed out after 10000 ms)
Apr 05 10:28:57 raspberrypi npm[515]:   zigbee2mqtt:info 4/5/2019, 10:28:57 AM MQTT publish: topic 'zigbee2mqtt/salon_switch_canape', payload '{"battery":86,"linkquality":39,"duration":0,"counter":1,"brightness":255,"action":"on-null"}'
```

Koenkk, 2019-04-07:
I'm not sure what's cause of this, @ugrug could you take a look at https://github.com/Koenkk/zigbee2mqtt/issues/531#issuecomment-480211038 ?

ugrug, 2019-04-07:
I'll check it!

Bastiencc, 2019-04-09:
Hi @Koenkk , @ugrug and @thehaxxa 

I've update to 1.3.0 and after remove/re-repair all my remote, it seems to work well : I have not experienced any disconnects since 2days.... Wait and see !

Thanks

thehaxxa, 2019-04-13:
> I've updated to 1.3.0 and after remove/re-pair all my remote, it seems to work well : I have not experienced any disconnects since 2days.... Wait and see !

I waited almost 3 weeks before it disconnected again, I think there is some change that occurs during use. Hard to diagnose...

Bastiencc, 2019-04-15:
In my case, it's the very first time I have no disconnect for severals days.

Bastiencc, 2019-04-15:
I get my first disconnect today, on two device... The issue is still here :(
I press the 4 buttons and it works again

Bastiencc, 2019-04-16:
This morining I get some disconnects. When I try to re-paire, I get : 
```
Apr 16 11:00:17 raspberrypi npm[2520]:   zigbee2mqtt:error 4/16/2019, 11:00:17 AM Cannot get the Node Descriptor of the Device: 0x001788010408cd64 (Error: Timed out after 10000 ms)
Apr 16 11:00:37 raspberrypi npm[2520]:   zigbee2mqtt:error 4/16/2019, 11:00:37 AM Cannot get the Node Descriptor of the Device: 0x001788010408cd64 (Error: Timed out after 10000 ms)
```
or : 
```
Apr 16 11:03:35 raspberrypi npm[2615]:   zigbee2mqtt:info 4/16/2019, 11:03:35 AM New device 'undefined' with address 0x001788010408cd64 connected!
Apr 16 11:03:35 raspberrypi npm[2615]:   zigbee2mqtt:info 4/16/2019, 11:03:35 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x001788010408cd64","meta":{}}'
Apr 16 11:03:35 raspberrypi npm[2615]:   zigbee2mqtt:warn 4/16/2019, 11:03:35 AM Device with modelID 'undefined' is not supported.
Apr 16 11:03:35 raspberrypi npm[2615]:   zigbee2mqtt:warn 4/16/2019, 11:03:35 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

MattTheBrofessor, 2019-04-26:
I have the same issue. I connected it a and only programmed the ON and OFF button. This worked fine for like a week but I hardly used it. Then I used Node-Red to program the dimmer, when this worked fine with a light that used the Hue Bridge. Then I used Zigbee2Mqtt to connect a bulb (tried INNR and Aqara) and then it started failing. Don't know if the two are connected. 

My Hue Motion Sensor is also not working. It works just one time and then it disconnects. Both devices were updated to the last software version, so that should not be the problem.

Bastiencc, 2019-05-02:
An other idea: I change my wifi channel to 11 (and zigbee still to channel 11), and now it seems to be good since 15 days...

closed 2019-07-08 by stale[bot]

marothe, 2019-09-16:
Having the same issue - the hue dimmer switch randomly loses connection and you have to re-pair them to Z2M. Did anybody of you guys find a solution for this?

nxdefiant, 2019-09-17:
Only had this problem once so far...

Bastiencc, 2019-09-17:
Same issue here... :(

rm2kdev, 2019-09-25:
Im having the same issue, i've made an extensive thread about my issue here (exact same as you guys 1.6.0)

https://github.com/Koenkk/zigbee-shepherd-converters/issues/637

I've ordered another ccusb to sniff

ahue, 2019-10-23:
Same issue here. Re-pairing possible. Sometimes modelID is detected, sometimes 'undefined'. However pressing the switch does not cause it to light up green and no message is received by zigbee2mqtt. Got another one working again after endless trials but now two others fell out and I cant reproduce re-pairing strategy. Something with stopping zigbee2mqtt, resetting the switch and removing the battery I recall helped last time, but cannot reproduce now.

JohannCR, 2021-03-21:
Same issue. Any progress ?

hacker-h, 2021-11-12:
I can confirm the issue for latest 3.0.X firmware

rrubin0, 2022-07-31:
Im experiencing this too. Repairing is fast but not easy to teach the wife to do. 
All hue lights and xiaomi devices work flawlessly. Im running latest-dev version.

josepki, 2022-10-18:
Same issue here, I have 4 dimmers and all of them have the same problem, they stop working after a few days. Can it be related with pairing other devices? I'm in the process of building my zigbee network and I pair new devices from time to time.

jjimenezlopez, 2022-10-18:
Same issue here, I have 4 dimmers and all of them have the same problem, they stop working after a few days. Can it be related with pairing other devices? I'm in the process of building my zigbee network and I pair new devices from time to time.

hacker-h, 2022-10-18:
@jjimenezlopez Experiencing the issue after weeks of unchanged network with only ~10 devices on a sonoff 3.0 dongle, so this seems to be unrelated

hacker-h, 2022-10-18:
The most interesting observation I made so far is that with two remotes at different locations/networks:
One loses pairing randomly after a few weeks or even days or hours.
The other one never loses it at all.
Since both networks run with the same hardware, software and similar devices I suppose this could also be a hardware issue (some have it, some do not).

jjimenezlopez, 2022-10-18:
@hacker-h thanks for the reply. I have a sonoff 3.0 dongle also.

Do you have the Hue bridge running? I wonder if that can affect some way.

hacker-h, 2022-10-18:
I have no hue bridge, but I just rechecked the config and there is a major difference between the faulty and the working remote:
the faulty runs on firmware version 6.1.1.28573, build date 20190410
the good one runs on firmware version 5.45.1.17846, build date 20160302
Can somebody else confirm this?

jjimenezlopez, 2022-10-18:
I can confirm that my dimmer runs 6.1.1.28573 too! Not sure I can downgrade to try though…

rrubin0, 2022-10-19:
I’ve had this happen too but I just hold down all 4 buttons for a few
seconds and it reboots , reconnects, and works fine again and the problem
is intermittent and infrequent enough to get by. Sonoff 3 radio as well.

gerrytb, 2022-12-05:
I am experimentign the same issue. 
The firmwares version of the dimmer is 6.1.1.28573 too.
Z2M Version is 1.28.4

Has somebody find a solution?

culdaraul, 2022-12-07:
I have similar issues. I have two "older" HUE DimSwitch and one "new" model, the latter has no issues at all. The other two are giving me headaches - completely unreliable even if I'm 2m away from the Sonoff zigbee stick. I have couple of other zigbee devices (32 in total: hue lamps, sonoff relays, ikea switches and lamps) which are working flawlessly but I'm starting to hate Philips. These 2 older dimswitches are using the latest fw.

fdski, 2023-02-01:
same problem here. My dimmer is pretty old, replaced the battery, it registers on the network, then nothing works. 
4 button reset causes it to announce itself and that's all. 
Debug 2023-02-01 23:13:15Device '0x00178801020e72c0' announced itself
Info 2023-02-01 23:13:15MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00178801020e72c0","ieee_address":"0x00178801020e72c0"},"type":"device_announce"}'
Debug 2023-02-01 23:13:17Retrieving state of '0x00178801020e72c0' after reconnect
and that's all, nothing works till next reset, then it announces itself and nothing again. 

I tried the "reset factory" method from https://www.zigbee2mqtt.io/devices/324131092621.html

jimp12004, 2025-01-14:
Old topic, but sadly in 2025 still relevant ...  same behaviour on my side ... and its driving me crazy ... 

3 dimmers have worked 2 years without any flaw until 6 months ago. Then any of them suddenly stoped working at the same time. Repairing (4 buttons) does help immediately for every device, but its frustrating since its hard to argue for "smart home" to my family if its just not working as the good old simple light switch ...

The error occurs nearly every day, i notice it more in the evening hours, but also this morning first time. I think this is subjective observation, since i don't need the dimmers over the day but more in the morning and evening.

Pressing a button does not turn the led light on the dimmer into red (which indicates no connection) but is still green

What i tried (without success):
- factory resettet everything - hue bridge, hue account (removed bridge), app, i actually simulated a new purchase of a complete hue system
- switched hue zigbee channel several times
- assigned a static wifi channel to my wifi router at the other side of the 2,4GHz band to absolutely not interfer with zigbee
- placed the hue 20m away from any wifi router
- firmware updated to the latest version
- changed batteries several times

The most suggested issue that i found was interference but I don't think thats the issue, because it would have more unpredictable behaviour, especially the playing around with the devices positions. Also the reconnect works absolutely flawless til the next drop out.
Hardware error on the dimmers is very very unlikely, all 3 of the must have the same hardware error at the same time - no chance ...

My guess (ordered by probaility) is that:
- the bridge has a hardware fault
- the firmware/device software of the bridge/dimmer switch has a unknown/unfixed bug
- interferences

Any updates on your side? Sad, but I am gonna drop this HUE garbage if there is no solution in the next weeks, especially because any other budget gadget is working without any issue ...

davepastry, 2025-07-31:
Same issue here, picked up a bunch of NIB hue dimmer switches, I can get them added into mqtt and they work perfectly for a little bit, even managed to get them updated to the current OTA firmware, but by the next morning none of them work and no actions are reported in exposes in mqtt when I push the buttons, I'm 10 feet away from my zigbee coordinator and my apartment has plenty of zigbee routers in place. 

somebody help me I bought a whole case of these lol.

fdski, 2025-07-31:
i got mine to work about a year ago. Not sure what changed, just paired them and they are fine since then. 
What i can advise is this: be mindful how you pair your devices. Default way of "enable pairing" is a shortcut to trouble. When you do that, all devices "listen" for new devices. Let's say, one of your zigbee devices, battery driven one, picks up a new device. You'd think in mesh it would "optmise itlself", in my experiance, it does not. It stays as a "child" to whatever connected first. If that parent device dies or falls off the network (or goes to sleep), you're screwed. 

When pairing, enable pairing on wired device that you know will be in vicinity of your dimmer switch, then pair with that. Ever since i did that, problems for me disappeared. My network consists of wired wall light switches - that are always powered and can easily perform repeater function, and PIR detectors, temperature detectors, wireless buttons and such. If any of the latter acts as a "gateway" to the network, it isn't going to work very long. 

Good luck, hope this helps.

IntCleastwood, 2025-11-09:
I don't know if this is a different issue on your/my side, but i never had a partial failure of just some of the dimmers. It was always that every dimmer (total of 5) did not respond anymore for a certain time, mostly 20-30min ... then they suddenly worked again but always all at once.
In the time were the dimmers were off they just started working again when re-pairing them one by one ... i could not resolve this by just moving the dimmer closer to the bridge.

So i think it has not necessarly to do with the distance between a dimmer and the bridge but more the bridge being inside a radio-interferenced area which then will make the bridge completely loosing its capability to communicate with any device.

It works rock solid at the moment, even with being connected with a wire to a wifi bridge instead of a solid ethernet connection to the main switch ...