Koenkk/zigbee2mqtt #532: Osram Smart+ Switch Mini Support

pixeldoc2000, 2018-10-24:
I'm trying to add support for my Osram Smart+ Switch Mini.

# database.db
`{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f00101cf178","nwkAddr":43753,"manufId":4364,"manufName":"OSRAM","powerSource":"Battery","modelId":"Lightify Switch Mini\u0000","epList":[1,2,3],"status":"online","joinTime":1540399537,"endpoints":{"1":{"profId":260,"epId":1,"devId":2064,"inClusterList":[0,1,32,4096,64768],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Lightify Switch Mini\u0000","dateCode":"201711181530470","powerSource":3,"swBuildId":"e.1.9.0M"}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryVoltage":30,"batteryVoltMinThres":24,"batteryVoltThres1":26,"batteryVoltThres2":27,"batteryVoltThres3":28,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{"checkinInterval":0,"longPollInterval":187,"shortPollInterval":4,"fastPollTimeout":720}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Lightify Switch Mini\u0000","dateCode":"201711181530470","powerSource":3,"swBuildId":"e.1.9.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Lightify Switch Mini\u0000","dateCode":"201711181530470","powerSource":3,"swBuildId":"e.1.9.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"lOunfbIm6OpIfTkY"}
`

# devics.js
`    {
        zigbeeModel: ['Lightify Switch Mini\u0000'],
        model: 'AC0251100NJ',
        vendor: 'OSRAM',
        description: 'Smart+ Mini Switch',
        supports: '',
        fromZigbee: [fz.AC0251100NJ_off, fz.AC0251100NJ_on],
        //fromZigbee: [fz.AC0251100NJ_test],
        fromZigbee: [],
    },
`

# fromZigbee.js
`    AC0251100NJ_test: {
        cid: 'genOnOff',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            console.log(msg.data)
        },
    },
    AC0251100NJ_on: {
        cid: 'genOnOff',
        type: 'cmdOn',
        convert: (model, msg, publish, options) => {
            return {action: 'on'};
        },
    },
    AC0251100NJ_off: {
        cid: 'genOnOff',
        type: 'cmdOff',
        convert: (model, msg, publish, options) => {
            return {action: 'off'};
        },
    },
`

# Top (ON) Button shot press debug log
`Oct 24 18:41:05 hassbian npm[18035]: { groupid: 0,
Oct 24 18:41:05 hassbian npm[18035]:   clusterid: 6,
Oct 24 18:41:05 hassbian npm[18035]:   srcaddr: 43753,
Oct 24 18:41:05 hassbian npm[18035]:   srcendpoint: 1,
Oct 24 18:41:05 hassbian npm[18035]:   dstendpoint: 1,
Oct 24 18:41:05 hassbian npm[18035]:   wasbroadcast: 1,
Oct 24 18:41:05 hassbian npm[18035]:   linkquality: 49,
Oct 24 18:41:05 hassbian npm[18035]:   securityuse: 0,
Oct 24 18:41:05 hassbian npm[18035]:   timestamp: 5740574,
Oct 24 18:41:05 hassbian npm[18035]:   transseqnumber: 0,
Oct 24 18:41:05 hassbian npm[18035]:   len: 3,
Oct 24 18:41:05 hassbian npm[18035]:   data: { '0': 17, '1': 13, '2': 1 },
Oct 24 18:41:05 hassbian npm[18035]:   zclMsg:
Oct 24 18:41:05 hassbian npm[18035]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Oct 24 18:41:05 hassbian npm[18035]:      manufCode: 0,
Oct 24 18:41:05 hassbian npm[18035]:      seqNum: 13,
Oct 24 18:41:05 hassbian npm[18035]:      cmdId: 'on',
Oct 24 18:41:05 hassbian npm[18035]:      payload: {} } }
Oct 24 18:41:05 hassbian npm[18035]: cmd { groupid: 0,
Oct 24 18:41:05 hassbian npm[18035]:   clusterid: 6,
Oct 24 18:41:05 hassbian npm[18035]:   srcaddr: 43753,
Oct 24 18:41:05 hassbian npm[18035]:   srcendpoint: 1,
Oct 24 18:41:05 hassbian npm[18035]:   dstendpoint: 1,
Oct 24 18:41:05 hassbian npm[18035]:   wasbroadcast: 1,
Oct 24 18:41:05 hassbian npm[18035]:   linkquality: 49,
Oct 24 18:41:05 hassbian npm[18035]:   securityuse: 0,
Oct 24 18:41:05 hassbian npm[18035]:   timestamp: 5740574,
Oct 24 18:41:05 hassbian npm[18035]:   transseqnumber: 0,
Oct 24 18:41:05 hassbian npm[18035]:   len: 3,
Oct 24 18:41:05 hassbian npm[18035]:   data: { '0': 17, '1': 13, '2': 1 },
Oct 24 18:41:05 hassbian npm[18035]:   zclMsg:
Oct 24 18:41:05 hassbian npm[18035]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Oct 24 18:41:05 hassbian npm[18035]:      manufCode: 0,
Oct 24 18:41:05 hassbian npm[18035]:      seqNum: 13,
Oct 24 18:41:05 hassbian npm[18035]:      cmdId: 'on',
Oct 24 18:41:05 hassbian npm[18035]:      payload: {} } }
Oct 24 18:41:05 hassbian npm[18035]:   zigbee2mqtt:info 2018-10-24 18:41:05 MQTT publish, topic: 'zigbee2mqtt/0x000d6f00101cf178', payload: '{"action":"on","linkquality":49}'
`

# Bottom (OFF) Button shot press debug log
`Oct 24 18:42:34 hassbian npm[18035]: { groupid: 0,
Oct 24 18:42:34 hassbian npm[18035]:   clusterid: 6,
Oct 24 18:42:34 hassbian npm[18035]:   srcaddr: 43753,
Oct 24 18:42:34 hassbian npm[18035]:   srcendpoint: 2,
Oct 24 18:42:34 hassbian npm[18035]:   dstendpoint: 1,
Oct 24 18:42:34 hassbian npm[18035]:   wasbroadcast: 1,
Oct 24 18:42:34 hassbian npm[18035]:   linkquality: 60,
Oct 24 18:42:34 hassbian npm[18035]:   securityuse: 0,
Oct 24 18:42:34 hassbian npm[18035]:   timestamp: 6021609,
Oct 24 18:42:34 hassbian npm[18035]:   transseqnumber: 0,
Oct 24 18:42:34 hassbian npm[18035]:   len: 3,
Oct 24 18:42:34 hassbian npm[18035]:   data: { '0': 17, '1': 14, '2': 0 },
Oct 24 18:42:34 hassbian npm[18035]:   zclMsg:
Oct 24 18:42:34 hassbian npm[18035]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Oct 24 18:42:34 hassbian npm[18035]:      manufCode: 0,
Oct 24 18:42:34 hassbian npm[18035]:      seqNum: 14,
Oct 24 18:42:34 hassbian npm[18035]:      cmdId: 'off',
Oct 24 18:42:34 hassbian npm[18035]:      payload: {} } }
Oct 24 18:42:35 hassbian npm[18035]: cmd { groupid: 0,
Oct 24 18:42:35 hassbian npm[18035]:   clusterid: 6,
Oct 24 18:42:35 hassbian npm[18035]:   srcaddr: 43753,
Oct 24 18:42:35 hassbian npm[18035]:   srcendpoint: 2,
Oct 24 18:42:35 hassbian npm[18035]:   dstendpoint: 1,
Oct 24 18:42:35 hassbian npm[18035]:   wasbroadcast: 1,
Oct 24 18:42:35 hassbian npm[18035]:   linkquality: 60,
Oct 24 18:42:35 hassbian npm[18035]:   securityuse: 0,
Oct 24 18:42:35 hassbian npm[18035]:   timestamp: 6021609,
Oct 24 18:42:35 hassbian npm[18035]:   transseqnumber: 0,
Oct 24 18:42:35 hassbian npm[18035]:   len: 3,
Oct 24 18:42:35 hassbian npm[18035]:   data: { '0': 17, '1': 14, '2': 0 },
Oct 24 18:42:35 hassbian npm[18035]:   zclMsg:
Oct 24 18:42:35 hassbian npm[18035]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Oct 24 18:42:35 hassbian npm[18035]:      manufCode: 0,
Oct 24 18:42:35 hassbian npm[18035]:      seqNum: 14,
Oct 24 18:42:35 hassbian npm[18035]:      cmdId: 'off',
Oct 24 18:42:35 hassbian npm[18035]:      payload: {} } }
Oct 24 18:42:35 hassbian npm[18035]:   zigbee2mqtt:info 2018-10-24 18:42:35 MQTT publish, topic: 'zigbee2mqtt/0x000d6f00101cf178', payload: '{"action":"off","linkquality":60}'
`

Long Button press result in different cid like genLevelCtrl, cmdMoveWithOnOff like Osram 4x-Switch.

Im running zigbee2mqtt@0.1.8 on current hassbian.

Not quite shure how to go further.

tb-killa, 2018-10-25:
Don´t check them in detail but yes seem to be in most stuff equal.

@pixeldoc2000 please provide full debug log with ``DEBUG=zigbee-shepherd* npm start`` and provide comments for each section of test like short press up / down - longpress up / down and the release state.
from this informations we could check and rework.

pixeldoc2000, 2018-11-14:
Looks like this has a Firmware Quirk / Bug.

With Firmware 01090000 the Device sends zigbeeModel: ['Lightify Switch Mini\u0000']. After Update to Firmware 010b0000 (current as of 2018-11-14) via Osram Lightify Gateway they fixed it to zigbeeModel: ['Lightify Switch Mini'].

# devices.js
```
    {
        zigbeeModel: ['Lightify Switch Mini', 'Lightify Switch Mini\u0000'],
        model: 'AC0251100NJ',
        vendor: 'OSRAM',
        description: 'Smart+ Mini Switch',
        supports: '',
        fromZigbee: [fz.AC0251100NJ_off, fz.AC0251100NJ_on],
        //fromZigbee: [fz.AC0251100NJ_test],
        toZigbee: [],
    },
```

Now basic ON/OFF Function is working.

pixeldoc2000, 2018-11-14:
Long Press Top Button:
```
2018-11-14 23:10:36 - debug: Received zigbee message of type 'cmdMoveWithOnOff' with data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":38}}' of device 'Lightify Switch Mini' (0x000d6f00101cf178)
2018-11-14 23:10:36 - warn: No converter available for 'AC0251100NJ' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":38}}'
2018-11-14 23:10:38 - debug: Received zigbee message of type 'cmdStop' with data '{"cid":"genLevelCtrl","data":{}}' of device 'Lightify Switch Mini' (0x000d6f00101cf178)
2018-11-14 23:10:38 - warn: No converter available for 'AC0251100NJ' with cid 'genLevelCtrl', type 'cmdStop' and data '{"cid":"genLevelCtrl","data":{}}'
```

Long Press Middle Button:
```
2018-11-14 23:11:44 - debug: Received zigbee message of type 'cmdMoveToSaturation' with data '{"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}' of device 'Lightify Switch Mini' (0x000d6f00101cf178)
2018-11-14 23:11:44 - warn: No converter available for 'AC0251100NJ' with cid 'lightingColorCtrl', type 'cmdMoveToSaturation' and data '{"cid":"lightingColorCtrl","data":{"saturation":254,"transtime":2}}'
2018-11-14 23:11:45 - debug: Received zigbee message of type 'cmdMoveHue' with data '{"cid":"lightingColorCtrl","data":{"movemode":1,"rate":25}}' of device 'Lightify Switch Mini' (0x000d6f00101cf178)
2018-11-14 23:11:45 - warn: No converter available for 'AC0251100NJ' with cid 'lightingColorCtrl', type 'cmdMoveHue' and data '{"cid":"lightingColorCtrl","data":{"movemode":1,"rate":25}}'

```

Long Press Buttom Button
```
2018-11-14 23:12:08 - debug: Received zigbee message of type 'cmdMove' with data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}' of device 'Lightify Switch Mini' (0x000d6f00101cf178)
2018-11-14 23:12:08 - warn: No converter available for 'AC0251100NJ' with cid 'genLevelCtrl', type 'cmdMove' and data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":38}}'
2018-11-14 23:12:09 - debug: Received zigbee message of type 'cmdStop' with data '{"cid":"genLevelCtrl","data":{}}' of device 'Lightify Switch Mini' (0x000d6f00101cf178)
2018-11-14 23:12:09 - warn: No converter available for 'AC0251100NJ' with cid 'genLevelCtrl', type 'cmdStop' and data '{"cid":"genLevelCtrl","data":{}}'
```

Simular to https://github.com/Koenkk/zigbee2mqtt/issues/159 (Switch 4x EU-LIGHTIFY)

pixeldoc2000, 2018-11-17:
# devices.js
```
    {
        zigbeeModel: ['Lightify Switch Mini', 'Lightify Switch Mini\u0000'],
        model: 'AC0251100NJ',
        vendor: 'OSRAM',
        description: 'Smart+ Switch Mini',
        supports: 'ToDo',
        fromZigbee: [fz.AC0251100NJ_off, fz.AC0251100NJ_on, fz.STS_PRS_251_battery],
        toZigbee: [],
        /*
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('lightingColorCtrl', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryVoltage', 300, 3600, 0, cb),
            ];
             execute(device, actions, callback);
        },
        */
    },
```
https://github.com/Koenkk/zigbee-shepherd-converters/tree/lightify_switch

Now reports Battery Info:
```
2018-11-17 03:41:12 - debug: Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'Lightify Switch Mini' (0x000d6f00101cf178)
2018-11-17 03:41:12 - info: MQTT publish, topic: 'zigbee2mqtt/osram_smart_mini_switch', payload: '{"battery":"100.00","voltage":3000,"linkquality":65,"action":"on"}'
```

Occasionally it sends a bunch of `devInterview` data
```
2018-11-17 03:40:36 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:37 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:37 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:38 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:38 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:40 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:41 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:42 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:43 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:44 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:44 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:45 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:45 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:46 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:47 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:48 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:49 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:50 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
2018-11-17 03:40:50 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f00101cf178"'
```

jamestutton, 2019-01-16:
How did you pair these switches?  Have a bunch of these and have taken your code and some stuff from the tradfri dimmer and got it sort of working.  but dont think they are pairing correctly.  Long hold up and Down near the USB and it flashes blue then RED but no pairing messages recieved.  They will randomly drop in and get the state etc from them but dont think they pair correctly and the switches seem to control all the OSRAM lights in the house cant get them to control just one.

closed 2019-03-24 by stale[bot]

xTim18, 2024-11-11:
Looks like I am running into the same issue. When connecting the switch mini with my home assistance via zigbee2mqtt the interview starts but the device is recogniced as not supported - How would I confirm if the switch is communicating as 'Lightify Switch Mini\u0000' with old firmware or 'Lightify Switch Mini' with new firmware?
Hance any ideas how to fix this without Lightify Gateway?