Koenkk/zigbee2mqtt #534: IKEA bulbs not sending status

pixeldublu, 2018-10-25:
Hello,

Just brought some bulbs and a dimmer, LED1545G12, LED1536G5  and ICTC-G-1

Managed to get them connected but the bulbs do not send any status. 

The dimmer works perfectly ( + other xiaomi devices )

Got no errors in logs, bulbs are connected successfully :(

Things tried:
- both stable and edge versions
- deleting database and rejoining ( like 10 times :) )
- even tried a different fresh flashed CC2531

Any ideas what else i can try ? 

Thanks in advance.

pixeldublu, 2018-10-25:
With bulbs paired and bridge restarted. On first turn on of the bulb:

`[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2018-10-25T08:43:49: PM2 log: Launching in no daemon mode
2018-10-25T08:43:49: PM2 log: App [npm:0] starting in -fork mode-
2018-10-25T08:43:49: PM2 log: App [npm:0] online
> zigbee2mqtt@0.1.8 start /app
> node index.js
  zigbee2mqtt:info 2018-10-25 08:43:53 Logging to directory: '/share/zigbee2mqtt/log/2018-10-25.08-43-53'
  zigbee2mqtt:debug 2018-10-25 08:43:54 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2018-10-25 08:43:54 Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 2018-10-25 08:43:54 Starting zigbee2mqtt version 0.1.8 (commit #d0cf61b)
  zigbee2mqtt:info 2018-10-25 08:43:54 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-10-25 08:43:55 zigbee-shepherd started
  zigbee2mqtt:info 2018-10-25 08:43:55 Coordinator firmware version: '20180507'
  zigbee2mqtt:debug 2018-10-25 08:43:55 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a63","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014d9d7f9","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180507},"startTime":1540446235,"joinTimeLeft":0}
  zigbee2mqtt:info 2018-10-25 08:43:55 Currently 4 devices are joined:
  zigbee2mqtt:info 2018-10-25 08:43:55 0xd0cf5efffe16dd82 (0xd0cf5efffe16dd82): LED1536G5 - IKEA TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (Router)
  zigbee2mqtt:info 2018-10-25 08:43:55 0x90fd9ffffeeab453 (0x90fd9ffffeeab453): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Router)
  zigbee2mqtt:info 2018-10-25 08:43:55 0xd0cf5efffe26054b (0xd0cf5efffe26054b): LED1536G5 - IKEA TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (Router)
  zigbee2mqtt:info 2018-10-25 08:43:55 0x00158d0002a4ab24 (0x00158d0002a4ab24): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
  zigbee2mqtt:info 2018-10-25 08:43:55 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2018-10-25 08:43:55 Connecting to MQTT server at mqtt://192.168.1.7
  zigbee2mqtt:info 2018-10-25 08:43:55 zigbee-shepherd ready
  zigbee2mqtt:info 2018-10-25 08:43:55 Connected to MQTT server
  zigbee2mqtt:info 2018-10-25 08:43:55 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-10-25 08:43:56 MQTT publish, topic: 'homeassistant/light/0xd0cf5efffe16dd82/light/config', payload: '{"brightness":true,"color_temp":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0xd0cf5efffe16dd82/set","state_topic":"zigbee2mqtt/0xd0cf5efffe16dd82","availability_topic":"zigbee2mqtt/bridge/state","name":"0xd0cf5efffe16dd82","unique_id":"0xd0cf5efffe16dd82_light_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-10-25 08:43:56 MQTT publish, topic: 'homeassistant/light/0x90fd9ffffeeab453/light/config', payload: '{"brightness":true,"color_temp":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0x90fd9ffffeeab453/set","state_topic":"zigbee2mqtt/0x90fd9ffffeeab453","availability_topic":"zigbee2mqtt/bridge/state","name":"0x90fd9ffffeeab453","unique_id":"0x90fd9ffffeeab453_light_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-10-25 08:43:56 MQTT publish, topic: 'homeassistant/light/0xd0cf5efffe26054b/light/config', payload: '{"brightness":true,"color_temp":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0xd0cf5efffe26054b/set","state_topic":"zigbee2mqtt/0xd0cf5efffe26054b","availability_topic":"zigbee2mqtt/bridge/state","name":"0xd0cf5efffe26054b","unique_id":"0xd0cf5efffe26054b_light_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-10-25 08:43:56 MQTT publish, topic: 'homeassistant/sensor/0x00158d0002a4ab24/action/config', payload: '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","json_attributes":["battery","voltage","angle","side","from_side","to_side","brightness","angle_x_absolute","angle_y_absolute","angle_z","angle_y","angle_x","unknown_data"],"force_update":true,"state_topic":"zigbee2mqtt/0x00158d0002a4ab24","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d0002a4ab24","unique_id":"0x00158d0002a4ab24_action_zigbee2mqtt"}'
  zigbee2mqtt:debug 2018-10-25 08:43:56 Soft reset timeout disabled
  zigbee2mqtt:info 2018-10-25 08:43:56 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002a4ab24', payload: '{"angle_x":3,"angle_y":2,"angle_z":87,"angle_x_absolute":87,"angle_y_absolute":88,"linkquality":92}'
  zigbee2mqtt:debug 2018-10-25 08:44:07 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:07 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:07 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:08 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:08 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:09 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
[... cut ...]
  zigbee2mqtt:debug 2018-10-25 08:44:10 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:11 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:11 Received zigbee message of type 'devInterview' with data '"0x90fd9ffffeeab453"'
  zigbee2mqtt:debug 2018-10-25 08:44:11 Received zigbee message of type 'devIncoming' with data '"0x90fd9ffffeeab453"' of device 'TRADFRI bulb E27 WS opal 980lm' (0x90fd9ffffeeab453)`

ryanbeaton, 2018-10-25:
Maybe because of old firmware?

> zigbee2mqtt:info 2018-10-25 08:43:55 Coordinator firmware version: '20180507'

ryanbeaton, 2018-10-25:
Sorry I mean only for the 'devInterview' weirdness. I think the Ikea bulbs don't send status.

https://github.com/Koenkk/zigbee2mqtt/issues/369

pixeldublu, 2018-10-25:
Will try later to download and flash new z firmware.

 By toggle i mean turning it on and off via switch. Not using controller or dimmer.

Thanks!

Koenkk, 2018-10-25:
@pixeldublu the bulb doesn't report it status when it is switched on/off using the switch (because there is no power anymore on the bulb)

pixeldublu, 2018-10-25:
Off its obvious... but when turning on there is no status?

tb-killa, 2018-10-25:
Like Zigbee specification for details:
**Joining Announce**
All Zigbee devices send a ZDO Device Announce broadcast transmission when they join a Zigbee network (ZDO cluster ID 0x0013).

We could use some sort of pinging like every 5 seconds and if timeout the device is offline marked.
if device is offline marked and we receive **"ZDO Device Announce broadcast" with payload includes [16-bit address] + [64-bit address] (little-endian byte order (LSB first))** we could mark the device with online and do pinging again like a working queue.

pixeldublu, 2018-10-25:
Seems like i had the wrong idea. 

Sending messages to the bulbs work. Now i have to figure out all values :D

Thanks all!

evildad, 2018-10-25:
I think we should reopen this issue.With the tradfri gateway and the Hass Component switching a bulb on, immediately changes its state to on.
@Koenkk does pinging the bulbs really have such a negative impact?

evildad, 2018-11-01:
@Koenkk are you currently working on the Tradfri state issue? As you are also using them aren't you annoyed of this problem too? ;)

Koenkk, 2018-11-01:
@evildad I'm currently not actively working on this, all my tradfri bulbs are not behind a power switch (as I don't want to power off routers in my zigbee network).

wimpie007, 2018-11-05:
for me, this is a bit of a showstopper. Looking into the conbee solution, but i'm not sure it is implemented there too...

Koenkk, 2018-11-05:
When turning the device off using the switch and having debug log enabled, do you see: `Received zigbee message of type 'devStatus' with data '"online"' of device` in the log?

wimpie007, 2018-11-06:
turning the device OFF with hardware switch doesn't give any log-entry. ( "log_level": "debug",
  "zigbee_shepherd_debug": true)

Koenkk, 2018-11-06:
@wimpie007 and when turning it on?

wimpie007, 2018-11-06:
2018-11-6 08:30:37 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:37 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:38 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:38 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:39 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:39 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:39 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:39 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:40 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:41 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:41 - debug: Received zigbee message of type 'devInterview' with data '"0x000b57fffea710af"'
2018-11-6 08:30:41 - debug: Received zigbee message of type 'devIncoming' with data '"0x000b57fffea710af"' of device 'TRADFRI bulb E27 CWS opal 600lm' (0x000b57fffea710af)
2018-11-6 08:30:41 - info: Device incoming...
2018-11-6 08:30:41 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-11-6 08:30:41 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI bulb E27 CWS opal 600lm' (0x000b57fffea710af)

Koenkk, 2018-11-16:
@wimpie007 this `devStatus` online only seems to happen once. Need to check the behaviour with the TRADFRI gateway.

FHeilmann, 2018-11-22:
Subscribing to this as I have the same issue, my GU10 tradfri lightbulbs don't report state, which causes the lightswitch in HA to go out of sync with the bulb.

danpowell88, 2018-12-18:
I see the following in the logs when I turn on a hue bulb (note i modified the log lines to output the message

> 2018-12-18 08:39:34 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:34 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:34 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:35 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:35 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:35 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:35 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:35 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:53 - warn: Message without device! Message: 0x00178801035609f4
> 2018-12-18 08:39:53 - info: Device incoming...Mesasge:0x00178801035609f4
> 2018-12-18 08:39:53 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
> 2018-12-18 08:39:54 - warn: Message without device! Message:0x00124b001202345c
> 2018-12-18 08:39:54 - warn: Message without device!Message:0x00124b001202345c
> 2018-12-18 08:39:54 - info: Device incoming...Message:0x00124b001202345c
> 2018-12-18 08:39:54 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
> 2018-12-18 08:40:27 - info: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"state":false,"linkquality":47,"description":"18372/0x0017880102D94A5B","type":"NEIGHBOR","rssi":23,"device":{"ieeeAddr":"0x00124b001202345c","friendlyName":"Router","type":"Router","nwkAddr":17448,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.router","status":"online"}}'
> 2018-12-18 08:40:27 - info: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"state":false,"linkquality":47,"description":"12792/0x0017880102D96397","type":"PARENT","rssi":48,"device":{"ieeeAddr":"0x00124b001202345c","friendlyName":"Router","type":"Router","nwkAddr":17448,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.router","status":"online"}}'
> 2018-12-18 08:40:27 - info: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"state":false,"linkquality":47,"description":"12158/0x0017880102D937F7","type":"NEIGHBOR","rssi":1,"device":{"ieeeAddr":"0x00124b001202345c","friendlyName":"Router","type":"Router","nwkAddr":17448,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.router","status":"online"}}'
> 2018-12-18 08:40:27 - info: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"state":false,"linkquality":47,"description":"13539/0x0017880102E33E20","type":"NEIGHBOR","rssi":31,"device":{"ieeeAddr":"0x00124b001202345c","friendlyName":"Router","type":"Router","nwkAddr":17448,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.router","status":"online"}}'
> 2018-12-18 08:40:27 - info: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"state":false,"linkquality":47,"description":"18372/0x0017880102D94A5B","type":"NEIGHBOR","rssi":25,"device":{"ieeeAddr":"0x00124b001202345c","friendlyName":"Router","type":"Router","nwkAddr":17448,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.router","status":"online"}}'
> 2018-12-18 08:40:27 - info: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"state":false,"linkquality":47,"description":"1335/0x0017880104E90115","type":"NEIGHBOR","rssi":5,"device":{"ieeeAddr":"0x00124b001202345c","friendlyName":"Router","type":"Router","nwkAddr":17448,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.router","status":"online"}}'

@Koenkk I'd like to try and handle these, can you suggest where I should start to handle these notifications for the light turning on and the passing it to mqtt?

Interestingly the 2345c device is my router which also shows message without device, is that perhaps as its relaying the message as well?

Koenkk, 2018-12-18:
@danpowell88 does this also happen when turning it off for the second time? (as I remember it didn't with my ikea bulbs)

danpowell88, 2018-12-18:
@Koenkk It does appear it only happens once after a boot of zigbee2mqtt and then no matter how many times i do it its not triggered again.

Is the best bet then to ping all known devices (that are powered and not end devices) that havent had any communication in x timeframe and request a status update from them and publish that?

Koenkk, 2018-12-18:
@danpowell88 I think so, an example how to ping devices is implemented here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/routerPollXiaomi.js

RaveGun, 2018-12-19:
Hi, I am also interested in the topic. So far I replaced 2 normal bulbs with 2 HUE warm white bulbs.

Let's say for one bulb I will be really great to get the feedback if it was turned back on from the wall switch as this one it is constantly forgotten on.

The option that the light starts on full blast on power on it what I need as I am the only one using the home automation in the family.

So pinging or any other option that reports the current real status will be great to have.

Thank you.

danpowell88, 2018-12-19:
I think a ping of online devices would reliably mark them as offline if they are manually switched off (assuming it doesnt cause congestion on the zigbee network), the main part I'm not sure about is notifications when they get switched back on, I think Zigbee-shepherd might be swallowing this for any lights that have already been joined in a session

Koenkk, 2018-12-20:
The best way to do this is to add the following `configure` method.

```
    {
        zigbeeModel: ...
        model: ...
        vendor: ...
        description: ...
        supports: ..
        fromZigbee: ...
        toZigbee: ..
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('genBasic', 'modelId', 10, 60, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

This should produce 1 message every 60 seconds which serves as a heartbeat. A timeout could be given when e.g. 3 messages are missed.

Perhaps somebody could start implementing this in a new extension (https://github.com/Koenkk/zigbee2mqtt/tree/master/lib/extension).

danpowell88, 2018-12-20:
I might take a look at this when I get a chance, but without getting notifications when the device is back online (at least for hue bulbs) its not all that useful for me.

Any suggestions for getting the power on notifications ?

Koenkk, 2018-12-20:
Exactly!

danpowell88, 2018-12-23:
I think with https://github.com/Koenkk/zigbee2mqtt/pull/739 all that might need to happen now is the config in the converters to request a report, @Koenkk ?

danpowell88, 2018-12-23:
I've made some progress, after analyzing the hue traffic it looks like the hue hub polls for on/off state and brightness every second.

```
const logger = require('../util/logger');
const interval = 10000;

/**
 * This extensions polls Xiaomi Zigbee routers to keep them awake.
 */
class RouterPollHue {
    constructor(zigbee, mqtt, state, publishDeviceState) {
        this.zigbee = zigbee;
        this.timer = null;
    }

    onZigbeeStarted() {
        this.startTimer();
    }

    startTimer() {
        this.clearTimer();
        this.timer = setInterval(() => this.handleInterval(), interval);
    }

    clearTimer() {
        if (this.timer) {
            clearTimeout(this.timer);
            this.timer = null;
        }
    }

    stop() {
        this.clearTimer();
    }

    handleInterval() {

        logger.info('polling interval');

        this.zigbee.getAllClients()
            .filter((d) => d.manufId === 4107) // Filter Xiaomi devices
            .filter((d) => d.type === 'Router') // Filter routers
            .filter((d) => d.powerSource && d.powerSource !== 'Battery') // Remove battery powered devices
            .forEach((d) => {         
                
                logger.info(JSON.stringify(d));

                const device =  this.zigbee.shepherd.find(d.ieeeAddr, 11);
                if (device) {
                   
                    device.read('genOnOff', 'onOff');
                    logger.info('polled status of e:11 ' + d.ieeeAddr);
                } 
                
                const device2 =  this.zigbee.shepherd.find(d.ieeeAddr, 242);
                if (device2) {
                   
                    var x = device2.read('genOnOff', 'onOff');
                    logger.info('polled status of e242: ' + d.ieeeAddr);
                    logger.info(JSON.stringify(x));
                }  
            });
    }
}

module.exports = RouterPollHue;
```

I'm not entirely sure how the endpoints work and I've just hard coded 11,242 as thats what the stringified device came back with, however I never see any messages come back with the state so I must be doing something wrong here.

@Koenkk any ideas?

Koenkk, 2018-12-25:
@danpowell88 thanks for finding out how hue does this, however I think that the `report` configuration is a better option. With the `read` we produce 2 zigbee messages, with the `report` only 1 message is produced (as the device send this message on a regular interval) which will reduce load on the zigbee network.

danpowell88, 2018-12-26:
hmm I wonder if the phillips hues do not support the reporting feature.

I have the following:

`[32m zigbee2mqtt:info[39m 2018-12-26 20:46:05 0x00178801035609f4 (0x00178801035609f4): 8718696449691 - Philips Hue White Single bulb B22 (Router)`

And used the method you had above

```
        zigbeeModel: ['LWB010'],
        model: '8718696449691',
        vendor: 'Philips',
        description: 'Hue White Single bulb B22',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: generic.light_onoff_brightness().fromZigbee,
        toZigbee: generic.light_onoff_brightness().toZigbee,
		        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('genBasic', 'modelId', 1, 10, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

After starting up I get 
`[31m zigbee2mqtt:error[39m 2018-12-26 20:46:05 Failed to configure 0x00178801035609f4 0x00178801035609f4 undefined`

The undefined was the contents of `msg` that I added, I also had the same issue if I requested `'genOnOff', 'onOff'`

I also ran with `-e DEBUG=zigbee-shepherd-converters:devices` , this didn't report any additional info during that particular error, however it was working as I got an error about another device a few minutes later. 

```
2018-12-26T10:54:08.213Z zigbee-shepherd-converters:devices Configured '(cb) => ep1.bind('genOnOff', coordinator, cb)' with result 'Error: Timed out after 10000 ms'

[31m zigbee2mqtt:error[39m 2018-12-26 20:54:08 Failed to configure unknown 0x0017880103e44d48 undefined
```

@Koenkk any ideas on where to go from here? is the endpoint parameter to the find method correct?

Koenkk, 2018-12-27:
@danpowell88 let's first try the solution from #761

Koenkk, 2018-12-29:
Device availability has been implemented. See #775 on how to use it.

closed 2018-12-29 by Koenkk