Koenkk/zigbee2mqtt #540: Smartthings Water Leak Sensor (2016 model) issues

dmwyatt, 2018-10-26:
I'm attempting to add support for this device, but I almost can't even get it to pair.

After freshly flashing my CC2531, every time except once I get:

`Received zigbee message of type 'devStatus' with data '"online"' of device 'undefined' (0x24fd5b0001087d5e)`

One time I managed to get it to say something about the device being called `moisturev4`, but the device wouldn't report any events.  Every time I attempt to repair I now get the `undefined` device error.

Here's a debug log via `DEBUG=* npm start`.

http://termbin.com/x4ws

Koenkk, 2018-10-26:
Start with:

- stop zigbee2mqtt
- remove all references of it from `database.db`
- start zigbee2mqtt
- pair
- provide log

Koenkk, 2018-10-29:
@kirovilya do you know why the model number is not announced correctly?

kirovilya, 2018-10-29:
@Koenkk It is another case? 
Strange, but in the logs there is no information about the manufacturer and model.
"already in network"  at the end of the log, as if the device is already paired.

This new `moisturev4` device is not yet registered with our library. Therefore, it did not handle any events.
Try to remove the `database.db` and try again to pair the device.
Some links:
https://community.hubitat.com/t/smartthings-water-leak-sensor/880/15
https://github.com/prjct92eh2/Xiaomi/blob/master/devicetypes/prjct92eh2/xiaomi-aqara-water-sensor.src/xiaomi-aqara-water-sensor.groovy

dmwyatt, 2018-10-29:
Actually, that last log I provided _is_ after removing `database.db`.

Koenkk, 2018-10-29:
@dmwyatt can you find out how you paired it when you saw `moisturev4`. That is the first step, then we can look how to make it report it's state.

dmwyatt, 2018-10-29:
I wish I could.  As far as I can tell, I always pair it in the exact same
way.

As described on samsungs site here:
https://support.smartthings.com/hc/en-us/articles/205957630-SmartThings-Water-Leak-Sensor-previous-generations-

...I hold down the connect button for 5 seconds and release when led starts
blinking red.  The LED blinks blue twice and then it starts to attempt
pairing.

I've done it a dozen times now.  I'll try some more later today.

dmwyatt, 2018-10-29:
Ahha! 

It gives the `unknown` device when attempting to pair when I have my Iris Smart Plug (which is a router) plugged in...even if I don't have it paired to anything.  Which doesn't make much sense to me, but whatever.

Whenever I pair with the Iris Smart Plug disconnected, I get it identified as `moisturev4`.  I will work through adding support today or tomorrow...

dmwyatt, 2018-10-29:
Nope, I lied.  I couldn't get the device to do anything after it paired with the `moisturev4` name.  By that I mean, there was zero activity in the logs (with `DEBUG=* blah blah` on) when I set the sensor in water.

I then deleted all devices from the configuration, deleted database.db, and reflashed for good measure.  Then, attempting to pair in the exact same way I got the `undefined` device again.

dmwyatt, 2018-11-02:
I've tried ten times in a row to pair this thing today and every time it's `unknown`.

Which exact steps should I follow between attempts to repair?  Remove the device it adds to `configuration.yaml`?  Delete `database.db`?  Re-flash my CC2531?

dmwyatt, 2018-11-02:
Hey, I got it to pair with the `moisturev4` name again!  I do not know if this is the reason it worked, but I tried pairing it while the sensor probe was in water.  I would like to try re-pairing again to see if this is the solution to getting it to pair consistently, but after trying so many times, I'm afraid to "unpair" it again.

[Here's the database.db after pairing.](http://termbin.com/dttz)

closed 2019-01-16 by stale[bot]