Koenkk/zigbee2mqtt #542: New Version: ICTC-G-1 (Ikea Tradfri Dimmer)

HKC001, 2018-10-27:
I might have found a new version of the ICTC-G-1.

Running **zigbee2mqtt-0.1.8** and **firmware version: '20180815'**.

Debug log:

```
zigbee2mqtt:info 2018-10-27 10:07:45 Succesfully configured 0x000b57fffe97cc66 0x000b57fffe97cc66
  zigbee2mqtt:debug 2018-10-27 10:07:45 Received zigbee message of type 'undefined' with data '{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":4,"cmdId":"moveWithOnOff","payload":{"movemode":0,"rate":70}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
{ groupid: 0,
  clusterid: 8,
  srcaddr: 59017,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 7,
  securityuse: 0,
  timestamp: 2106616,
  transseqnumber: 0,
  len: 5,
  data: { '0': 17, '1': 4, '2': 5, '3': 0, '4': 70 },
  zclMsg: 
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
     manufCode: 0,
     seqNum: 4,
     cmdId: 'moveWithOnOff',
     payload: { movemode: 0, rate: 70 } } }
cmd { groupid: 0,
  clusterid: 8,
  srcaddr: 59017,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 7,
  securityuse: 0,
  timestamp: 2106616,
  transseqnumber: 0,
  len: 5,
  data: { '0': 17, '1': 4, '2': 5, '3': 0, '4': 70 },
  zclMsg: 
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
     manufCode: 0,
     seqNum: 4,
     cmdId: 'moveWithOnOff',
     payload: { movemode: 0, rate: 70 } } }
  zigbee2mqtt:debug 2018-10-27 10:07:45 Received zigbee message of type 'cmdMoveWithOnOff' with data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":70}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
  zigbee2mqtt:warn 2018-10-27 10:07:45 No converter available for 'ICTC-G-1' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":70}}'
```

tb-killa, 2018-10-27:
Hi @HKC001 thanks for the report.
Does your Value with the ``No converter available`` messages return by usingen the rotation left or right ?
Would you please provide full debug log with comments for the left and right rotation.
From what i expect it could be ``movemode":0`` for ``left`` and ``movemode":1`` for ``right``.

Thanks.

HKC001, 2018-10-28:
Odd, but it seems the movemode is reversed, ``1`` for ``left`` and ``0`` for ``right``.

LEFT
```
zigbee2mqtt:debug 2018-10-28 09:45:16 Received zigbee message of type 'undefined' with data '{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":38,"cmdId":"move","payload":{"movemode":1,"rate":195}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
{ groupid: 0,
  clusterid: 8,
  srcaddr: 28125,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 0,
  securityuse: 0,
  timestamp: 516708,
  transseqnumber: 0,
  len: 5,
  data: { '0': 17, '1': 38, '2': 1, '3': 1, '4': 195 },
  zclMsg: 
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
     manufCode: 0,
     seqNum: 38,
     cmdId: 'move',
     payload: { movemode: 1, rate: 195 } } }
cmd { groupid: 0,
  clusterid: 8,
  srcaddr: 28125,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 0,
  securityuse: 0,
  timestamp: 516708,
  transseqnumber: 0,
  len: 5,
  data: { '0': 17, '1': 38, '2': 1, '3': 1, '4': 195 },
  zclMsg: 
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
     manufCode: 0,
     seqNum: 38,
     cmdId: 'move',
     payload: { movemode: 1, rate: 195 } } }
  zigbee2mqtt:debug 2018-10-28 09:45:16 Received zigbee message of type 'cmdMove' with data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":195}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
  zigbee2mqtt:warn 2018-10-28 09:45:16 No converter available for 'ICTC-G-1' with cid 'genLevelCtrl', type 'cmdMove' and data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":195}}'
  zigbee2mqtt:warn 2018-10-28 09:45:16 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-10-28 09:45:17 MQTT publish, topic: 'zig/dim_1', payload: '{"brightness":168}'
  zigbee2mqtt:debug 2018-10-28 09:45:17 Received zigbee message of type 'undefined' with data '{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":39,"cmdId":"stopWithOnOff","payload":{}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
  zigbee2mqtt:info 2018-10-28 09:45:18 MQTT publish, topic: 'zig/dim_1', payload: '{"brightness":69}'
```
RIGHT
```
zigbee2mqtt:debug 2018-10-28 09:45:45 Received zigbee message of type 'undefined' with data '{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":40,"cmdId":"moveWithOnOff","payload":{"movemode":0,"rate":195}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
{ groupid: 0,
  clusterid: 8,
  srcaddr: 28125,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 5,
  securityuse: 0,
  timestamp: 606126,
  transseqnumber: 0,
  len: 5,
  data: { '0': 17, '1': 40, '2': 5, '3': 0, '4': 195 },
  zclMsg: 
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
     manufCode: 0,
     seqNum: 40,
     cmdId: 'moveWithOnOff',
     payload: { movemode: 0, rate: 195 } } }
cmd { groupid: 0,
  clusterid: 8,
  srcaddr: 28125,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 5,
  securityuse: 0,
  timestamp: 606126,
  transseqnumber: 0,
  len: 5,
  data: { '0': 17, '1': 40, '2': 5, '3': 0, '4': 195 },
  zclMsg: 
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
     manufCode: 0,
     seqNum: 40,
     cmdId: 'moveWithOnOff',
     payload: { movemode: 0, rate: 195 } } }
  zigbee2mqtt:debug 2018-10-28 09:45:45 Received zigbee message of type 'cmdMoveWithOnOff' with data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":195}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
  zigbee2mqtt:warn 2018-10-28 09:45:45 No converter available for 'ICTC-G-1' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":195}}'
  zigbee2mqtt:warn 2018-10-28 09:45:45 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:info 2018-10-28 09:45:45 MQTT publish, topic: 'zig/dim_1', payload: '{"brightness":69}'
  zigbee2mqtt:debug 2018-10-28 09:45:46 Received zigbee message of type 'undefined' with data '{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":41,"cmdId":"stopWithOnOff","payload":{}}' of device 'TRADFRI wireless dimmer' (0x000b57fffe97cc66)
  zigbee2mqtt:info 2018-10-28 09:45:46 MQTT publish, topic: 'zig/dim_1', payload: '{"brightness":158}'
```

tb-killa, 2018-10-29:
Question: Does the dimmer work in all functionality ?
Maybe its easier to ignore this type of message.?

HKC001, 2018-10-31:
@tb-killa I tested it for 2 days, and it works... but not very smoothly. Turning the knob has inconsistent results, sometimes the change is big, small or rarely not working at all. Overall it feels less 'smooth' than the test model at Ikea.

But it basically still works, so I'll close this issue, with a request that errors get fixed if anyone ever has the time. :)

closed 2018-10-31 by HKC001