Koenkk/zigbee2mqtt #545: Samsung SmartThings Multipurpose Sensor (2018)

ericstewart22, 2018-10-28:
Hello,

Could someone help me add support for the Samsung SmartThings Multipurpose Sensor (2018)? M/N is IM6001-MPP01 ([link](https://www.smartthings.com/products/smartthings-multipurpose-sensor)).

I have a CC2531 that I flashed using a NodeMCU and the modified firmware for the Arduino. The CC2531 is connected to a Raspberry Pi running Hass.io only a few feet away. I'm using the following configuration for Hass.io zigbee2mqtt-edge:

```
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": false,
  "permit_join": true,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://192.168.1.10",
  "serial_port": "/dev/ttyACM0",
  "log_level": "debug",
  "disable_led": true,
  "zigbee_shepherd_devices": true
}
```

devices.js and converters/ directory were copied into `share` directory from `zigbee-shepherd-converters repo`.

I've tried multiple times to add support for the device but I can't figure it out. I can pair it but then nothing. No open/close, no tilt/vibration, no temperature; no messages at all. With zigpy/bellows/zha_new I could get open/close and temperature.

I added an entry to devices.js:
```
    {
        zigbeeModel: ['multi'],
        model: 'IM6001-MPP01',
        vendor: 'SmartThings',
        description: 'MultiPurpose Sensor',
        supports: 'contact, temperature, tilt, vibration',
        fromZigbee: [],
        toZigbee: [],
    },
```

Here's the link to the log: [link](https://pastebin.com/P6JZvQsN)

Here's the database configuration: [link](https://pastebin.com/EbhskGa1)

Here's a link to the Samsung SmartThings Device Handler: [link](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/PROD_2018.09.11/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy)

Here's a link to the issue I created for the same device for zha_new: `[link](https://github.com/Yoda-x/ha-zha-new/issues/46)`

tb-killa, 2018-10-28:
Hi welcome to this project and nice to see that there is a new device joining :)

Lets start with something general:

Your EndDevice is from ``manufacturer:Samjin`` so we need something other on configuration side as the normal ``manufacturer:SmartThings`` need (thanks to the ``Samsung SmartThings Device Handler``).

Please replace your ``devices.js`` entry with this ones and report back via ``DEBUG``log and if you see some ``No converter ..`` Messages .?

```
{
	zigbeeModel: ['multi'],
	model: 'IM6001-MPP01',
	vendor: 'Samjin-SmartThings',
	description: 'MultiPurpose Sensor',
	supports: 'contact, temperature, tilt, vibration',
	fromZigbee: [],
	toZigbee: [],
	configure: (ieeeAddr, shepherd, coordinator, callback) => {
		const device = shepherd.find(ieeeAddr, 1);
		const actions = [
			(cb) => device.bind('genPowerCfg', coordinator, cb),
			(cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 30, 3600, 0, cb),
			(cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
			(cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 0, cb),
			(cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
		];

		execute(device, actions, callback);
	},
},
```
We need to figure what cluster ``0xfc02`` mean and how we could add them to zigbee-shepherd ?
@Koenkk Do you know about this ?

ericstewart22, 2018-10-28:
Thanks for the help. I used the following configuration which produced the following logs. I don't see anything in the logs at the moment when I open/close the sensor.

`    {
    	zigbeeModel: ['multi'],
    	model: 'IM6001-MPP01',
    	vendor: 'Samjin-SmartThings',
    	description: 'MultiPurpose Sensor',
    	supports: 'contact, temperature, tilt, vibration',
    	fromZigbee: [fz.generic_temperature],
    	toZigbee: [],
    	configure: (ieeeAddr, shepherd, coordinator, callback) => {
    		const device = shepherd.find(ieeeAddr, 1);
    		const actions = [
    			(cb) => device.bind('genPowerCfg', coordinator, cb),
    			(cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 30, 3600, 0, cb),
    			(cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
    			(cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 0, cb),
    			(cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
    		];
    
    		execute(device, actions, callback);
    	},
    },`

[logs](https://pastebin.com/VmRy6dSc)

Eric

tb-killa, 2018-10-28:
Hi thanks for the report.

Please insert to ``fromZigbee`` this to supress the ``devchange`` converter: ``fz.ignore_temperature_change``.
For the ``open/close`` we need some sort of new converter.

Lets test this:
add to ``fromZigbee`` this too ``fz.IM6001-MPP01__contact``

and this to ``converters/fromZigbee.js``

```
fz.IM6001-MPP01__contact: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            console.log(msg.data)
        },
    },
```

please provide debug-log with ``DEBUG=zigbee-shepherd* npm start`` as with this we also get the converter configuration messages too.

please let the device run with little more time.

ericstewart22, 2018-10-28:
Here's the error I get now:

[logs](https://pastebin.com/7xCaPXNg)

tb-killa, 2018-10-29:
Hi thanks.
Please remove this line and test again.
```
cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 30, 3600, 0, cb),
```
Question: what type of Firmware do you Flash because in the debug log the output of Version is undefined!

ericstewart22, 2018-10-31:
Thanks for the help. Sorry for the delay. I attached the new logs. To be clear here's the configuration I'm using...

First, link to [logs](https://pastebin.com/vQmZwKsB).
The sensor was typically "closed" so "contact: false" seems incorrect to me. I tried "opening" the sensor and leaving it open. Opening it doesn't result in a message and leaving it doesn't seem to change the contact or zoneStatus message.

devices.js

I'm running zigbee2mqtt from Hass.io (i.e. docker) so I modified includes to make sure they were being referenced correctly:
```
const fz = require('/share/zigbee2mqtt/converters/fromZigbee');
const tz = require('/share/zigbee2mqtt/converters/toZigbee');
```
...
```
    {
    	zigbeeModel: ['multi'],
    	model: 'IM6001-MPP01',
    	vendor: 'Samjin-SmartThings',
    	description: 'MultiPurpose Sensor',
    	supports: 'contact, temperature, tilt, vibration',
    	fromZigbee: [fz.generic_temperature,fz.IM6001_MPP01_contact,fz.ignore_temperature_change],
    	toZigbee: [],
    	configure: (ieeeAddr, shepherd, coordinator, callback) => {
    		const device = shepherd.find(ieeeAddr, 1);
    		const actions = [
    			(cb) => device.bind('genPowerCfg', coordinator, cb),
    			(cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
    			(cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 0, cb),
    			(cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
    		];
    
    		execute(device, actions, callback);
    	},
    },
```
fromZigbee.js

Modifying the require in devices.js meant that it couldn't find debounce, so I tried commenting it out for the time being and it seems to be working...

`//const debounce = require('./debounce');`
...
```
    IM6001_MPP01_contact: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            console.log(msg.data)
        },
    },
```

I don't have a CCDebugger so I used a NodeMCU to flash with a modified firmware from [link](https://github.com/kirovilya/files/blob/master/CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex) as per step #7 from [here](https://github.com/Koenkk/zigbee2mqtt/wiki/Alternative-firmware-flashing-methods).

I had issues flashing using NodeMCU so I documented it for others. In case it's relevant.. [link](https://github.com/Koenkk/zigbee2mqtt/issues/78)

pfischbach, 2018-12-04:
Updated to Coordinator firmware version: '20180815' and zigbee2mqtt version 0.2.0 (commit #178f667)
Now both sensors are trying to pair.
Added changes to devices.js but can't get multi sensor to work. water sensor is not added to devices.js since I don't know what to add.
Log from  multi sensor
[Log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2642114/Log.txt)

pfischbach, 2018-12-05:
Tested with fresh flash of firmware and empty data base and configuration file.
Recommend changes to devices.js added, no changes to converters added.
After pairing I don't get any data back from device.
See files for logs.

[Log Samsung Muli Sensor 2018.txt](https://github.com/Koenkk/zigbee2mqtt/files/2650948/Log.Samsung.Muli.Sensor.2018.txt)

pfischbach, 2018-12-06:
After a power reset of the ZigBee stick I got the following log

`zigbee2mqtt:error 2018-12-5 19:50:22 Failed to configure 0x286d9700010285ef 0x286d9700010285ef
zigbee2mqtt:debug 2018-12-5 19:50:46 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
zigbee2mqtt:warn 2018-12-5 19:50:46 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
zigbee2mqtt:warn 2018-12-5 19:50:46 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.`

Koenkk, 2018-12-06:
That looks good, does the sensor report anything when trigger the contact?

pfischbach, 2018-12-07:
Yes, there is a cyclic telegram and a telegram when I actuate the door contact.
The messages are all the same.
`  zigbee2mqtt:debug 2018-12-6 19:29:56 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:29:56 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:29:56 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-6 19:30:26 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:30:26 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:30:26 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-6 19:30:57 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:30:57 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:30:57 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-6 19:31:27 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:31:27 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:31:27 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-6 19:31:57 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:31:57 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:31:57 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-6 19:32:28 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:32:28 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:32:28 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-6 19:32:58 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:32:58 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:32:58 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-6 19:33:28 Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2018-12-6 19:33:28 No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2018-12-6 19:33:28 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
`

Koenkk, 2018-12-07:
@pfischbach If I understand correctly, no different message is reported when you trigger the door contact.

The device did not successfully enroll. Can you add the following; repair the device and provide the log?

```js
    {
    	zigbeeModel: ['multi'],
    	model: 'IM6001-MPP01',
    	vendor: 'Samjin-SmartThings',
    	description: 'MultiPurpose Sensor',
    	supports: 'contact, temperature, tilt, vibration',
    	fromZigbee: [fz.generic_temperature],
    	toZigbee: [],
    	configure: (ieeeAddr, shepherd, coordinator, callback) => {
    		const device = shepherd.find(ieeeAddr, 1);
    		const actions = [
                        (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 23}, cb),
                        (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                        (cb) => device.read('ssIasZone', 'zoneStatus', cb),
    		];
    
    		execute(device, actions, callback);
    	},
    },
```

pfischbach, 2018-12-11:
@Koenkk Thanks!
Added the config, re-flashed the firmware and deleted the database and config devices.
Tested 2 times, see log. Not sure how to proceed.
No success, looks like the device connects but this time it is not sending cyclic telegrams.
[database and config](https://pastebin.com/3KKGYRgYl)
[debug log](https://pastebin.com/pmfkcTBD)
[debug log 2nd](https://pastebin.com/ahnYwywd)

Koenkk, 2018-12-11:
Can you provide the log when running with `DEBUG=* npm start` (need to figure out why configuring fails).

pfischbach, 2018-12-12:
I am new to zigbee2mqtt, using hassio. Enabled zigbee-shepherd debug in the hassio config.

[log](https://pastebin.com/tsYquNGN)

Koenkk, 2018-12-12:
`zigbee_shepherd_debug` is the correct option but I don't see the debug logging, can you check again?

pfischbach, 2018-12-12:
[log 0x286d9700010285ef](https://pastebin.com/E237HAd7)

Koenkk, 2018-12-13:
@danielwelch `zigbee_shepherd_debug` as expected, makes the zigbee-shepherd debug logging appear. However there is also debug logging from zigbee-shepherd-converters which is currently no logged. Is there a way to enable this? (same for #663)

pfischbach, 2018-12-13:
The Hassio log does not log the zigbee-shepherd details into the default file. 
The log window in the HAS web interface does not show all the logs. Window is limited in size and scrolls.
Not sure if there is a different std out file the log goes to.

closed 2019-02-18 by stale[bot]

pfischbach, 2019-02-20:
Started working on this Sensor again.
Had to setup a Raspberry outside of HAS

Can you provide the log when running with `DEBUG=* npm start` (need to figure out why configuring fails).

[debug](https://pastebin.com/mY3BmdmT)
[device.js](https://pastebin.com/JR5kZBwB)

pfischbach, 2019-02-20:
Now I am getting messages after restart
But configuration still fails
zigbee2mqtt:debug 2/20/2019, 4:46:56 PM Received zigbee message of type 'attReport' with data '{"cid":"ssIasZone","data":{"zoneStatus":0}}' of device 'multi' (0x286d9700010285ef)
  zigbee2mqtt:warn 2/20/2019, 4:46:56 PM No converter available for 'IM6001-MPP01' with cid 'ssIasZone', type 'attReport' and data '{"cid":"ssIasZone","data":{"zoneStatus":0}}'
  zigbee2mqtt:warn 2/20/2019, 4:46:56 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.

mclaudiopt, 2019-04-30:
hello
any clues to get this working?

mclaudiopt, 2019-05-12:
hi there. when will the sensor be available on non edge version? thanks

Koenkk, 2019-05-13:
@mclaudiopt should already be working in latest edge.

mclaudiopt, 2019-06-15:
i am on edge but temperature does not update
´``
  zigbee2mqtt:warn 6/15/2019, 9:38:55 PM Failed to configure 0x286d970001010cd0 (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
  zigbee2mqtt:warn 6/15/2019, 9:38:55 PM This can be ignored if the device is working properly
  zigbee2mqtt:debug 6/15/2019, 9:39:10 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'multi' (0x286d970001010cd0) of endpoint 1
  zigbee2mqtt:info 6/15/2019, 9:39:10 PM MQTT publish: topic 'zigbee2mqtt/0x286d970001010cd0', payload '{"contact":true,"battery_low":false,"linkquality":84,"temperature":24.07,"battery":"0.00","voltage":2500}'
  zigbee2mqtt:debug 6/15/2019, 9:39:10 PM Configuring 0x286d970001010cd0 (0x286d970001010cd0) ...
  zigbee2mqtt:debug 6/15/2019, 9:39:10 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x286d970001010cd0"' of device 'multi' (0x286d970001010cd0) of endpoint 1
  zigbee2mqtt:warn 6/15/2019, 9:39:12 PM Failed to configure 0x286d970001010cd0 (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
  zigbee2mqtt:warn 6/15/2019, 9:39:12 PM This can be ignored if the device is working properly
```
@Koenkk

mclaudiopt, 2019-07-13:
@Koenkk do you have any news on this sensor?
i can pair and open/close signal is read correctly. temperature only is read on new pairing

also i get this error
zigbee2mqtt:warn 7/13/2019, 9:08:07 PM Failed to configure Sensor Smartthings (0x286d970001010cd0) ('Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.') (attempt #1)

can you help?

Koenkk, 2019-07-15:
@mclaudiopt no as I don't have the sensor myself, after pairing on zigbee2mqtt 1.5.0, you should see a `Succesfully configured..` for this device after following: http://www.zigbee2mqtt.io/devices/IM6001-MPP01#pairing. Then temperature should work.

mclaudiopt, 2019-07-20:
Hello @Koenkk thanks for the info. i am on hassio and the addon is not yet available for version 1.5. is there anyother way to use it without the addon updated?
thanks

Koenkk, 2019-07-21:
@mclaudiopt seems it's available now, but check: https://github.com/danielwelch/hassio-zigbee2mqtt#warning-breaking-changes-in-version-151

mclaudiopt, 2019-07-21:
hi there @Koenkk already updated....
this erros still appear
```
  zigbee2mqtt:warn 7/21/2019, 8:58:10 PM Failed to configure 0x286d970001010cd0 (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
  zigbee2mqtt:warn 7/21/2019, 8:58:10 PM This can be ignored if the device is working properly
```

and temperature does not update:-(

Koenkk, 2019-07-22:
@mclaudiopt did you re-pair and follow http://www.zigbee2mqtt.io/devices/IM6001-MPP01#pairing? If that doesn't work I'm afraid I cannot help anymore (as I also don't have this device). Perhaps somebody else from this thread can help.

mclaudiopt, 2019-07-22:
@Koenkk now it is working  ! 
thank you so much
is there anyway to skip loggin this errror 
```
  zigbee2mqtt:warn 7/21/2019, 8:58:31 PM Failed to configure 0x286d970001010cd0 (0x286d970001010cd0) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #2)
  zigbee2mqtt:warn 7/21/2019, 8:58:31 PM This can be ignored if the device is working properly
```

Koenkk, 2019-07-22:
Not yet, but you can just ignore that for now.

thardie, 2020-09-09:
I was having the temperature not showing up problem. I did the paring with the suggested magnet every 1 second thing, but do it close to the coordinator. That made things work properly for me. Hopefully that helps someone.