Koenkk/zigbee2mqtt #547: Reconnect a reset Ikea Bulb previously paired.

borpin, 2018-10-28:
A new Ikea Bulb.  I did have it paired with the zigbee2mqtt gateway then accidentally paired it with a n Ikea remote.  I reset the bulb (and the remote) and now I cannot get it to connect to the gateway.  The gateway reports it is paired known.

When trying to connect I get the message `Device incoming` then an mqtt message, then nothing.

I suspect I need to unpair it first. How do I do that?

BudBundi, 2018-10-28:
Turn the power to the lamp on/off 6x times https://www.youtube.com/watch?v=mJm9YpPrGzk

borpin, 2018-10-28:
Not sure what I did but I seem to be talking with the bulb.

This is a fabulous bit of code and resource on setting it up - cheers!

closed 2018-10-28 by borpin